blueman-project/blueman #1594: Bluetooth randomly freezes; stop & restart Bluetooth works

PJW1820, 2021-08-24:
Have 2 devices connected via Bluetooth: 1) BT5.1 Mouse; 2) Blackweb On-Ear Headphones.
Mouse randomly freezes (whether Headphones connected or not).  Work around: turn Bluetooth OFF and then turn Bluetooth ON; click both left & right mouse buttons.

blueman: 5.53-0ubuntu3.3
BlueZ: 5.53-0ubuntu3.3
Distribution: Ubuntu 20.04.3 LTS
Desktop environment: GNOME
Architecture: amd64
Kernel: Linux 5.4.0-81-generic (x86_64) Version: #91-Ubuntu SMP Thu Jul 15 19:09:17 UTC 2021
-Loaded Modules-
rfcomm		: Bluetooth RFCOMM ver 1.11
bnep		: Bluetooth BNEP ver 1.3
btusb		: Generic Bluetooth USB driver ver 0.8
btrtl		: Bluetooth support for Realtek devices ver 0.1
btbcm		: Bluetooth support for Broadcom devices ver 0.1
btintel		: Bluetooth support for Intel devices ver 0.1
bluetooth		: Bluetooth Core ver 2.22

Additional note: Display Bluetooth Devices, "BT5.1 Mouse", Received signal strength 100% (too much), Link Quality: 100% & transmit power level 100% (too high); Blackweb headphones: Received signal strength 50% (optimal), Link Quality: 100% & transmit power level 51% (optimal);

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.  Sorry not that technically savvy

cschramm, 2021-08-28:
> blueman: 5.53-0ubuntu3.3

Not a blueman version. You can check `dpkg -l blueman` but I suppose it's 2.1.3-2ubuntu1.1.

This clearly does not sound like a blueman issue. I'd watch `bluetoothctl` and check if the device disconnects when it happens. If not, it's probably not even related to Bluetooth but some more generic issue with the device.

PJW1820, 2021-08-29:
Thank you Christopher for your reply.
Actually the current version of blueman is 2.1.2-1ubuntu0.2.
I noticed on your website that the most current version is 2.2.2.
Website <https://github.com/blueman-project/blueman/wiki/Packaged-versions>
shows the current version for Ubuntu 20.04 is 2.1.2 but is flagged as
outdated.
Should I update to 2.2.2?  I have downloaded & extracted the files.
Do you have a package that I can run? Or instructions? ( I am not
technically savvy so please keep it simple)

Peter

cschramm, 2021-08-29:
Your problem does not seem blueman related at all, so there's limited value in updating to a more recent version.

PJW1820, 2021-08-29:
Ok, thanks.

closed 2021-11-05 by github-actions[bot]