blueman-project/blueman #1595: dnsmasq failed to start: dnsmasq: unknown interface pan0

basiliscos, 2021-08-25:
blueman: 2.2.1
BlueZ: 5.60
Distribution: void linux
Desktop environment: awesome

When I try to enable NAP (with enabled `dnsmasq`) I got the following error:

```
g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.Failed: Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/blueman/main/DbusService.py", line 126, in _handle_method_call
    ok(method(*args))
  File "/usr/lib/python3.9/site-packages/blueman/plugins/mechanism/Network.py", line 53, in _enable_network
    nc.apply_settings()
  File "/usr/lib/python3.9/site-packages/blueman/main/NetConf.py", line 441, in apply_settings
    self.dhcp_handler.do_apply()
  File "/usr/lib/python3.9/site-packages/blueman/main/NetConf.py", line 113, in do_apply
    raise NetworkSetupError(f"dnsmasq failed to start: {error_msg}")
blueman.main.NetConf.NetworkSetupError: dnsmasq failed to start: dnsmasq: unknown interface pan0
 (0)
```

What can I do?

cschramm, 2021-08-29:
Please run blueman-mechanism with `--loglevel info` (something like `sudo killall blueman-mechanism; sudo /usr/lib/blueman/blueman-mechanism --loglevel info` should do the trick) and try to enable NAP again to see the `dnsmasq` call. It should have `--interface=pan1` so that I don't see where `pan0` could come from. Do you have `/etc/dnsmasq.conf` and does it mention `pan0`?

basiliscos, 2021-08-30:
> Please run blueman-mechanism with --loglevel info (something like sudo killall blueman-mechanism; sudo /usr/lib/blueman/blueman-mechanism --loglevel info should do the trick) and try to enable NAP again to see the dnsmasq call.

I have tried that, but then when I run `blueman-applet` (from non-root user) it just exits:

```
blueman-mechanism 08.12.30 INFO     blueman-mechanism:63 <module>  : Starting blueman-mechanism
blueman-mechanism 08.12.30 INFO     MechanismApplication:79 __init__  : loading Rfcomm
blueman-mechanism 08.12.30 INFO     MechanismApplication:79 __init__  : loading Ppp
blueman-mechanism 08.12.30 INFO     MechanismApplication:79 __init__  : loading RfKill
blueman-mechanism 08.12.30 INFO     MechanismApplication:79 __init__  : loading Network
blueman-mechanism 08.13.00 INFO     MechanismApplication:25 tick      : Exiting
```
Is there any other way? 

> Do you have /etc/dnsmasq.conf and does it mention pan0?

Yes, I have added that as `interface = pan0`. That does not help. By default there is no any `pan0` in `dnsmasq.conf` and the error is also the same.

cschramm, 2021-08-30:
The output is from blueman-mechanism, not blueman-applet and you need to kill other blueman-applet instances first, as described on the Troubleshooting page that you removed from the issue template.

Adding `pan0` to the configuration does not make sense. blueman sets up `pan1` and adds `--interface=pan1` to the call accordingly. The big question is how your dnsmasq comes to try pan0 instead.

basiliscos, 2021-09-01:
No idea how that did happen, but now everything seems working fine. 

Thanks a lot!

closed 2021-09-01 by basiliscos

cschramm, 2021-09-01:
So `pan0` was already configured in `dnsmasq.conf` and seemingly takes precedence over `--interface=pan1`? :thinking:

basiliscos, 2021-09-01:
Yes, I have noted that `dnsmasq.conf` takes precedence over command line options. And this is what I need, because I have to specify other default routing address.