blueman-project/blueman #1598: Unable to open "Devices..." in blueman-applet

alphapanda01, 2021-09-04:
blueman: 2.2.1.r1.g08bbd44a-1
BlueZ: 5.61-1
Distribution: Arch Linux (5.13.13-arch1-1)
Desktop environment: bspwm

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

[Blueman Log](https://pastebin.com/raw/Z8KHW44w)

When every I try to click on `Devices...` option in the blueman-applet, I get an error message in the log and the manager window doesn't load.

[image]

This is part of error which comes up when I click on `Devices...`

ton, 2021-09-08:
I am encountering the same issue. Starting `blueman-manager` from a terminal works without problems.

alphapanda01, 2021-09-09:
@ton that method doesn't work for me, as I can't type unless I connect my keyboard (via bluetooth).

cschramm, 2021-09-17:
Do you use systemd?

If yes, does `dbus-send --dest=org.freedesktop.systemd1 /org/freedesktop/DBus org.freedesktop.systemd1.Activator.ActivationRequest string:blueman-manager.service` trigger blueman-manager?

alphapanda01, 2021-09-18:
@cschramm yes, I do use systemd. 
And no, that command doesn't trigger blueman-manager.

cschramm, 2021-09-18:
What about `systemctl --user start blueman-manager.service` and what does `systemctl --user status blueman-manager.service` tell after trying?

alphapanda01, 2021-09-20:
Sorry for replying late.
This is the output of `systemctl --user status blueman-manager.service`
```
× blueman-manager.service - Bluetooth Manager
     Loaded: loaded (/usr/lib/systemd/user/blueman-manager.service; static)
     Active: failed (Result: exit-code) since Mon 2021-09-20 18:44:54; 3s ago
    Process: 550025 ExecStart=/usr/bin/blueman-manager (code=exited, status=1/FAILURE)
   Main PID: 550025 (code=exited, status=1/FAILURE)
        CPU: 205ms

Sep 20 18:44:53  systemd[831]: Starting Bluetooth Manager...
Sep 20 18:44:54  blueman-manager[550025]: Unable to init server: Could not connect: Connection refused
Sep 20 18:44:54  blueman-manager[550025]: Unable to init server: Could not connect: Connection refused
Sep 20 18:44:54  systemd[831]: Started Bluetooth Manager.
Sep 20 18:44:54  blueman-manager[550025]: Unable to init server: Could not connect: Connection refused
Sep 20 18:44:54  blueman-manager[550025]: cannot open display:
Sep 20 18:44:54  systemd[831]: blueman-manager.service: Main process exited, code=exited, status=1/FAILURE
Sep 20 18:44:54  systemd[831]: blueman-manager.service: Failed with result 'exit-code'.
```

cschramm, 2021-09-20:
Something's not properly set up in your graphical session. Probably the `DISPLAY` variable is missing in the systemd session.

People keep reporting such issues and always come to the conclusion to have some configuration errors. I don't know what configuration errors that are exactly, though. :man_shrugging: See e.g. #1086, #1524, #1541 with solutions like `systemctl --user import-environment` or `dbus-update-activation-environment DISPLAY`.

alphapanda01, 2021-09-20:
aah, I'm very sorry for causing trouble for you, for this simple matter.
I forgot exporting my display at the start. Even the command `dbus-update-activation-environment DISPLAY` seems to work.
I did have my speculations, that I messed up somewhere when I read the systemd output, but was too dumb to notice it.

Thank you, it works now!!!

closed 2021-09-20 by alphapanda01