blueman-project/blueman #1606: Unable to connect (C4)Logitech Adapter

GuiyomeH, 2021-09-26:
blueman: 2.0.8
BlueZ: bluetoothctl 5.50
Distribution: AntiX 19.4 (based on Ubuntu 10)
Desktop environment: ICEWM 2.7.0
pulseaudio: 12.2

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Hello everyone,

I have some issue to connect my (C4)Logitech Adapter (Bluetooth audio adapter for HiFi).

I installed pulseaudio-bluetooth-module as described in https://wiki.debian.org/BluetoothUser/a2dp.

Using Blueman, I get the following error message:
`Connection failed: blueman.bluez.errors.DBusFailedError: Protocol not available`

I ran Blueman in debug mode to get details:
[https://gist.github.com/GuiyomeH/3532e2500c198bc376a1edee2b1f0119](url)

Sorry for I could not achieve to run blueman-report, it looks like it could not connect to GitHub Gist.

I also launched a few bluetoothctl commands.

List controllers:
```
[bluetooth]# list
Controller XX:XX:XX:XX:XX:XX BlueZ 5.50 [default]
Controller YY:YY:YY:YY:YY:YY BlueZ 5.50 Dongle
```

As my default controller is no longer working, I plugged a bluetooth dongle. Thus, switching to the dongle:
```
[bluetooth]# select YY:YY:YY:YY:YY:YY
Controller YY:YY:YY:YY:YY:YY BlueZ 5.50 Dongle [default]
```

Listing bluetooth devices available:
```
[bluetooth]# devices
Device ZZ:ZZ:ZZ:ZZ:ZZ:ZZ (C4)Logitech Adapter
```

Get info on device:
```
[bluetooth]# info ZZ:ZZ:ZZ:ZZ:ZZ:ZZ
Device ZZ:ZZ:ZZ:ZZ:ZZ:ZZ (public)
	Name: (C4)Logitech Adapter
	Alias: (C4)Logitech Adapter
	Class: 0x00240400
	Icon: audio-card
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: no
	LegacyPairing: no
	UUID: Audio Sink                (0000110b-0000-1000-8000-<Many numbers>)
```

Trying to connect:
```
[bluetooth]# connect ZZ:ZZ:ZZ:ZZ:ZZ:ZZ
Attempting to connect to ZZ:ZZ:ZZ:ZZ:ZZ:ZZ
Failed to connect: org.bluez.Error.Failed
```

Can anyone help on this issue?

Do not hesitate to ask further details or suggest any step I might have been missing to provide enough details.

Thanks.

cschramm, 2021-09-28:
The required package's name is actually `pulseaudio-module-bluetooth`. I would definitely not expect to see the protocol not available error if it's in place. Did you restart pulseaudio (or the whole system?). Check `systemctl status bluetooth.service` for additional error information.

GuiyomeH, 2021-10-01:
> The required package's name is actually `pulseaudio-module-bluetooth`. I would definitely not expect to see the protocol not available error if it's in place. Did you restart pulseaudio (or the whole system?). Check `systemctl status bluetooth.service` for additional error information.

Hi @cschramm ,
Thanks for your reply.

My bad, I inverted words in the module. Actually I should have wrote `pulseaudio-module-bluetooth`. Here are details about the package installed (appologize for french text):

```
$ apt show pulseaudio-module-bluetooth -a
Package: pulseaudio-module-bluetooth
Version: 13.0-3~bpo10+1
Priority: optional
Section: sound
Source: pulseaudio
Maintainer: Pulseaudio maintenance team <pkg-pulseaudio-devel@lists.alioth.debian.org>
Installed-Size: 272 kB
Depends: libc6 (>= 2.14), libcap2 (>= 1:2.10), libdbus-1-3 (>= 1.9.14), libpulse0 (= 13.0-3~bpo10+1), libsbc1, pulseaudio (= 13.0-3~bpo10+1), bluez (>= 5.23)
Homepage: https://www.pulseaudio.org
Download-Size: 87,1 kB
APT-Sources: http://deb.debian.org/debian buster-backports/main amd64 Packages
Description: module Bluetooth pour le serveur de sons PulseAudio
 PulseAudio, précédemment connu sous le nom de Polypaudio, est un serveur de
 sons pour les systèmes POSIX et Win32. Il peut remplacer le serveur de sons
 ESD, en apportant une latence plus faible, une qualité de mixage et de
 ré-échantillonnage supérieure et une meilleure architecture générale.
 .
 Ce module permet à PulseAudio de fonctionner avec des périphériques
 Bluetooth, tels qu'un micro-casque ou une passerelle audio
 .
 Le nom de ce module est module-bluetooth

Package: pulseaudio-module-bluetooth
Version: 12.2-4+deb10u1
Priority: optional
Section: sound
Source: pulseaudio
Maintainer: Pulseaudio maintenance team <pkg-pulseaudio-devel@lists.alioth.debian.org>
Installed-Size: 261 kB
Depends: libc6 (>= 2.14), libcap2 (>= 1:2.10), libdbus-1-3 (>= 1.9.14), libpulse0 (= 12.2-4+deb10u1), libsbc1, pulseaudio (= 12.2-4+deb10u1), bluez (>= 5.23)
Homepage: http://www.pulseaudio.org
Tag: admin::hardware, admin::kernel, devel::lang:c, devel::library,
 hardware::TODO, implemented-in::c, role::devel-lib, role::plugin,
 use::driver, works-with::audio
Download-Size: 83,0 kB
APT-Manual-Installed: yes
APT-Sources: http://ftp.fr.debian.org/debian buster/main amd64 Packages
Description: module Bluetooth pour le serveur de sons PulseAudio
 PulseAudio, précédemment connu sous le nom de Polypaudio, est un serveur de
 sons pour les systèmes POSIX et Win32. Il peut remplacer le serveur de sons
 ESD, en apportant une latence plus faible, une qualité de mixage et de
 ré-échantillonnage supérieure et une meilleure architecture générale.
 .
 Ce module permet à PulseAudio de fonctionner avec des périphériques
 Bluetooth, tels qu'un micro-casque ou une passerelle audio
 .
 Le nom de ce module est module-bluetooth
```

I restarted the computer without success.

As AntiX is not using Systemd, I ran the following command:

```
$ service bluetooth status
[ ok ] bluetooth is running.
```

And to get logs:
```
$ sudo more /var/log/syslog | grep bluetooth
Oct  1 22:38:45 MyPC bluetoothd[2185]: a2dp-sink profile connect failed for ZZ:ZZ:ZZ:ZZ:ZZ:ZZ: Protocol not available
```

Does this match the information you were looking for?

Another additional information, the device works fine with my phone and used to work with the previous OS of this computer.

Thanks.

cschramm, 2021-10-02:
Are you sure pulseaudio is running in your session? Does e.g. `pacmd list-modules` work and show you the Bluetooth / BlueZ modules? There should be an `/etc/xdg/autostart/pulseaudio.desktop` but I'm not sure if IceWM picks that up by default.

GuiyomeH, 2021-10-02:
Thanks @cschramm , you're right, pulseaudio was not running. Just `pulseaudio --start` did it.
I'll look for IceWM doc to know how to start it with session.

Issue solved!

Cheers.

closed 2021-10-02 by GuiyomeH