blueman-project/blueman #1607: blueman-applet: recent connections list is empty after suspend-resume cycle

chebro, 2021-09-27:
## System
blueman: 2.2.2-1
BlueZ: 5.61-1
Distribution: Arch Linux
Desktop environment: NA (i3wm)

## Description

The recent connections button is greyed out (empty) whenever my device is resumed from suspend. (yes this is a duplicate of #1131, i have opened this since it is closed)

## Logs

As suggested in #1131 i followed these steps

> 1. kill the blueman-applet 
> 2. run blueman-applet in a terminal
> 3. Attempt to connect to at least one device
> 4. Suspend and resume
> 5. Paste the complete output to a gist and post the link here

here is the terminal output

```bash
$ blueman-applet
blueman-applet 19.59.52 WARNING  DiscvManager:119 update_menuitems: warning: Adapter is None
```

I have also replicated the issue in debug mode (using `blueman-applet --loglevel debug`), here is the complete log: http://0x0.st/-Yth.txt

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2021-09-28:
Thanks for the report. I could replicate the issue. It is a logic problem. When a device gets logically removed, which is usually the case for all devices on suspend, we remove its recently connected services from the list stored in memory but when a device gets logically added, we do not add to the list. A quick and minimal fix for 2.2 is in #1610 but this also motivated me to a little refactoring of the plugin for 2.3 in #1609.

Prakhar-mehta20, 2021-09-30:
I am also facing this same issue, bluetooth won't work after i suspend,

cschramm, 2021-09-30:
@Prakhar-mehta20: That does not sound like the same issue at all.

PR #1610 (2021-10-03): Restore recent connection items on resume
Closes #1607
commit e2f4232c (2021-10-03): Restore recent connection items on resume

Closes #1607
closed 2021-10-03 by cschramm