blueman-project/blueman #1616: No PIN/passcode display, cannot pair keyboards.

bkw777, 2021-10-09:
blueman: 2.1.4
BlueZ: 5.56-0ubuntu4.2
Distribution: xubuntu 21.04
Desktop environment: xfce

* [Y] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Trying to pair a Logitech MX Keys and a Logitech G915 TKL, both keyboards require a passcode to pair, but blueman does not display the passcode, and the pairing attempt always fails.

I was able to pair both keyboards by using bluetooothctl in a terminal following this example:
https://wiki.archlinux.org/title/Bluetooth_keyboard

In bluetoothctl, after issuing the pair command, bluetoothctl displays a passcode, and when that passcode is entered on the keyboard followed by Enter, pairing suceeds.

Trying to follow https://github.com/blueman-project/blueman/wiki/Troubleshooting
The only thing that looks remotely related is the part about python versions and notifications. My python versions matched (python3), and I pasted the example and got the expected sample desktop notification.

After pairing manually with bluetoothctl, the devices operate normally, blueman shows their connection status, and may be connected/disconnected from blueman as long as I don't delete them.

bkw777, 2021-10-09:
Also the passcode is random and different for every pairing attempt, not 0000 or 123456 or any static pre-set passcode.

bkw777, 2021-10-10:
I don't know if it provides any useful info, but here is a log of pairing the same device in bluetoothctl.
```
bkw@fw:~$ bluetoothctl
Agent registered
[CHG] Controller 10:3D:1C:AF:DB:23 Pairable: yes
[MX Anywhere 3]# pairable on
Changing pairable on succeeded
[MX Anywhere 3]# scan
Missing on/off argument
[MX Anywhere 3]# scan on
Discovery started
[CHG] Controller 10:3D:1C:AF:DB:23 Discovering: yes
[NEW] Device E2:9A:C2:C7:30:8C MX Keys
[NEW] Device 6F:52:EE:46:79:34 6F-52-EE-46-79-34
[NEW] Device 08:EF:3B:B2:9E:CD LG LASC55(CD)
[NEW] Device C6:EA:EA:6F:FA:9C C6-EA-EA-6F-FA-9C
[MX Anywhere 3]# pair E2:9A:C2:C7:30:8C
Attempting to pair with E2:9A:C2:C7:30:8C
[CHG] Device E2:9A:C2:C7:30:8C Connected: yes
[agent] Passkey: 137156
[NEW] Primary Service (Handle 0xdbb5)
	/org/bluez/hci0/dev_E2_9A_C2_C7_30_8C/service000a
	00001801-0000-1000-8000-00805f9b34fb
	Generic Attribute Profile
[NEW] Characteristic (Handle 0x61e4)
	/org/bluez/hci0/dev_E2_9A_C2_C7_30_8C/service000a/char000b
	00002a05-0000-1000-8000-00805f9b34fb
	Service Changed
[NEW] Descriptor (Handle 0x76a4)
	/org/bluez/hci0/dev_E2_9A_C2_C7_30_8C/service000a/char000b/desc000d
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
[NEW] Primary Service (Handle 0xdbb5)
	/org/bluez/hci0/dev_E2_9A_C2_C7_30_8C/service000e
	0000180a-0000-1000-8000-00805f9b34fb
	Device Information
[NEW] Characteristic (Handle 0x8694)
	/org/bluez/hci0/dev_E2_9A_C2_C7_30_8C/service000e/char000f
	00002a29-0000-1000-8000-00805f9b34fb
	Manufacturer Name String
[NEW] Characteristic (Handle 0x8f44)
	/org/bluez/hci0/dev_E2_9A_C2_C7_30_8C/service000e/char0011
	00002a24-0000-1000-8000-00805f9b34fb
	Model Number String
[NEW] Characteristic (Handle 0x97f4)
	/org/bluez/hci0/dev_E2_9A_C2_C7_30_8C/service000e/char0013
	00002a25-0000-1000-8000-00805f9b34fb
	Serial Number String
[NEW] Characteristic (Handle 0xa004)
	/org/bluez/hci0/dev_E2_9A_C2_C7_30_8C/service000e/char0015
	00002a27-0000-1000-8000-00805f9b34fb
	Hardware Revision String
[NEW] Characteristic (Handle 0xa7e4)
	/org/bluez/hci0/dev_E2_9A_C2_C7_30_8C/service000e/char0017
	00002a26-0000-1000-8000-00805f9b34fb
	Firmware Revision String
[NEW] Characteristic (Handle 0xb6a4)
	/org/bluez/hci0/dev_E2_9A_C2_C7_30_8C/service000e/char0019
	00002a28-0000-1000-8000-00805f9b34fb
	Software Revision String
[NEW] Characteristic (Handle 0xbce4)
	/org/bluez/hci0/dev_E2_9A_C2_C7_30_8C/service000e/char001b
	00002a50-0000-1000-8000-00805f9b34fb
	PnP ID
[NEW] Primary Service (Handle 0xdbb5)
	/org/bluez/hci0/dev_E2_9A_C2_C7_30_8C/service001d
	0000180f-0000-1000-8000-00805f9b34fb
	Battery Service
[NEW] Characteristic (Handle 0xc9f4)
	/org/bluez/hci0/dev_E2_9A_C2_C7_30_8C/service001d/char001e
	00002a19-0000-1000-8000-00805f9b34fb
	Battery Level
[NEW] Descriptor (Handle 0xe314)
	/org/bluez/hci0/dev_E2_9A_C2_C7_30_8C/service001d/char001e/desc0020
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
[NEW] Primary Service (Handle 0xdbb5)
	/org/bluez/hci0/dev_E2_9A_C2_C7_30_8C/service0048
	00010000-0000-1000-8000-011f2000046d
	Vendor specific
[NEW] Characteristic (Handle 0xefa4)
	/org/bluez/hci0/dev_E2_9A_C2_C7_30_8C/service0048/char0049
	00010001-0000-1000-8000-011f2000046d
	Vendor specific
[NEW] Descriptor (Handle 0xfbf4)
	/org/bluez/hci0/dev_E2_9A_C2_C7_30_8C/service0048/char0049/desc004b
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
[CHG] Device E2:9A:C2:C7:30:8C UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Device E2:9A:C2:C7:30:8C UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Device E2:9A:C2:C7:30:8C UUIDs: 0000180a-0000-1000-8000-00805f9b34fb
[CHG] Device E2:9A:C2:C7:30:8C UUIDs: 0000180f-0000-1000-8000-00805f9b34fb
[CHG] Device E2:9A:C2:C7:30:8C UUIDs: 00001812-0000-1000-8000-00805f9b34fb
[CHG] Device E2:9A:C2:C7:30:8C UUIDs: 00010000-0000-1000-8000-011f2000046d
[CHG] Device E2:9A:C2:C7:30:8C ServicesResolved: yes
[CHG] Device E2:9A:C2:C7:30:8C Paired: yes
Pairing successful
[CHG] Device E2:9A:C2:C7:30:8C WakeAllowed: yes
[CHG] Device E2:9A:C2:C7:30:8C Modalias: usb:v046DpB35Bd0011
[DEL] Device 6F:52:EE:46:79:34 6F-52-EE-46-79-34
[DEL] Device 08:EF:3B:B2:9E:CD LG LASC55(CD)
[NEW] Device 6F:52:EE:46:79:34 6F-52-EE-46-79-34
[NEW] Device 08:EF:3B:B2:9E:CD LG LASC55(CD)
[MX Anywhere 3]# pairable off
Changing pairable off succeeded
[CHG] Controller 10:3D:1C:AF:DB:23 Pairable: no
[DEL] Device C6:EA:EA:6F:FA:9C C6-EA-EA-6F-FA-9C
[MX Anywhere 3]# quit
bkw@fw:~$ 
```

cschramm, 2021-10-10:
Perfect, thanks. So... I'd expect to see our agent's `DisplayPasskey` being called in the blueman log. Instead we just receive `Authentication Canceled` from BlueZ. A bluetoothd debug log should give more insights on why that's happening. What I suspect is that blueman's agent is not registered as the default (The bluetoothd debug output from `src/agent.c` is the most interesting for that).