blueman-project/blueman #1618: Issues with Blueman on Kali Linux for Raspberry Pi 4

CaptCaddburry, 2021-10-20:
blueman: 2.2.2-1
BlueZ:5.61-1+kali
Distribution: http://http.kali.org/kali kali-rolling/main armhf Packages
Desktop environment: xfce

I have been trying to set up blueman-services on my raspberry pi running Kali Linux. When I hit enter to start the services, the terminal comes up with an error:

blueman-services 10.24.10 ERROR   Functions:313_netmask_for_ifacename: siocgifnetmask failed
File "/usr/lib/python3/dist-packages/blueman/main/Services.py", line 144, in on_selection_changed
    self.set_page(rowid)
File "/usr/lib/python3/dist-packages/blueman/main/Services.py", line 128, in set_page
    inst.on_load(self.box)
File "/usr/lib/python3/dist-packages/blueman/plugins/services/Network.py", line 39, in on_load
    ipiface = iPad dress.ip_interface('/'.join(cast(Tuple[str, str], netifs[iface])))
TypeError: sequence item 1: expected str instance, NoneType found

I did have my Pi installed with the standard Raspbian OS and the blueman service worked perfectly with it. But once I have used the Raspbian Imaging Software to change my OS to Kali Linux, now the blueman service doesn't work. I was wondering if anyone knew of any issues with it, I didn't want to have to reformat my sd card again to see if there were any differences in the code or if it was pulling a different version number than what Kali is pulling.

qkum, 2022-03-03:
**I just got the exact same error message today with `kali-linux-2022.1-raspberry-pi-zero-2-w-armhf-img-xz`.**

The only thing I have changed since installation is `sudo apt-get update & upgrade`.

The rest is all default.

**Here is the output of your python script:**
[image]

Thanks for your time & help 🏆

qkum, 2022-03-03:
I did pair it with my phone to test if bluetooth worked on this version....it did not on the 2021.4 version.

Maybe it is because it is looking for my phone? (that is a device it could be looking for that is not available atm)

qkum, 2022-03-03:
Can you tell me where the file is that I have to edit to change the same settings from the terminal? 

- I have been looking/googling but without much luck for now. I have used many hours on this bluetooth jazz already (days). The answer would be much appreciated 💯

qkum, 2022-03-03:
As I spotted that you guys are wise people using **Python**  I have begun troubleshooting it myself.

Python is after all the language of the future and every wise newbie starts with learning Python.

I can, for now, say that 'iface' returns 'lo' as the first interface and nothing ' ' as the second.
I found out by adding a "print(iface)" in the for loop.

Here is my interfaces:

[image]

So, there clearly is eth0 with internet connection and it all that it is missing.

I will try to connect to my wifi to see if it changes anything.

qkum, 2022-03-03:
It did not change anything to connect to my wifi.

It still gets 'lo' as the first interface and nothing as second in the loop.

**This is the error it keeps throwing:**
- For every interation, it prints "iface" and the content of iface. 
```
blueman-services 05.45.42 ERROR    Functions:313 _netmask_for_ifacename: siocgifnetmask failed
blueman-services 05.45.42 ERROR    Functions:313 _netmask_for_ifacename: siocgifnetmask failed
iface
lo
iface

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/main/Services.py", line 144, in on_selection_changed
    self.set_page(rowid)
  File "/usr/lib/python3/dist-packages/blueman/main/Services.py", line 128, in set_page
    inst.on_load(self.box)
  File "/usr/lib/python3/dist-packages/blueman/plugins/services/Network.py", line 41, in on_load
    ipiface = ipaddress.ip_interface('/'.join(cast(Tuple[str, str], netifs[iface])))
TypeError: sequence item 1: expected str instance, NoneType found
```

cschramm, 2022-03-09:
I guess the structure has a different size in your cases, however, I could not replicate that with a kalilinux/kali-rolling docker image.

Could you please provide the output of the following Python script?

```py
import array
import fcntl
import socket
import struct

names = array.array('B', 4096 * b'\0')

sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)

max_bytes_out = struct.unpack('iL', fcntl.ioctl(sock.fileno(), 0x8912, struct.pack('iL', 4096, names.buffer_info()[0])))[0]

namestr = names.tobytes()

print(namestr[:max_bytes_out])
```

cschramm, 2022-03-09:
Ah, gosh, I see. https://github.com/blueman-project/blueman/pull/1022 introduced a fixed size for `struct ifreq` but that depends on the size of `long` and thus the assumption is only correct for 64 Bit systems...

cschramm, 2022-03-09:
I think the 40 should actually be a 32 for you. Can you test #1687?

infirit, 2022-03-28:
Oops 😨 . @qkum can you test this?

Would any 32bit system do @cschram? I'll setup 32bit VM to test if @qkum can't.

infirit, 2022-03-28:
> ```python
> import array
> import fcntl
> import socket
> import struct
> 
> names = array.array('B', 4096 * b'\0')
> 
> sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
> 
> max_bytes_out = struct.unpack('iL', fcntl.ioctl(sock.fileno(), 0x8912, struct.pack('iL', 4096, names.buffer_info()[0])))[0]
> 
> namestr = names.tobytes()
> 
> print(namestr[:max_bytes_out])
> ```

Here is the output on 32bit debian 11. I'll test the PR later this week.

```$ python3 ifreq.py 
b'lo\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\x00\x00\x7f\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00enp0s3\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\x00\x00\n\x00\x02\x0f\x00\x00\x00\x00\x00\x00\x00\x00'```

PR #1687 (2022-03-29): Fix network interface iteration on 32 bit systems
The size of struct ifreq is dominated by the size of struct ifmap whose size depends on two long fields, so that it is actually just 32 bytes on 32 bit systems.

Fixes #1618
commit fc99cf5b (2022-03-29): Fix network interface iteration on 32 bit systems

The size of struct ifreq is dominated by the size of struct ifmap whose size depends on two long fields, so that it is actually just 32 bytes on 32 bit systems.

Fixes #1618
commit d0685cb2 (2022-06-09): Fix network interface iteration on 32 bit systems

The size of struct ifreq is dominated by the size of struct ifmap whose size depends on two long fields, so that it is actually just 32 bytes on 32 bit systems.

Fixes #1618
closed 2022-03-29 by cschramm