blueman-project/blueman #1629: A2DP audio device PAIRED but fails to connect

BishalMahanta900, 2021-11-03:
Nov 03 00:38:04 LinuxPC bluetoothd[4130]: src/profile.c:ext_io_disconnected() Unable to>
Nov 03 01:02:08 LinuxPC bluetoothd[4130]: src/service.c:btd_service_connect() a2dp-sink>
Nov 03 01:02:10 LinuxPC bluetoothd[4130]: /org/bluez/hci0/dev_84_56_14_50_FF_98/sep1/fd>
Nov 03 01:37:43 LinuxPC bluetoothd[4130]: src/profile.c:ext_io_disconnected() Unable to>
Nov 03 01:37:52 LinuxPC bluetoothd[4130]: Terminating
Nov 03 01:37:52 LinuxPC bluetoothd[4130]: Endpoint unregistered: sender=:1.88 path=/Med>
Nov 03 01:37:52 LinuxPC bluetoothd[4130]: Endpoint unregistered: sender=:1.88 path=/Med>
Nov 03 01:37:52 LinuxPC systemd[1]: Stopping Bluetooth service...
Nov 03 01:37:52 LinuxPC bluetoothd[4130]: Endpoint unregistered: sender=:1.88 path=/Med>
Nov 03 01:37:52 LinuxPC bluetoothd[4130]: Endpoint unregistered: sender=:1.88 path=/Med>
Nov 03 01:37:52 LinuxPC bluetoothd[4130]: Endpoint unregistered: sender=:1.88 path=/Med>
Nov 03 01:37:52 LinuxPC bluetoothd[4130]: Endpoint unregistered: sender=:1.88 path=/Med>
Nov 03 01:37:52 LinuxPC bluetoothd[4130]: Endpoint unregistered: sender=:1.88 path=/Med>
Nov 03 01:37:52 LinuxPC bluetoothd[4130]: Endpoint unregistered: sender=:1.88 path=/Med>
Nov 03 01:37:52 LinuxPC bluetoothd[4130]: Stopping SDP server
Nov 03 01:37:52 LinuxPC bluetoothd[4130]: Exit
Nov 03 01:37:52 LinuxPC systemd[1]: bluetooth.service: Deactivated successfully.
Nov 03 01:37:52 LinuxPC systemd[1]: Stopped Bluetooth service.
-- Boot f1b41100107d4238bafc40fe05188985 --
Nov 03 10:58:16 LinuxPC systemd[1]: Starting Bluetooth service...
Nov 03 10:58:16 LinuxPC bluetoothd[4109]: Bluetooth daemon 5.61
Nov 03 10:58:16 LinuxPC systemd[1]: Started Bluetooth service.
Nov 03 10:58:16 LinuxPC bluetoothd[4109]: Starting SDP server
Nov 03 01:37:52 LinuxPC systemd[1]: Stopped Bluetooth service.
-- Boot f1b41100107d4238bafc40fe05188985 --
Nov 03 10:58:16 LinuxPC systemd[1]: Starting Bluetooth service...
Nov 03 10:58:16 LinuxPC bluetoothd[4109]: Bluetooth daemon 5.61
Nov 03 10:58:16 LinuxPC systemd[1]: Started Bluetooth service.
Nov 03 10:58:16 LinuxPC bluetoothd[4109]: Starting SDP server
Nov 03 10:58:17 LinuxPC bluetoothd[4109]: Bluetooth management interface 1.21 initializ>
Nov 03 10:58:17 LinuxPC bluetoothd[4109]: profiles/sap/server.c:sap_server_register() S>
Nov 03 10:58:17 LinuxPC bluetoothd[4109]: sap-server: Operation not permitted (1)
Nov 03 11:01:45 LinuxPC bluetoothd[4109]: src/service.c:btd_service_connect() a2dp-sink>
Nov 03 11:02:23 LinuxPC bluetoothd[4109]: src/service.c:btd_service_connect() a2dp-sink>
Nov 03 11:02:29 LinuxPC bluetoothd[4109]: src/service.c:btd_service_connect() a2dp-sink>
Nov 03 11:03:49 LinuxPC bluetoothd[4109]: src/service.c:btd_service_connect() a2dp-sink>
Nov 03 11:04:14 LinuxPC bluetoothd[4109]: src/service.c:btd_service_connect() a2dp-sink>
Nov 03 11:07:50 LinuxPC bluetoothd[4109]: Terminating
Nov 03 11:07:50 LinuxPC systemd[1]: Stopping Bluetooth service...
Nov 03 11:07:50 LinuxPC bluetoothd[4109]: Stopping SDP server
Nov 03 11:07:50 LinuxPC bluetoothd[4109]: Exit
Nov 03 11:07:50 LinuxPC systemd[1]: bluetooth.service: Deactivated successfully.
Nov 03 11:07:50 LinuxPC systemd[1]: Stopped Bluetooth service.
-- Boot 560e098672824485ad787445cd775ee1 --
Nov 03 11:09:12 LinuxPC systemd[1]: Starting Bluetooth service...
Nov 03 11:09:12 LinuxPC bluetoothd[3541]: Bluetooth daemon 5.61
Nov 03 11:09:12 LinuxPC systemd[1]: Started Bluetooth service.
Nov 03 11:09:12 LinuxPC bluetoothd[3541]: Starting SDP server
Nov 03 11:09:12 LinuxPC bluetoothd[3541]: Bluetooth management interface 1.21 initializ>
Nov 03 11:09:12 LinuxPC bluetoothd[3541]: profiles/sap/server.c:sap_server_register() S>
Nov 03 11:09:12 LinuxPC bluetoothd[3541]: sap-server: Operation not permitted (1)
Nov 03 11:09:34 LinuxPC bluetoothd[3541]: src/service.c:btd_service_connect() a2dp-sink>
Nov 03 11:24:11 LinuxPC bluetoothd[3541]: src/service.c:btd_service_connect() a2dp-sink>
Nov 03 11:24:16 LinuxPC bluetoothd[3541]: src/service.c:btd_service_connect() a2dp-sink>
Nov 03 11:24:17 LinuxPC bluetoothd[3541]: src/service.c:btd_service_connect() a2dp-sink>
Nov 03 11:24:37 LinuxPC bluetoothd[3541]: src/service.c:btd_service_connect() a2dp-sink>
lines 3752-3778/3778 (END)
Nov 03 10:58:17 LinuxPC bluetoothd[4109]: Bluetooth management interface 1.21 initialized
Nov 03 10:58:17 LinuxPC bluetoothd[4109]: profiles/sap/server.c:sap_server_register() Sap driver initialization failed.
Nov 03 10:58:17 LinuxPC bluetoothd[4109]: sap-server: Operation not permitted (1)
Nov 03 11:01:45 LinuxPC bluetoothd[4109]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 84:56:14:50:FF:98: Protocol not available
Nov 03 11:02:23 LinuxPC bluetoothd[4109]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 84:56:14:50:FF:98: Protocol not available
Nov 03 11:02:29 LinuxPC bluetoothd[4109]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 84:56:14:50:FF:98: Protocol not available
Nov 03 11:03:49 LinuxPC bluetoothd[4109]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 84:56:14:50:FF:98: Protocol not available
Nov 03 11:04:14 LinuxPC bluetoothd[4109]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 84:56:14:50:FF:98: Protocol not available
Nov 03 11:07:50 LinuxPC bluetoothd[4109]: Terminating
Nov 03 11:07:50 LinuxPC systemd[1]: Stopping Bluetooth service...
Nov 03 11:07:50 LinuxPC bluetoothd[4109]: Stopping SDP server
Nov 03 11:07:50 LinuxPC bluetoothd[4109]: Exit
Nov 03 11:07:50 LinuxPC systemd[1]: bluetooth.service: Deactivated successfully.
[... 71 lines cut ...]
Nov 03 10:58:17 LinuxPC bluetoothd[4109]: sap-server: Operation not permitted (1)
Nov 03 11:01:45 LinuxPC bluetoothd[4109]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 84:56:14:50:FF:98: Protocol not available
Nov 03 11:02:23 LinuxPC bluetoothd[4109]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 84:56:14:50:FF:98: Protocol not available
Nov 03 11:02:29 LinuxPC bluetoothd[4109]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 84:56:14:50:FF:98: Protocol not available
Nov 03 11:03:49 LinuxPC bluetoothd[4109]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 84:56:14:50:FF:98: Protocol not available
Nov 03 11:04:14 LinuxPC bluetoothd[4109]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 84:56:14:50:FF:98: Protocol not available
Nov 03 11:07:50 LinuxPC bluetoothd[4109]: Terminating
Nov 03 11:07:50 LinuxPC systemd[1]: Stopping Bluetooth service...
Nov 03 11:07:50 LinuxPC bluetoothd[4109]: Stopping SDP server
Nov 03 11:07:50 LinuxPC bluetoothd[4109]: Exit
Nov 03 11:07:50 LinuxPC systemd[1]: bluetooth.service: Deactivated successfully.
Nov 03 11:07:50 LinuxPC systemd[1]: Stopped Bluetooth service.
-- Boot 560e098672824485ad787445cd775ee1 --
Nov 03 11:09:12 LinuxPC systemd[1]: Starting Bluetooth service...
Nov 03 11:09:12 LinuxPC bluetoothd[3541]: Bluetooth daemon 5.61
Nov 03 11:09:12 LinuxPC systemd[1]: Started Bluetooth service.
Nov 03 11:09:12 LinuxPC bluetoothd[3541]: Starting SDP server
Nov 03 11:09:12 LinuxPC bluetoothd[3541]: Bluetooth management interface 1.21 initialized
Nov 03 11:09:12 LinuxPC bluetoothd[3541]: profiles/sap/server.c:sap_server_register() Sap driver initialization failed.
Nov 03 11:09:12 LinuxPC bluetoothd[3541]: sap-server: Operation not permitted (1)
Nov 03 11:09:34 LinuxPC bluetoothd[3541]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 84:56:14:50:FF:98: Protocol not available
Nov 03 11:24:11 LinuxPC bluetoothd[3541]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 84:56:14:50:FF:98: Protocol not available
Nov 03 11:24:16 LinuxPC bluetoothd[3541]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 84:56:14:50:FF:98: Protocol not available
Nov 03 11:24:17 LinuxPC bluetoothd[3541]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 84:56:14:50:FF:98: Protocol not available
Nov 03 11:24:37 LinuxPC bluetoothd[3541]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 84:56:14:50:FF:98: Protocol not available

cschramm, 2021-11-03:
Your system seems to lack something that handles audio devices for BlueZ. In a typical setup that would be pulseaudio's bluetooth module.

BishalMahanta900, 2021-11-03:
> Your system seems to lack something that handles audio devices for BlueZ. In a typical setup that would be pulseaudio's bluetooth module.
[image]

cschramm, 2021-11-03:
You can check `pactl list | grep module-bluetooth-discover` and try `sudo pactl load-module module-bluetooth-discover` but it should be loaded in an Ubuntu setup.

BishalMahanta900, 2021-11-03:
> You can check `pactl list | grep module-bluetooth-discover` and try `sudo pactl load-module module-bluetooth-discover` but it should be loaded in an Ubuntu setup.

[image]

cschramm, 2021-11-03:
So the pulseaudio service does not seem to be running at all.

BishalMahanta900, 2021-11-03:
> So the pulseaudio service does not seem to be running at all.

yes, well in that case what should i do?

BishalMahanta900, 2021-11-03:
> Fix it. smile Check `systemctl --user status pulseaudio.service` as a starting point.

[image]

cschramm, 2021-11-04:
I don't think root is meant to have a user session bus at all. It's not the user you run the desktop session in, is it? If it is, I guess it's not expected to have audio, you could try `pulseaudio -D` though. Anyway, that seems like some concept of Kali Linux and gets way off topic here.

MasterSaha, 2021-11-05:
First - 
`pulseaudio -D`
Then,
`systemctl --user status pulseaudio.service`
Worked for me

Bishal-Mahanta, 2021-11-05:
[image]

cschramm, 2021-11-05:
Random guess: Due to the same broken desktop session, you also do not have a working notification daemon. The `blueman-applet --loglevel debug` output of the process should show more.

Subhamkonar84, 2021-11-06:
use this code this will definitely work---   `pulseaudio --start` this worked for me.

Bishal-Mahanta, 2021-11-07:
> Random guess: Due to the same broken desktop session, you also do not have a working notification daemon. The `blueman-applet --loglevel debug` output of the process should show more.

(blueman-applet:8648): Gtk-WARNING **: 10:38:59.736: Theme parsing error: gtk.css:7173:46: Junk at end of value for background-color
blueman-applet 10.38.59 INFO     PluginManager:85 load_plugin: ['NMPANSupport', 'SerialManager', 'NetUsage', 'DhcpClient', 'DiscvManager', 'StandardItems', 'Battery', 'GameControllerWakelock', 'Networking', 'ExitItem', 'StatusIcon', 'DisconnectItems', 'AppIndicator', 'RecentConns', 'Menu', 'NMDUNSupport', 'KillSwitch', 'DBusService', 'AutoConnect', 'ConnectionNotifier', 'PPPSupport', 'PowerManager', 'TransferService', 'ShowConnected', 'AuthAgent']
blueman-applet 10.38.59 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 10.38.59 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 10.38.59 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Address': '50:5B:C2:FA:61:94', 'AddressType': 'public', 'Name': 'LinuxPC', 'Alias': 'LinuxPC', 'Class': 1048576, 'Powered': True, 'Discoverable': False, 'DiscoverableTimeout': 60, 'Pairable': True, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['0000110e-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '00001133-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '00001104-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00005005-0000-1000-8000-0002ee000001'], 'Modalias': 'usb:v1D6Bp0246d053D', 'Roles': ['central', 'peripheral']}
blueman-applet 10.38.59 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 10.38.59 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'>
blueman-applet 10.38.59 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 10.38.59 WARNING  PluginManager:147 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-applet 10.38.59 WARNING  PluginManager:147 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 10.38.59 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 10.38.59 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 10.38.59 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 10.38.59 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Battery.Battery'>
blueman-applet 10.38.59 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'>
blueman-applet 10.38.59 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 10.38.59 INFO     Networking:35 load_nap_settings: Loading NAP settings
blueman-applet 10.39.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 10.39.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 10.39.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.AppIndicator.AppIndicator'>
blueman-applet 10.39.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 10.39.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
blueman-applet 10.39.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 10.39.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 10.39.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier'>
blueman-applet 10.39.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 10.39.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 10.39.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 10.39.00 INFO     KillSwitch:106 io_event  : killswitch registered 1
blueman-applet 10.39.00 INFO     KillSwitch:122 io_event  : State: True
blueman-applet 10.39.00 INFO     PowerManager:178 update_power_state: off False | foff False | on True | current state True | new state True
blueman-applet 10.39.00 INFO     KillSwitch:106 io_event  : killswitch registered 2
blueman-applet 10.39.00 INFO     KillSwitch:122 io_event  : State: True
[... cut ...]
blueman-applet 10.39.00 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.DBusService.DBusService object at 0x7f2e24661f70>
blueman-applet 10.39.00 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <StatusIcon.StatusIcon object at 0x7f2e21b3dd00 (blueman+plugins+applet+StatusIcon+StatusIcon at 0x1e2d440)>
blueman-applet 10.39.00 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.Menu.Menu object at 0x7f2e21b86be0>

closed 2021-11-11 by Bishal-Mahanta