blueman-project/blueman #1636: Devices option not working on blueman-applet

snprajwal, 2021-11-25:
blueman: 2.2.3 -1
BlueZ: 5.62-1
Distribution: Arch Linux 5.15.4-arch1-1
Desktop environment: i3 4.20.1

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

The Devices option in the dropdown menu on `blueman-applet` does nothing. I need to manually launch the window with `blueman-manager`.

cschramm, 2021-11-26:
Immediately sounds like https://github.com/blueman-project/blueman/issues/1598#issuecomment-922962739 to me.

snprajwal, 2021-11-26:
My DISPLAY was set, but running `dbus-update-activation-environment DISPLAY` fixed it. Any idea as to why this happens and if there's a fix for it?

cschramm, 2021-11-26:
You can see various cases, solutions and conclusions in the tickets referenced there. It's always about "setup-heavy" environments like i3, Sway or bspwm and the conclusions sound to me like "I missed something when setting it up". The problem is simply that dbus-daemon and / or systemd need to know the desktop session and that's not properly handled in those cases.

I don't really see why that happens. systemd ships [`/etc/X11/xinit/xinitrc.d/50-systemd-user.sh`](https://github.com/systemd/systemd/blob/main/xorg/50-systemd-user.sh), so it's standard configuration for X11, completely independent from any desktop setup. Desktop managers load those xinitrc scripts (at least LightDM in my case) and `xinit` / `startx` load them as well. One thing that might go wrong in those cases is that people write custom `xinitrc` files and miss including `/etc/X11/xinit/xinitrc.d` in them. Sway actually leaves it to user configuration but the Arch Linux package ships a `/etc/sway/config.d/50-systemd-user.conf` counterpart, so in case of Arch Linux the maintainers jump in for the user and it seems like there's not much room to do anything wrong.

closed 2021-11-26 by cschramm