blueman-project/blueman #1639: my device pairs, connect and disconnect constantly

RECKONNE, 2021-12-03:
blueman: Hey how are you guys? I have a problem, my device pairs, connect and disconnect all the time is a loop
BlueZ:
Distribution:
Desktop environment:Kali

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2021-12-04:
If the device is something specific like a HID, I'd check `dmesg`. In any case I'd proceed with the logs of `bluetooth.service` and try again with debugging enabled (see the troubleshooting page you mentioned) if there's nothing interesting.

voidshine, 2021-12-09:
This is a problem with blueman. It handles the happy paths of "accept" and "always" but the "deny" option is lacking serious implementation. Deny just errors out and forgets the user's intention to NOT connect to the device. Paired with a device implementation that repeatedly requests authorization, this creates an endless stream of requests and popups that won't take no for an answer. You have to Remove the device or turn off bluetooth altogether.

Being forced to remove/unpair is a drag because when you do finally want to connect to the device, you have to go through the full pairing process again with physical button press. Would be really nice if blueman gave the user more direct control to NOT auto-connect some aggressive devices. Here's a relevant bit of code in blueman/main/applet/BluezAgent.py :

```
    def _on_authorize_service(self, device: str, uuid: str, ok: Callable[[], None],
                              err: Callable[[BluezErrorRejected], None]) -> None:
        def on_auth_action(action: str) -> None:
            logging.info(action)

            if action == "always":
                Device(obj_path=device).set("Trusted", True)
            if action == "always" or action == "accept":
                ok()
            else:
                err(BluezErrorRejected("Rejected"))
```

More is needed in the `else` clause. Respect the user's input and don't let the device keep hammering the system with popups. Ideally, device pairing configurations could be marked as "tentative" to indicate that the user wants to stay paired but only connect when initiated by the user. The flag could then be set when authorization is denied, and the state cycling problem would be solved.

Cheers and thanks for a cool project! I particularly like how the Help|Report a Problem menu comes straight here.

cschramm, 2021-12-09:
Interesting case. So you have a device that keeps connecting, no matter how often you / your system rejects it and just turning the device off is no solution to you. What kind of device is that?

I'm wondering if blueman is the right layer to handle such a thing. It can of course keep a state that you want to reject connections from a specific or just any device and then auto-deny any authorization, however, that would still mean authorization requests and denials flying around between bluetoothd and blueman and I wonder with what frequency. Your description sounds like the device immediately starts another connection attempt after rejection, so the system would be rather busy with doing nothing. :thinking:

infirit, 2021-12-10:
We could allow users to `Block` devices. Never tried it but should be easy to add a button in the DeviceMenu for.

voidshine, 2021-12-10:
Yes, some devices aggressively retry immediately every time they're denied. [Esinkin Wireless Audio Receiver for Music Streaming Sound System](https://www.amazon.com/gp/product/B07FZ3G7SC/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) is one example that makes reproducing the problem easy. This thing even requests again if the first is passively ignored, so blueman stacks many popups if left alone with the device paired.

The `Block` idea sounds fine to me, sure beats the current behavior! We shouldn't get a hidden deny/retry/deny/retry loop if the system just doesn't talk to the device as if it's paired and ready to connect..... until we unblock or actively choose to connect. Then the host/laptop starts talking to the speaker and connection proceeds as it really, really wants to do whenever it has the chance. :)

cschramm, 2021-12-10:
> This thing even requests again if the first is passively ignored, so blueman stacks many popups if left alone with the device paired.

Ouch. We could certainly improve on that. I'd need to have a look at the bluetoothd logic (I'm actually surprised that it allows multiple pending requests for the same service), but I guess cancelling a request for a service for which we already have a pending one should be a solution there (it might cancel the pending request as well, though).

So, what's the frequency of those popups? And do you get even more if you reject some?

> We shouldn't get a hidden deny/retry/deny/retry loop if the system just doesn't talk to the device

The problem is blueman is far from "the system". Best we could do from the blueman perspective in terms of blocking is to auto-deny requests to not disturb the user. However, I just realized that BlueZ does offer an API to block devices. You can try to `block` and `unblock` with `bluetoothctl` to see if it has the desired effect. All we'd need to do is offer a UI for the device property then.

infirit, 2021-12-10:
We are dealing with 2 different issues here. The @RECKONNE with connecting and reconnecting devices (which is likely solved be removing and re-pairing/connecting). 

And @voidshine which is talking about denying profiles authorisations. Denying the connection of the profile probably makes the device disconnect but this is just a side effect. What you are asking for is to block a device from connecting at all.

@cschramm we might need to start using the `replaces_id` on the notification which should tell the daemon to close the previous one.

voidshine, 2021-12-10:
It may be a separate issue, I just followed the guideline of looking for similar issues before creating a new one. I found what appears to be the cause of at least one of the problems, if not both, in the Python code...but it's not so much a bug as just missing logic.

@cschramm You are right on! The `block` & `unblock` bluez API sorts this right out! I tested with `bluetoothctl` and it does just what I need. Stops the requests, leaves the device paired but doesn't talk to it, then when I unblock it allows popups and reconnect again. I can just use this for now until blueman adds UI for it!

The stacking popup period is ~40 seconds. Pressing Deny results in a new popup, but how fast it comes is inconsistent. If it's been stacking for a while, it's instant; after blocking for a moment, the device seemed less aggressive and popups came more slowly.

Should we split this off to a feature request for exposing `block` and `unblock` capability in the blueman UI? Once that functionality is hooked up, calling it to prevent popup loops might be easier.

closed 2022-02-05 by cschramm