blueman-project/blueman #1654: Microphone not working?

RobertLSnead, 2021-12-24:
blueman: 2.1.2
Distribution: applet?
Desktop environment: ubuntu 20.04

* [ X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Not sure if I am doing something wrong or if there is no support for this?
Trying to use my microphone and nothing.
Audio works great, just unable to do any voice recording.
Anyone have any ideas or suggestions on this or if you have the same problem?
-Thanks

cschramm, 2021-12-25:
I assume you're using pulseaudio for Bluetooth audio. Does the microphone show up there?

RobertLSnead, 2021-12-30:
I'm not really sure, I'm just using the standard setup for ubuntu with the blueman install, when I select the mic on the settings from ubuntu all I get is static.

cschramm, 2021-12-30:
Check `pavucontrol`.

fredthomke, 2022-01-04:
Try this:
While your BT device is connected via Blueman, right-click the device in the Blueman control screen and choose Audio Profile > "Headset Head Unit (HSP/HFP)".  
The mic won't work unless you are in "Headset" mode as opposed to High Fidelity Playback (A2DP).  At least that's how it is on my two LM20 PCs.  The playback audio quality suffers a bit in Headset mode as compared with High Fidelity, though.  Like AM vs FM radio quality.

closed 2022-03-14 by github-actions[bot]