blueman-project/blueman #1655: blueman-sendto (send a note) fails and returns with "Unable to find service record"

gregg00, 2021-12-26:
blueman: 2.1.4-1+b2
BlueZ: I don't think i have bluez installed
Distribution: raspian os
Desktop environment: LXDE-pi

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

I am able to connect and pair to the device, but when i try to send a note, i get the error "Unable to find service record".  I see the same "unable to find service record" message when i run `blueman-applet --loglevel debug` and `cat /var/log/syslog`

cschramm, 2022-01-04:
Does blueman-manager offer the option to send a file in the device's right-click menu? Please provide the device info that you can find there.

gregg00, 2022-01-18:
I can only "send a note" to my pinetime, with blueman-manager.
I'm traveling at the moment, but when I get home I plan to try to directly connect to the pinetime rather than trying to connect over bluetooth.  As far as I understand, obex is a necessary part of the bluetooth stack if you want to transfer data over bluetooth, and it looks like my pinetime is missing obex.

cschramm, 2022-01-18:
The options to send a file should be disabled if the device does not advertise an OBEX Object Push service (ID 0x1105).

closed 2022-03-20 by gregg00