blueman-project/blueman #1664: Enhancement: 'Bluetooth Troubleshooter' a.k.a. reset bluetooth (not just restart service)

prajjwald, 2022-01-09:
blueman: 2.2.2-1
BlueZ: 5.60-0ubuntu2.1
Distribution: Ubuntu
Desktop environment: Gnome 4.x

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

TL;DR: **Add a 'troubleshoot bluetooth' menu option to blueman to allow for force resetting bluetooth beyond just turning bluetooth on/off or restarting the bluetooth service.**

### Details (my current script is in [prajjwald/bluetooth_resetter](https://github.com/prajjwald/bluetooth_resetter) ):

 - I occasionally run into issues where my bluetooth adapter is stuck in a bad state.  The only solution I had before was to restart my computer so things would connect again. 
- I found by experimentation that while restarting the bluetooth service (or turning bluetooth off and then on) did not help, Removing (rmmod) various bluetooth related modules, then adding them back (modprobe) was equivalent to a system restart, and works.
- The issue can be **artificially** replicated by connecting to a bluetooth device, turning the device off while connected, and then turning it back on in 30 seconds or so.  The system will be unable to reconnect unless
  - the system is restarted
  - the rmmod/modprobe procedure outlined above is followed.

I have a workaround in my script, but for better user experience, it would be nice if blueman offered a similar option (e.g. **troubleshoot bluetooth** or similar) out of the box.

cschramm, 2022-02-05:
I don't think such a driver / firmware / chipset issue is a common enough case.

I actually cannot remember stumbling across any issues that got solved only by re-loading bluetooth modules. Did you look into the cause of your issue?

prajjwald, 2022-02-23:
no, after discovering that resetting Bluetooth using the script works I haven't dug deeper.

The use case that would be somewhat common would be one where someone either goes out of Bluetooth range of the computer or they turn the Bluetooth device off while still connected.  In both of these cases - coming back into range or turning the Bluetooth device back on will not allow for re-connection - I was able to reliably replicate this behavior.

This behavior is reminiscent of early Linux USB device behavior - ejecting a USB device without unmounting would result in the mount path being non-responsive for a long time.

cschramm, 2022-02-23:
I cannot remember any case where `rmmod && modprobe` was required to reconnect to devices, although I'd expect that to help with possible driver and firmware issues that lead to a stuck state. What you describe seems to be bound to your specific setup and not be a common case (of course an Ubuntu 21.10 user with the same chip (and similar packages) might easily experience the same but that's still not common to me).

infirit, 2022-02-24:
I don't think Blueman is the right place to fix broken drivers or firmware.

prajjwald, 2022-02-25:
fair enough.  I don't have sufficient exposure to other chipsets, and if this is a problem that's not common across the board, not having a 'hard reset' option is OK.

closed 2022-02-25 by prajjwald