blueman-project/blueman #1665: NAP nor working for me

infirit, 2022-01-10:
I have been looking at using an existing bridge to use for the bnep device. The problem I am having that NAP is just not working with the standard setup we have with dnsmasq and iptables rules. I checked everything, manually set it up and before I go chasing my tail is it working for at all for you @cschramm? If so any ideas what might be going on?

ps, everything looks fine, rules, bridge. I see bnep0 being added to the bridge, states change properly, pan1 has the ip and netmask I set etc.

cschramm, 2022-01-11:
Works fine for me (checked with 72bebc6f and 7ed1b6b1). Checklist:

- [ ] Default route set to NAP (`ip route list default`)
- [ ] NAP `ping`able
- [ ] Internet `ping`able
- [ ] Hostnames resolvable

infirit, 2022-01-11:
Cool. thanks for checking :+1:

closed 2022-03-20 by github-actions[bot]