blueman-project/blueman #1666: UnicodeDecodeError: 'utf-8' codec can't decode byte 0xa0 in position 0: invalid start byte

mustaqimM, 2022-01-13:
blueman: `2.2.3`
BlueZ: `5.62`
Distribution: `Void Linux`
Desktop environment: `swaywm`

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

[blueman-applet log](https://gist.github.com/mustaqimM/16197ac65457530ac2566e52fd6e4041)
---
Hello, if I connect my bluetooth headphones on first try, it connects then disconnects. If I try again it connects without a problem. The relevant line when it disconnects:

`UnicodeDecodeError: 'utf-8' codec can't decode byte 0xa0 in position 0: invalid start byte`

This didn't occur before but I don't know when it first started

cschramm, 2022-02-05:
Hi, what context is that line from? :thinking:

cschramm, 2022-02-05:
The UnicodeDecodeError is a harmless side effect (#1241).

From the blueman log it seems like bluetoothd is crashing shortly after connecting to the device. Its logs should tell you more.

closed 2022-04-14 by github-actions[bot]