blueman-project/blueman #1668: When Bluetooth connects, Wifi Disconnects automatically

Sweetstud, 2022-01-16:
blueman: 2.1.4
BlueZ:
Distribution: Mx Linux 21
Desktop environment:XFCE

Good day. My problems is when i use my Dell Inspirion  N7110 and i connect to my bluetooth headset, the wifi disconnects. Is there any way around this ?

cschramm, 2022-01-17:
Sounds like a firmware issue with some combo device. `dmesg` might shed some light. This is clearly out of scope of blueman and also even BlueZ.

Sweetstud, 2022-01-18:
How do I access dmesg?

cschramm, 2022-01-19:
`sudo dmest`

Sweetstud, 2022-01-21:
I don't know if this is any help?

$ sudo dmesg
[    0.000000] microcode: microcode updated early to revision 0x2f, date = 2019-02-17
[    0.000000] Linux version 5.10.0-11-amd64 (debian-kernel@lists.debian.org) (gcc-10 (Debian 10.2.1-6) 10.2.1 20210110, GNU ld (GNU Binutils for Debian) 2.35.2) #1 SMP Debian 5.10.92-1 (2022-01-18)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.10.0-11-amd64 root=UUID=cee3bd8c-4248-43ae-a213-3ef826591e77 ro quiet splash
[    0.000000] Disabled fast string operations
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009d7ff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009d800-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000bac8efff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bac8f000-0x00000000badd3fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000badd4000-0x00000000bade0fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000bade1000-0x00000000bae0bfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000bae0c000-0x00000000bae0dfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000bae0e000-0x00000000bae0ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000bae10000-0x00000000bae12fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000bae13000-0x00000000baf17fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000baf18000-0x00000000baf1bfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000baf1c000-0x00000000baf1efff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000baf1f000-0x00000000baf9efff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000baf9f000-0x00000000baffefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000bafff000-0x00000000baffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bb000000-0x00000000bf9fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed08000-0x00000000fed08fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed10000-0x00000000fed19fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffd80000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000023e9fefff] usable
[    0.000000] BIOS-e820: [mem 0x000000023e9ff000-0x000000023fdfffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.6 present.
[    0.000000] DMI: Dell Inc.          Dell System Inspiron N7110/0YH79Y, BIOS A13 09/05/2012
[    0.000000] tsc: Fast TSC calibration using PIT
[    0.000000] tsc: Detected 2292.505 MHz processor
[    0.000787] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000790] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000800] last_pfn = 0x23e9ff max_arch_pfn = 0x400000000
[    0.000805] MTRR default type: uncachable
[    0.000806] MTRR fixed ranges enabled:
[    0.000807]   00000-9FFFF write-back
[    0.000808]   A0000-BFFFF uncachable
[    0.000810]   C0000-FFFFF write-protect
[    0.000811] MTRR variable ranges enabled:
[    0.000812]   0 base 0FFC00000 mask FFFC00000 write-protect
[    0.000814]   1 base 000000000 mask F80000000 write-back
[    0.000815]   2 base 080000000 mask FC0000000 write-back
[    0.000816]   3 base 0BC000000 mask FFC000000 uncachable
[    0.000817]   4 base 0BB000000 mask FFF000000 uncachable
[    0.000818]   5 base 100000000 mask F00000000 write-back
[    0.000819]   6 base 200000000 mask FC0000000 write-back
[    0.000820]   7 base 23FE00000 mask FFFE00000 uncachable
[    0.000821]   8 disabled
[    0.000822]   9 disabled
[    0.001604] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.001856] last_pfn = 0xbb000 max_arch_pfn = 0x400000000
[    0.013786] reserving inaccessible SNB gfx pages
[    0.014398] RAMDISK: [mem 0x2e911000-0x3347ffff]
[    0.014403] ACPI: Early table checksum verification disabled
[    0.014408] ACPI: RSDP 0x00000000000F00E0 000024 (v02 DELL  )
[    0.014413] ACPI: XSDT 0x00000000BAFFE120 00008C (v01 DELL   QA09     00000002 LOHR 00000002)
[... 772 lines cut ...]
[   19.415991] Bluetooth: BNEP socket layer initialized
[   19.437975] NET: Registered protocol family 38
[   19.535230] r8169 0000:03:00.0: firmware: direct-loading firmware rtl_nic/rtl8105e-1.fw
[   19.535926] RTL8208 Fast Ethernet r8169-0-300:00: attached PHY driver [RTL8208 Fast Ethernet] (mii_bus:phy_addr=r8169-0-300:00, irq=IGNORE)
[   19.823913] r8169 0000:03:00.0 eth0: Link is Down
[   19.838119] iwlwifi 0000:01:00.0: Radio type=0x2-0x2-0x1
[   19.938671] iwlwifi 0000:01:00.0: Radio type=0x2-0x2-0x1
[   20.027245] iwlwifi 0000:01:00.0: Radio type=0x2-0x2-0x1
[   20.132720] iwlwifi 0000:01:00.0: Radio type=0x2-0x2-0x1
[... cut ...]
[   21.745170] wlan0: send auth to 68:ff:7b:6d:59:f9 (try 1/3)
[   21.747176] wlan0: authenticated
[   21.749784] wlan0: associate with 68:ff:7b:6d:59:f9 (try 1/3)
[   21.771414] wlan0: RX AssocResp from 68:ff:7b:6d:59:f9 (capab=0xc11 status=0 aid=5)
[   21.777836] wlan0: associated
[   21.903514] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   22.314435] Bluetooth: RFCOMM TTY layer initialized
[   22.314442] Bluetooth: RFCOMM socket layer initialized
[   22.314449] Bluetooth: RFCOMM ver 1.11
WesleyBurris@WesleyBurris:~
$

cschramm, 2022-01-22:
Does not look to me like the disconnect happened there.

Sweetstud, 2022-01-23:
Hmmm ok thanks

closed 2022-04-02 by github-actions[bot]