blueman-project/blueman #1673: Bluetooth phone disconnecting all the time

Alvarocda, 2022-02-08:
blueman: 2.1.2 
BlueZ:
Distribution: Linux Mint 20.2 Uma

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

My bluetooth headset automatically disconnects all the time, usually, I can listen to 1 or 2 songs and it disconnects, all the time I need to access the blueman, forget my phone because it doesn't connect again and pair it again.
This is very annoying, I think during the day I need to do this process at least 50 times.

I tested it on a PC with Ubuntu and another with Mint and the problem happens with both.
The problem is not the phone or the adapter because in Windows it works perfectly without failures and on my cell phone too.

Gist with blueman debug log during the problem
https://gist.github.com/Alvarocda/be57496d42d01f0c2ede639309fa6674

My syslog
https://gist.github.com/Alvarocda/84cd959967f6f1f39ec3cb3fcd95f189

Alvarocda, 2022-02-09:
Here is the syslog with bluetooth -d param enabled.
https://gist.github.com/Alvarocda/b35211a9856207f95596d2e8c08b644e

At: 14:13:10 i connected to my headset

At 14:55:54 it disconnected

after he lost the connection, I accessed the blueman and tried to reconnect to the headset but when trying to reconnect it gave an error: **Resource temporarily unavailable**
Log of me trying to reconnect:
https://gist.github.com/Alvarocda/c8a173f701656455f11f4be9ac73cbcf

Unsuccessfully, I clicked the remove option from blueman and connected to the headset again and then it started working again
https://gist.github.com/Alvarocda/b85c04b3cdebfcbce83fc5ee12eb99f7

cschramm, 2022-02-09:
Reason 1 is `MGMT_DEV_DISCONN_TIMEOUT`, so it sounds like the device goes dark. You might be able to verify by checking `btmon`. It probably does not go physically dark, as re-adding it is enough to be able to re-connect. The reason why re-adding the device helps with reconnecting might be that BlueZ still holds a key (state typically in `/var/lib/bluetooth`) that the device dropped for some reason.

I can't provide much help here. The guys over at BlueZ might be more helpful. I would also try to test a newer / alternative driver / firmware as far as possible.

Alvarocda, 2022-02-09:
> Reason 1 is `MGMT_DEV_DISCONN_TIMEOUT`, so it sounds like the device goes dark. You might be able to verify by checking `btmon`. It probably does not go physically dark, as re-adding it is enough to [...]

January was the only month that it worked perfectly.
I would turn on the headset and the bluetooth would automatically connect with my headset, it wouldn't disconnect once and I didn't have any problems with pitch change.
Probably some update fixed the problem in December and now in February the problem has returned.

About the firmware, I saw on a forum that the latest firmware available is broken and they recommended I go back to an old firmware, but I'm not such an advanced linux user.
I'll try to go back to that firmware version and test.

https://bbs.archlinux.org/viewtopic.php?id=263806

> linux-firmware-20211216.f682ecb-1 -> intel/ibt-20-1-3.sfi Firmware revision 0.3 build 125 week 46 2021 (working)
> linux-firmware-20220119.0c6a7b3-2 -> intel/ibt-20-1-3.sfi Firmware revision 0.3 build 235 week 51 2021 (broken)

cschramm, 2022-02-09:
Your firmware should be much older. With Linux Mint 20 you're locked at Ubuntu 20.04, i.e. about 2 years in the past and far from 20211216 or anything. You should be able to follow past updates in `/var/log/apt`, but I doubt that there were significant ones (usually just security fixes).

Alvarocda, 2022-02-15:
I bought a new bluetooth adapter with RTL8761b chipset, this one has bluetooth 5.0, the old one was only 4.0.
I had a little trouble getting it to work because Linux didn't detect it automatically.

But after I managed to install it, I had no problems with disconnections.
I've been using my headphones for about 6 hours and so far, no problems.

But I would really like the bluetooth to connect by itself when I turn on my headphones, unfortunately it still isn't doing that.
I have to access blueman, forget about the device and make a new connection.

closed 2022-02-18 by Alvarocda