blueman-project/blueman #1674: Bluetooth will not turn on ......

dbt1957, 2022-02-09:
** Had a look through the various bt issues could not locate anything similar 

Fresh dualboot 20.04lts / win010 install on E6520 i7 win10 64-Bit

Bluetooth will not turn on

Moving the settings slider to on, still reflects turned off there

bt control in the upper right pull down menu shows " turn bt off" regardless of slider position 

pls ref attached ss 

Still a bit of a nub with ubuntu however nothing like this with several prior win10/20.04 dual boot installs

Any wisdom greatly appreciated

[image]

Regards

dbt

dbt1957, 2022-02-09:
Please forgive my ignorance 

Any suggestions as to where  a bit of a nub with ubuntu  might find help with this issue ?

dbt1957, 2022-02-09:
The tab on my bluetooth reflect https://github.com/blueman-project/blueman ( I assume for assitance)

infirit, 2022-02-09:
I'd suggest to ask on the Ubuntu forums or other support channel Ubuntu has these days.

closed 2022-02-09 by infirit