blueman-project/blueman #1681: Is there a way to bind a profile to a keyboard combination?

kendoori, 2022-02-22:
blueman: 2.1.2
BlueZ: 5.53
Distribution: Mint 20.3
Desktop environment: Cinnamon

Is there a way to bind a profile to a keyboard combination? For example, this profile is already set up. How to I reconnect it without using the tray icon?

```
`70:66:1B:C1:AE:53
public
Kenny's R-KS hearing aid
Kenny's R-KS hearing aid
0x240404
0x0000
audio-card
yes
yes
no
no
yes
0000110b-0000-1000-8000-00805f9b34fb Audio Sink
0000110c-0000-1000-8000-00805f9b34fb Remote Control Target
0000110d-0000-1000-8000-00805f9b34fb Advanced Audio
0000110e-0000-1000-8000-00805f9b34fb Remote Control
0000111e-0000-1000-8000-00805f9b34fb Handsfree
00001200-0000-1000-8000-00805f9b34fb PnP Information
bluetooth:v0282p05A6d9340
/org/bluez/hci0`
```

cschramm, 2022-02-22:
I'm not sure what you mean with profile. If it is about a connection to an audio device `bluetoothctl connect 70:66:1B:C1:AE:53` should do fine.

kendoori, 2022-02-22:
That's perfect!