blueman-project/blueman #1682: Bluetooth connects and disconnects immediately

agustinmenegat, 2022-02-22:
I search for my bluetooth speaker and it finds it correctly, when I connect it it tells me "connected successfully" and instantly disconnects it. I have tried to do it from "Bluetooth devices" and from the terminal. But both give me the same error.

z1atk0, 2022-02-24:
Exact same problem here, but with an old Nokia 2600 Classic. The phone connects, and after 5 seconds it disconnects again. Connecting/mounting and transferring files via `obexfs` manually works fine. Connecting an EPOS H3 Hybrid headset via `blueman-applet` also works fine.

Here's the debug log, the device info from `blueman-applet` and a screenshot with a rather vague error message:
```
[zlatko@disclosure:~]$ blueman-applet --loglevel debug
blueman-applet 15.10.00 INFO     PluginManager:85 load_plugin: ['DisconnectItems', 'KillSwitch', 'DBusService', 'Battery', 'GameControllerWakelock', 'PowerManager', 'StatusIcon', 'SerialManager', 'DiscvManager', 'PPPSupport', 'NMPANSupport', 'AppIndicator', 'DhcpClient', 'NMDUNSupport', 'RecentConns', 'Networking', 'ExitItem', 'TransferService', 'NetUsage', 'Menu', 'ConnectionNotifier', 'StandardItems', 'AutoConnect', 'AuthAgent', 'ShowConnected']
blueman-applet 15.10.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 15.10.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 15.10.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 15.10.00 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Address': '00:1A:7D:DA:71:08', 'AddressType': 'public', 'Name': 'BlueZ 5.60', 'Alias': 'BlueZ 5.60', 'Class': 8126724, 'Powered': True, 'Discoverable': True, 'DiscoverableTimeout': 0, 'Pairable': False, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['00001133-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00001104-0000-1000-8000-00805f9b34fb', '00001108-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '0000111f-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d053C', 'Roles': ['central', 'peripheral']}
blueman-applet 15.10.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 15.10.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 15.10.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 15.10.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Battery.Battery'>
blueman-applet 15.10.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'>
blueman-applet 15.10.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 15.10.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 15.10.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.PPPSupport.PPPSupport'>
blueman-applet 15.10.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DhcpClient.DhcpClient'>
blueman-applet 15.10.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 15.10.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 15.10.00 INFO     Networking:35 load_nap_settings: Loading NAP settings
blueman-applet 15.10.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 15.10.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 15.10.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier'>
blueman-applet 15.10.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 15.10.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 15.10.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 15.10.00 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 15.10.00 INFO     KillSwitch:106 io_event  : killswitch registered 0
blueman-applet 15.10.00 INFO     KillSwitch:122 io_event  : State: True
blueman-applet 15.10.00 INFO     PowerManager:178 update_power_state: off False | foff False | on True | current state True | new state True
blueman-applet 15.10.00 INFO     KillSwitch:85 _on_connman_vanished: net.connman vanished
blueman-applet 15.10.00 INFO     TransferService:234 _on_dbus_name_appeared: org.bluez.obex :1.149
blueman-applet 15.10.00 INFO     Applet:54 _on_dbus_name_appeared: org.bluez :1.459
[... cut ...]
0000112d-0000-1000-8000-00805f9b34fb SIM Access (SAP)
00001200-0000-1000-8000-00805f9b34fb PnP Information
00005001-0000-1000-8000-0002ee000001 Proprietary
00005002-0000-1000-8000-0002ee000001 Proprietary
00005003-0000-1000-8000-0002ee000001 Proprietary
00005557-0000-1000-8000-0002ee000001 Proprietary
00005601-0000-1000-8000-0002ee000001 Proprietary
bluetooth:v0001p0050d0000
/org/bluez/hci0
```
Screenshot:
[image]
`blueman-2.2.4`, `bluez-5.60`, `linux-5.15.25` on Slackware 14.2 32bit. Please let me know if you need any more information.

cschramm, 2022-02-27:
Did you check the bluetooth.service logs? I'd expect some hint there, probably even without debugging output.

z1atk0, 2022-02-28:
That's all from `grep blue /var/log/messages` at the relevant time - just DBus, nothing from `bluetoothd` itself:
```
Feb 24 15:10:00 disclosure dbus-daemon[23565]: [system] Activating service name='org.blueman.Mechanism' requested by ':1.466' (uid=1000 pid=13216 comm="/usr/bin/python3 /usr/local/bin/blueman-applet --l") (using servicehelper)
Feb 24 15:10:00 disclosure dbus-daemon[23565]: [system] Successfully activated service 'org.blueman.Mechanism'
Feb 24 15:10:11 disclosure dbus-daemon[23975]: [session uid=1000 pid=23970] Activating service name='org.blueman.Manager' requested by ':1.813583' (uid=1000 pid=13216 comm="/usr/bin/python3 /usr/local/bin/blueman-applet --l")
Feb 24 15:10:11 disclosure dbus-daemon[23975]: [session uid=1000 pid=23970] Successfully activated service 'org.blueman.Manager'
```
There's no `systemd` on Slackware, so there's no `bluetooth.service` either. `bluetoothd` is started via normal init scripts. I can try running it with the `-d` option and see if something pops up.

z1atk0, 2022-02-28:
No, running with `-d` doesn't make a difference. Just the usual startup messages, but nothing during the actual connect/disconnect:
```
Feb 28 06:42:03 disclosure bluetoothd[14110]: Bluetooth daemon 5.60
Feb 28 06:42:03 disclosure bluetoothd[14110]: Starting SDP server
Feb 28 06:42:03 disclosure bluetoothd[14110]: Bluetooth management interface 1.21 initialized
Feb 28 06:42:03 disclosure bluetoothd[14110]: Endpoint registered: sender=:1.75 path=/MediaEndpoint/A2DPSink/sbc
Feb 28 06:42:03 disclosure bluetoothd[14110]: Endpoint registered: sender=:1.75 path=/MediaEndpoint/A2DPSource/sbc
Feb 28 06:42:03 disclosure bluetoothd[14110]: Endpoint registered: sender=:1.75 path=/MediaEndpoint/A2DPSink/sbc_xq_453
Feb 28 06:42:03 disclosure bluetoothd[14110]: Endpoint registered: sender=:1.75 path=/MediaEndpoint/A2DPSource/sbc_xq_453
Feb 28 06:42:03 disclosure bluetoothd[14110]: Endpoint registered: sender=:1.75 path=/MediaEndpoint/A2DPSink/sbc_xq_512
Feb 28 06:42:03 disclosure bluetoothd[14110]: Endpoint registered: sender=:1.75 path=/MediaEndpoint/A2DPSource/sbc_xq_512
Feb 28 06:42:03 disclosure bluetoothd[14110]: Endpoint registered: sender=:1.75 path=/MediaEndpoint/A2DPSink/sbc_xq_552
Feb 28 06:42:03 disclosure bluetoothd[14110]: Endpoint registered: sender=:1.75 path=/MediaEndpoint/A2DPSource/sbc_xq_552
```

cschramm, 2022-03-09:
Disconnect reason 2 is `MGMT_DEV_DISCONN_LOCAL_HOST`, so it looks like something is causing the disconnect. Does pulseaudio give you anything?

z1atk0, 2022-03-09:
I just checked again, and there were indeed a few hints of `pulseaudio` and `NetworkManager` around that time:
```
Feb 28 06:40:54 disclosure dbus-daemon[23975]: [session uid=1000 pid=23970] Activating service name='org.blueman.Manager' requested by ':1.822135' (uid=1000 pid=4313 comm="/usr/bin/python3 /usr/local/bin/blueman-applet ")
Feb 28 06:40:55 disclosure dbus-daemon[23975]: [session uid=1000 pid=23970] Successfully activated service 'org.blueman.Manager'
Feb 28 06:42:02 disclosure NetworkManager[23639]: <info>  [1646026922.8097] device (00:21:09:61:0F:27): state change: disconnected -> unmanaged (reason 'removed', sys-iface-state: 'removed')
Feb 28 06:42:02 disclosure dbus-daemon[23565]: [system] Rejected send message, 4 matched rules; type="error", sender=":1.75" (uid=1000 pid=24088 comm="/usr/local/bin/pulseaudio --start --log-target=sys") interface="(unset)" member="(unset)" error name="org.freedesktop.DBus.Error.UnknownMethod" requested_reply="0" destination=":1.459" (uid=0 pid=3728 comm="/usr/sbin/bluetoothd ")
Feb 28 06:42:02 disclosure dbus-daemon[23565]: [system] Rejected send message, 4 matched rules; type="error", sender=":1.75" (uid=1000 pid=24088 comm="/usr/local/bin/pulseaudio --start --log-target=sys") interface="(unset)" member="(unset)" error name="org.freedesktop.DBus.Error.UnknownMethod" requested_reply="0" destination=":1.459" (uid=0 pid=3728 comm="/usr/sbin/bluetoothd ")
Feb 28 06:42:04 disclosure NetworkManager[23639]: <info>  [1646026924.0291] manager: (00:21:09:61:0F:27): new Bluetooth device (/org/freedesktop/NetworkManager/Devices/7)
Feb 28 06:42:04 disclosure NetworkManager[23639]: <info>  [1646026924.0332] device (00:21:09:61:0F:27): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
Feb 28 06:42:04 disclosure NetworkManager[23639]: <info>  [1646026924.0347] device (00:21:09:61:0F:27): state change: unavailable -> disconnected (reason 'none', sys-iface-state: 'managed')
Feb 28 06:42:04 disclosure dbus-daemon[23565]: [system] Activating service name='org.blueman.Mechanism' requested by ':1.474' (uid=1000 pid=4313 comm="/usr/bin/python3 /usr/local/bin/blueman-applet ") (using servicehelper)
Feb 28 06:42:04 disclosure dbus-daemon[23565]: [system] Successfully activated service 'org.blueman.Mechanism'
Feb 28 06:42:47 disclosure dbus-daemon[23975]: [session uid=1000 pid=23970] Activating service name='org.blueman.Manager' requested by ':1.822135' (uid=1000 pid=4313 comm="/usr/bin/python3 /usr/local/bin/blueman-applet ")
Feb 28 06:42:48 disclosure dbus-daemon[23975]: [session uid=1000 pid=23970] Successfully activated service 'org.blueman.Manager'
Feb 28 06:46:03 disclosure dbus-daemon[23975]: [session uid=1000 pid=23970] Activating service name='org.blueman.Manager' requested by ':1.822135' (uid=1000 pid=4313 comm="/usr/bin/python3 /usr/local/bin/blueman-applet ")
Feb 28 06:46:04 disclosure dbus-daemon[23975]: [session uid=1000 pid=23970] Successfully activated service 'org.blueman.Manager'
```
Does any of this explain the disconnect? :thinking: I certainly wouldn't know. :man_shrugging: :wink: Thanks for taking the time to go through all these logs! :+1:

cschramm, 2022-03-10:
Not really. The guys over at BlueZ or pulseaudio could be more helpful here.

closed 2022-05-17 by github-actions[bot]