blueman-project/blueman #1684: bluetouth doesnt see my keyboard but see everything else

damnikindadontcare, 2022-03-02:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

freakluis, 2022-03-06:
I'm right where you are. I've tried all of my bt keyboards: newer or older. Nothing seems to work. Could it be a matter of the kernel rather than a problem with this package?

cschramm, 2022-03-10:
Check `bluetoothctl` to see if your system knows the device in general, e.g. `bluetoothctl devices` for already discovered devices or `bluetoothctl scan on` for discovery. If it does not, check with another system if the device is visible at all. `btmon` could also help to check the physical layer but might be overwhelming of course.

kybernetik42, 2022-04-04:
This is at my Laptop Linuxmint:
[Linux:~$ sudo bluetoothctl
[sudo] Passwort für harald:             
Agent registered
[M720 Triathlon]# agent KeyboardOnly
Agent is already registered
[M720 Triathlon]# default-agent
Default agent request successful
[M720 Triathlon]# pairable on
Changing pairable on succeeded
[M720 Triathlon]# pair 34:88:5D:6F:D0:EE
Device 34:88:5D:6F:D0:EE not available
[NEW] Device 34:88:5D:6F:D0:EE Keyboard K380
[CHG] Device 34:88:5D:6F:D0:EE Connected: no
[DEL] Device 34:88:5D:6F:D0:EE Keyboard K380
[M720 Triathlon]# pair 34:88:5D:6F:D0:EE
Device 34:88:5D:6F:D0:EE not available
[M720 Triathlon]# agent KeyboardOnly
Agent is already registered
[M720 Triathlon]# agent KeyboardOnly
Agent is already registered
[M720 Triathlon]# pair 34:88:5D:6F:D0:EE
Device 34:88:5D:6F:D0:EE not available
[M720 Triathlon]# scan on
Discovery started
[CHG] Controller 9C:FC:E8:DC:BA:71 Discovering: yes
...
[CHG] Device C0:28:8D:4E:75:C8 ManufacturerData Key: 0x0003
[CHG] Device C0:28:8D:4E:75:C8 ManufacturerData Value:
  00 8c 09 00 00 08 c4 98 80 c4 4d 4b c0 28 8d 4e  ..........MK.(.N
  75 c8 12                                         u..             
....
[M720 Triathlon]# pair 34:88:5D:6F:D0:EE
Attempting to pair with 34:88:5D:6F:D0:EE
[CHG] Device 34:88:5D:6F:D0:EE Connected: yes
[agent] Passkey: 465628
[agent] Passkey: 465628
[agent] Passkey: 465628
[agent] Passkey: 465628
[agent] Passkey: 465628
[agent] Passkey: 465628
[agent] Passkey: 465628
[CHG] Device 34:88:5D:6F:D0:EE Modalias: usb:v046DpB342d4201
[CHG] Device 34:88:5D:6F:D0:EE UUIDs: 00001000-0000-1000-8000-00805f9b34fb
[CHG] Device 34:88:5D:6F:D0:EE UUIDs: 00001124-0000-1000-8000-00805f9b34fb
[CHG] Device 34:88:5D:6F:D0:EE UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device 34:88:5D:6F:D0:EE ServicesResolved: yes
[CHG] Device 34:88:5D:6F:D0:EE Paired: yes
Pairing successful
[CHG] Device 34:88:5D:6F:D0:EE ServicesResolved: no
[CHG] Device 34:88:5D:6F:D0:EE Connected: no
[CHG] Device 74:F0:7D:E3:55:20 RSSI: -79
[CHG] Device C0:28:8D:4E:75:C8 ManufacturerData Key: 0x0003
[CHG] Device C0:28:8D:4E:75:C8 ManufacturerData Value:
  00 8c 09 00 00 08 c4 98 80 c4 4d 4b c0 28 8d 4e  ..........MK.(.N
  75 c8 12                                         u..             
[NEW] Device 6B:41:45:81:34:50 6B-41-45-81-34-50
[CHG] Device 35:12:4A:87:3C:60 RSSI: -86
[CHG] Device 6A:A9:70:B0:32:BA ManufacturerData Key: 0x004c
[CHG] Device 6A:A9:70:B0:32:BA ManufacturerData Value:
  10 06 23 1a e4 50 c4 76                          ..#..P.v        
[NEW] Device 20:19:16:47:23:08 20-19-16-47-23-08
[NEW] Device 76:12:2D:33:C6:9E 76-12-2D-33-C6-9E
[NEW] Device 5F:A1:13:EB:1B:7E 5F-A1-13-EB-1B-7E
[NEW] Device 5E:38:A4:D3:E1:4F 5E-38-A4-D3-E1-4F
[NEW] Device 03:EA:60:46:C0:10 03-EA-60-46-C0-10
[NEW] Device 42:9B:23:27:5E:66 42-9B-23-27-5E-66
[CHG] Device 20:19:16:47:23:08 Name: FM 01 Slider
[CHG] Device 20:19:16:47:23:08 Alias: FM 01 Slider
[CHG] Device 6B:AE:77:15:44:21 RSSI: -103
[... 219 lines cut ...]
[CHG] Device 03:EA:60:46:C0:10 RSSI: -99
[CHG] Device C0:28:8D:4E:75:C8 ManufacturerData Key: 0x0003
[CHG] Device C0:28:8D:4E:75:C8 ManufacturerData Value:
  00 8c 09 00 00 08 c4 98 80 c4 4d 4b 12           ..........MK.   
[CHG] Device 6F:E2:E7:6E:0F:41 ManufacturerData Key: 0x004c
[CHG] Device 6F:E2:E7:6E:0F:41 ManufacturerData Value:
  01 00 00 00 00 00 00 00 00 00 00 08 00 00 00 00  ................
  00                                               .               
[CHG] Device 4E:F3:4C:25:06:56 RSSI: -90
[CHG] Device 74:24:33:AE:35:CE ManufacturerData Key: 0x004c
[CHG] Device 74:24:33:AE:35:CE ManufacturerData Value:
  10 07 5b 1f 5a 6c cd 4b c8                       ..[.Zl.K.       
[CHG] Device C0:28:8D:4E:75:C8 ManufacturerData Key: 0x0003
[CHG] Device C0:28:8D:4E:75:C8 ManufacturerData Value:
  00 8c 09 00 00 08 c4 98 80 c4 4d 4b c0 28 8d 4e  ..........MK.(.N
  75 c8 12                                         u..             
[M720 Triathlon]# scan off
Discovery stopped
....
[M720 Triathlon]# 

It is ending with this:
[M720 Triathlon]# connect 34:88:5D:6F:D0:EE
Attempting to connect to 34:88:5D:6F:D0:EE
Failed to connect: org.bluez.Error.Failed

kybernetik42, 2022-04-04:
@cschramm: The point, why I posted it, that I was able to connect the keyboard, but after boot, it was not reconizable. I must delete the keyboard from the list, and the pair again. But pairing again was not easy, it takes several tries, before the keyboard was paired again. Indeed, it is not exactly a contribution to this topic, you are right.

freakluis, 2022-04-04:
By the way, the problem I had had with my k380 keyboard was solved with this comment by @cschramm 

> Check `bluetoothctl` to see if your system knows the device in general, e.g. `bluetoothctl devices` for already discovered devices or `bluetoothctl scan on` for discovery. If it does not, check with [...]

closed 2022-06-13 by github-actions[bot]