blueman-project/blueman #1690: Device protocol not recognized

mitchejmm, 2022-03-13:
blueman:
BlueZ:
Distribution:
Desktop environment: Peppermint os 11

My bluetooth headset was recognized and the info shows 'paired', but when I attempt to connect, it fails.

[image]

ℹ Please use some service like [Pastebin](http://pastebin.com/) or [GitHub Gist](https://gist.github.com/) to post your logs and keep the thread clean and readable. Make sure to describe the exact steps you took when producing it.

->

mitchejmm, 2022-03-14:
Update:  After a reboot, the system now recognizes the bluetooth headphones AND connects to them, but the sound continues to play on the speakers.  I've seen references to information in the controller that I've never seen before (it has always been a ubiquitous turn on - connect - use experience)  I have no idea what particular setting is inhibiting the audio from reachinf the headset  (the neophyte that I am).

[image]

cschramm, 2022-03-14:
I could not spot the failure in your screenshots or description. From the issue title one could guess that you get a `Protocol not available` error message. That means you just do not have any piece of software that handles the audio connection (blueman does not and BlueZ itself doesn't either). The standard solution on Debian-based systems is the `pulseaudio-module-bluetooth` package that plugs the device into pulseaudio as a usual audio sink (and source).

mitchejmm, 2022-03-14:
Thank you for the information.  I’ve quite a learning curve, I know, as
this is the first time I’ve installed a Linux Distro.

I’ll remove the Blueman and try installing the bluetooth module, and report
back to you my experience.

One small ‘glitch’ I’m experiencing is a ‘jumping around’ of the cursor as
I type.  Wierd.

I’m also testing out different other hardware features, like the camera and
mike with Skype.

mitchejmm, 2022-03-14:
Is there a ‘app’ way of running bluetoothctrl?

I am approaching the setup of Peppermint from a ‘user’ perspective.

[I find it a bit frustrating that this facility is not included as a
standard feature]

Using the PulseAudio tutorial “setting up Bluetooth”, I’ve  gotten to the
point where it says, “Run bluetoothctrl” and goes through this rather long
list of configuration instructions.   I did a search on ‘how do I run
bluetoothctrl’ and was directed to “Linux command line bluetoothctrl
wrapper for nodejs”.  Yikes!!

Again, I’m a linux neophyte and not a programmer; just a brave soul trying
to get away from win10, trying to connect a pair
Bluetooth earphones to Peppermint

Any help would be greatly appreciated!

Thanx!

cschramm, 2022-03-15:
Not sure what an "app way" is to you. bluetoothctl is a user app(lication) with a text-based user interface. If you prefer a graphical one, blueman is what you want. All you need to do is get pulseaudio Bluetooth support by installing the pulseaudio-module-bluetooth package. According to https://peppermintos.com/guide/software/ the "preferred" method to install software is the "Software Manager" that they took from Linux Mint. After you got it, go back to blueman and you're probably able to connect (if you still get the same error, a restart, as you know it from Windows, should help).

You might need to select the desired audio sinks and sources in pulseaudio. Its settings are most probably available via menus somehow.

mitchejmm, 2022-03-15:
Thank you!

I reinstalled Blueman and rebooted, then the system connected to the
earphones.  One last action was to select the earphones instead of the
speakers… success!

… also thanx for the tip about Synaptic.

👍👍

Regards,

Jim Mitchell

closed 2022-03-16 by cschramm