blueman-project/blueman #1691: blueman-applet not starting after building last version

avma, 2022-03-16:
blueman:2.2.4
BlueZ: 5.63
Distribution: Linux Mint 20.3
Desktop environment: Cinnamon

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

avma, 2022-03-16:
[blueman-applet --loglevel debug.txt](https://github.com/blueman-project/blueman/files/8260968/blueman-applet.--loglevel.debug.txt)

avma, 2022-03-16:
The debug output for ver 2.2.4
[blueman-applet --loglevel debug_2.2.4.txt](https://github.com/blueman-project/blueman/files/8261759/blueman-applet.--loglevel.debug_2.2.4.txt)

cschramm, 2022-03-16:
So did you try to unmask blueman-mechanism.service?

avma, 2022-03-16:
No,  how do i do that?

cschramm, 2022-03-16:
That should work, yes. According to the D-Bus error message you masked the service in systemd, i.e. in addition to disabling it (removing the symlink to the unit file) you symlinked it to `/dev/null`.

infirit, 2022-03-16:
Where is `blueman-mechanism` installed? And if you run it as root what does it output?

avma, 2022-03-16:
It is located in '/usr/local/libexec'

this is the output of the command `sudo blueman-mechanism`

`❯       sudo ./blueman-mechanism
Traceback (most recent call last):
  File "./blueman-mechanism", line 6, in <module>
    from blueman.main.MechanismApplication import MechanismApplication
ModuleNotFoundError: No module named 'blueman'`

infirit, 2022-03-17:
Basically, don't install Blueman to your home dir. Blueman-mechanism runs as root and will not be able to find the python modules it needs. Everything to `/use/local` will be better.

avma, 2022-03-17:
How do I do that? 
I was following the build instruction on the README:
`./autogen.sh` `./configure` `make` `sudo make install`

cschramm, 2022-03-17:
To get a newer version you could try any of

- https://blueman.cschramm.eu/debian/ git snapshot packages
- building a proper .deb (check https://salsa.debian.org/cschramm/blueman/-/tree/main/debian)
- hacking the installation; something like `sudo mv /usr/local/lib/python3.8/site-packages/_blueman.so /usr/local/lib/python3.8/site-packages/blueman /usr/local/lib/python3.8/dist-packages/` could already do the trick

avma, 2022-03-18:
Ok, I'll give it a try....

avma, 2022-03-18:
yep, it worked, I now have  ver 2.2.1instaled. Thank you

closed 2022-03-18 by cschramm