blueman-project/blueman #1701: Add dots on left and right side of toolbar icon when bluetooth device is connected.

linusrg1, 2022-04-12:
blueman: 2.1.2
BlueZ: I don't know how to find this sorry. 
Distribution: Mint 20.3
Desktop environment: Cinnamon

Currently the icon remains the same even if a device is connected, it would be nice if, like blueberry there were 2 dots on the left and right side of the toolbar icon to signify that a device is connected.

cschramm, 2022-04-28:
Sounds like you're referring to some symbolic icon theme. The standard tray icon changes from white on blue to green on blue when a device is connected (with the standard plugins enabled).

linusrg1, 2022-04-29:
> Sounds like you're referring to some symbolic icon theme. The standard tray icon changes from white on blue to green on blue when a device is connected (with the standard plugins enabled).

It does not appear to be doing this for me on Cinnamon. Papirus could be interfering with it but I don't think that is it. Because Blueberry was able to switch its icon state just fine with papirus applied, and seeing how blueman has been around for a while and is quite popular, I dont see why the papirus team wouldnt of added support for it if they needed to.

cschramm, 2022-04-29:
Well, take a look at https://github.com/PapirusDevelopmentTeam/papirus-icon-theme/tree/master/Papirus/24x24/panel. blueman-tray and blueman-active both point at bluetooth-active. blueberry-tray and blueberry-tray-active do as well, so I don't see what you mean with blueberry switching its icon.

linusrg1, 2022-04-29:
> Well, take a look at https://github.com/PapirusDevelopmentTeam/papirus-icon-theme/tree/master/Papirus/24x24/panel. blueman-tray and blueman-active both point at bluetooth-active. blueberry-tray and blueberry-tray-active do as well, so I don't see what you mean with blueberry switching its icon.

When I had a device connected via bluetooth with blueberry, its icon was successfully able to switch over to it's device connected icon. The bluetooth icon with a dot on the left and right side of it, similar to the device connected bluetooth icon on android. With blueman however, I understand its version of this is changing its colour, but for me it is always white, even when a device is connected.

cschramm, 2022-04-30:
Ok, I got it. Papirus seems just very inconsistent there. blueman-tray, blueman-active, blueberry-tray and bluberry-tray-active all point to a single bluetooth-active icon. However, blueberry does support symbolic icons and if that is enabled, it uses blueberry-tray-symbolic and blueberry-tray-active-symbolic and those *are* different in Papirus: https://github.com/PapirusDevelopmentTeam/papirus-icon-theme/blob/master/Papirus/symbolic/status/blueberry-tray-active-symbolic.svg

In any case, this is not a blueman issue. If the default ShowConnected plugin is enabled, blueman switches its icon from blueman-tray to blueman-active and the shipped icons are different but your theme replaces them with two identical icons.

If you disable blueberry's `use-symbolic-icons` option Papirus actually breaks its icons in the exact same way, just not for the symbolic ones.

linusrg1, 2022-04-30:
> Ok, I got it. Papirus seems just very inconsistent there. blueman-tray, blueman-active, blueberry-tray and bluberry-tray-active all point to a single bluetooth-active icon. However, blueberry does [...]

ok ok thanks. So I should really be reporting this on papirus's github then ig.

closed 2022-06-24 by cschramm