blueman-project/blueman #1702: Is there a way to start blueman-manager minimized to the tray?

kendoori, 2022-04-14:
blueman: 2.12
BlueZ: 5.64
Distribution: Mint 20.3
Desktop environment: Cinnamon

* [X ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2022-04-16:
blueman-mechanism does not have such a feature and as far as I know there's no generic mechanism like a standard environment variable in GTK. Of course you can always fake it for any application by automatically starting and then minimizing it right away.

I wonder about your use case, though. There does not seem to be any point in starting blueman-manager when you do not want to interact with it.

kendoori, 2022-04-16:
@cschramm my use case is that I use it as a visual indicator in my tray area regarding the status of my connection.

kendoori, 2022-04-16:
I used the following workaround to run blueman-manager and close it to the tray on startup: https://askubuntu.com/a/1133211/2692

closed 2022-04-16 by kendoori

cschramm, 2022-04-16:
Those are distinct processes. You can run blueman-applet - typically on autostart anyway - for the tray icon. It does not have any main window and there is no point in starting blueman-manager as long you do not plan to interact with it.

kendoori, 2022-04-16:
> blueman-applet

That was the missing piece. I didn't realize that this is a separate element. That will solve this problem :-)