blueman-project/blueman #1703: Add bluetooth battery percentage support for audio devices

linusrg1, 2022-04-17:
blueman: 2.1.2
BlueZ:5.53-0ubuntu3.5
Distribution: Mint 20.3
Desktop environment: Cinnamon

It would be great if I had the ability to view the percentage of my bluetooth devices.

cschramm, 2022-04-28:
blueman 2.2 show notifications on connect and a status in manager windows for LE battery service.

There are dedicated applets for a permanent battery status and Cinnamon for sure has one.

daniejstriata, 2022-04-29:
I'm using blueman 2.2 myself and have not been able to get the headset's battery level after enabling experimental features and using upower.

cschramm, 2022-04-29:
@daniejstriata: BlueZ (>= 5.48) picks up LE battery services and upower (>= 0.99.7) picks up BlueZ battery services, so there seems to be no LE battery service in your case. What kind of battery service do you expect? Besides standard LE battery services I'm only aware of proprietary HID protocols.

daniejstriata, 2022-04-29:
My bluetooth headset. I would like to change it before it goes flat. On Android I can see the battery level without installing extra software.

cschramm, 2022-04-29:
So, after some research it seems like there are non-standard AT commands sent on RFCOMM for HFP/HSP devices that provide battery level information. Android seems to have pretty good support for them but BlueZ / pulseaudio / upower / younameit does not. You could give https://github.com/TheWeirdDev/Bluetooth_Headset_Battery_Level a try with which unfortunately I didn't have any luck with my devices so far.

daniejstriata, 2022-04-29:
I tried the too :-) All I got was device busy or something to that effect. I'll still try going forward and will revert here if I find anything concrete.

cschramm, 2022-05-02:
@MarijnS95 mentions upcoming support in pulseaudio in https://github.com/blueman-project/blueman/issues/1708#issuecomment-1115355790 :slightly_smiling_face:

MarijnS95, 2022-05-02:
Reference: https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/merge_requests/482

souravdas142, 2022-05-03:
for `PipeWire` there is no mechanism like that right? @MarijnS95 

@cschramm  why not look up and show battery information with periodic time ? I get battery information always from `dbus`
```bash
 $ dbus-send --print-reply=literal --system --dest=org.bluez /org/bluez/hci0/dev_F4_4E_FD_CF_B8_7E org.freedesktop.DBus.Properties.Get string:"org.bluez.Battery1" string:"Percentage"
   variant       byte 100
```
After Enabling the experimental feature in bluez. `upower` also reported the same.

MarijnS95, 2022-05-03:
@souravdas142 pipewire implemented the same mechanism. Now it's just waiting for stabilization of the BlueZ interface before this is generally available.

cschramm, 2022-05-03:
@souravdas142: Oh, yeah, I guess we're at fault there as we try to get the battery data only if the device has a battery service (`0000180f-0000-1000-8000-00805f9b34fb`), so we will probably miss battery data provided via the provider API. I don't remember why I implemented it that way and not based on the battery interface itself. Will give it a try.

souravdas142, 2022-05-03:
> Are you saying you see the battery in D-Bus even if the experimental feature is not enabled?

@cschramm  After enabling the "Experimental" feature I can see only then.

So can blueman for now conditionally check if Experimental feature is enabled for bluez, then only display battery information periodically until bluez API got stable and have this feature by default ?

MarijnS95, 2022-05-03:
> So can blueman for now conditionally check if Experimental feature is enabled for bluez

@souravdas142 just like what @cschramm you should only look for the presence of the interface itself, and of course also listen to `InterfacesAdded` as it is bound to show up late.

cschramm, 2022-05-03:
@souravdas142: You could give #1710 a try. It should pick up the provided battery data as well, but probably only works for the manager window, not the applet plugin / notifications.

souravdas142, 2022-05-04:
> @souravdas142: You could give #1710 a try. It should pick up the provided battery data as well, but probably only works for the manager window, not the applet plugin / notifications.

works, See : https://github.com/blueman-project/blueman/pull/1710#issuecomment-1115971180

closed 2022-06-06 by infirit