blueman-project/blueman #1707: Some inconsistency on UI

kuraga, 2022-05-02:
If Bluetooth is disconnected, Bluetooth Devices window is closed after an appropriate dialog.

But if Bluetooth Devices window is opened and then Bluetooth disconnects, window is nor closed nor cleared. Some inconsistency...

Thanks.

cschramm, 2022-05-02:
Not really sure what you mean.

I guess it's about disabled Bluetooth, i.e. existing adapters are not powered. In that case, blueman-manager shows a hint and a way to power the adapter(s) on start. If the window is already open when adapters get unpowered, it does not close, but it does get cleared.

Is that what you mean? I guess it would be a bit harsh and does not add value if we'd just close it.

kuraga, 2022-05-02:
@cschramm , thanks for reply!

> blueman-manager shows a hint and a way to power the adapter(s) on start.

...and close the window.

> If the window is already open when adapters get unpowered, it does not close, but it does get cleared.

On my system, it doesn't get cleared.

P.S. It's not so important, just for knowledge.

cschramm, 2022-05-02:
That could be a bug then. blueman-manager should only show devices bound to powered adapters. Please provide the basic information we ask for in the issue template and the output of `blueman-manager --loglevel debug` of when you disable Bluetooth.

kuraga, 2022-05-02:
```
blueman-manager 19.04.13 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Class': 0}
blueman-manager 19.04.13 INFO     ManagerDeviceList:369 row_update_event: ServicesResolved False
blueman-manager 19.04.13 INFO     ManagerDeviceList:369 row_update_event: Connected False
blueman-manager 19.04.13 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_1C_6E_4C_80_0D_6E {'ServicesResolved': False, 'Connected': False}
blueman-manager 19.04.13 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Powered': False, 'Discovering': False}
blueman-manager 19.04.13 INFO     PulseAudioUtils:324 __event_callback: 41 162
blueman-manager 19.04.13 DEBUG    PulseAudioProfile:46 on_pa_event: 41 162
blueman-manager 19.04.13 INFO     PulseAudioProfile:58 on_pa_event: card
blueman-manager 19.04.13 INFO     PulseAudioProfile:63 on_pa_event: remove
```

infirit, 2022-05-02:
unpacking returns a tuple, quick fix for that.

```patch
diff --git a/blueman/main/Manager.py b/blueman/main/Manager.py
index dab76bd1..0277e93e 100644
--- a/blueman/main/Manager.py
+++ b/blueman/main/Manager.py
@@ -94,7 +94,8 @@ class Blueman(Gtk.Application):
             def on_applet_signal(_proxy: AppletService, _sender: str, signal_name: str, params: GLib.Variant) -> None:
                 if signal_name == 'BluetoothStatusChanged':
                     status = params.unpack()
-                    bt_status_changed(status)
+                    if status:
+                        bt_status_changed(status[0])
 
             def on_dbus_name_vanished(_connection: Gio.DBusConnection, name: str) -> None:
                 logging.info(name)
```

Here is a proper log, we report that failed enabling bluetooth. Not sure why..

```
$ python apps/blueman-manager --loglevel debug
blueman-manager 21.04.47 INFO     PluginManager:87 load_plugin: ['Info', 'Services', 'Notes', 'PulseAudioProfile']
blueman-manager 21.04.47 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.manager.Info.Info'>
blueman-manager 21.04.47 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.manager.Services.Services'>
blueman-manager 21.04.47 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.manager.Notes.Notes'>
blueman-manager 21.04.47 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.manager.PulseAudioProfile.PulseAudioProfile'>
blueman-manager 21.04.47 INFO     PulseAudioUtils:193 pa_context_event: 1
blueman-manager 21.04.47 INFO     PulseAudioUtils:193 pa_context_event: 2
blueman-manager 21.04.47 INFO     Manager:122 on_dbus_name_appeared: org.bluez :1.11
blueman-manager 21.04.47 INFO     PulseAudioUtils:193 pa_context_event: 3
blueman-manager 21.04.47 INFO     PulseAudioUtils:193 pa_context_event: 4
blueman-manager 21.04.47 INFO     PulseAudioProfile:33 on_pa_ready: connected
blueman-manager 21.04.47 INFO     PulseAudioUtils:199 <lambda>  : 1
blueman-manager 21.04.47 INFO     PulseAudioUtils:324 __event_callback: 25 46
blueman-manager 21.04.47 DEBUG    PulseAudioProfile:46 on_pa_event: 25 46
blueman-manager 21.04.47 INFO     PulseAudioProfile:58 on_pa_event: card
blueman-manager 21.04.47 INFO     PulseAudioProfile:60 on_pa_event: change
Failed to enable bluetooth
blueman-manager 21.04.55 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Address': '3C:7C:3F:98:FE:BA', 'AddressType': 'public', 'Name': 'artix', 'Alias': 'Asus BT500', 'Class': 0, 'Powered': False, 'Discoverable': False, 'DiscoverableTimeout': 180, 'Pairable': True, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['00001133-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00001104-0000-1000-8000-00805f9b34fb', '00005005-0000-1000-8000-0002ee000001', '00001108-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '00001116-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '0000111f-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0540', 'Roles': ['central', 'peripheral']}
blueman-manager 21.04.55 ERROR    Manager:138 on_dbus_name_appeared: Default adapter not found, trying first available.
blueman-manager 21.04.55 DEBUG    DeviceList:190 set_adapter: Setting adapter to: hci1
blueman-manager 21.04.55 WARNING  DeviceList:197 set_adapter: Failed to set adapter, trying first available.
blueman-manager 21.04.55 DEBUG    DeviceList:190 set_adapter: Setting adapter to: None
blueman-manager 21.04.55 DEBUG    ManagerToolbar:69 on_adapter_changed: toolbar adapter /org/bluez/hci0
blueman-manager 21.04.55 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_D2_0C_FD_5A_21_20 {'Address': 'D2:0C:FD:5A:21:20', 'AddressType': 'random', 'Name': 'M720 Triathlon', 'Alias': 'M720 Triathlon', 'Appearance': 962, 'Icon': 'input-mouse', 'Paired': True, 'Trusted': True, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '0000180f-0000-1000-8000-00805f9b34fb', '00001812-0000-1000-8000-00805f9b34fb', '00010000-0000-1000-8000-011f2000046d'], 'Modalias': 'usb:v046DpB015d0007', 'Adapter': '/org/bluez/hci0', 'ServicesResolved': False, 'WakeAllowed': True}
blueman-manager 21.04.55 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_94_58_CB_CA_82_AE {'Address': '94:58:CB:CA:82:AE', 'AddressType': 'public', 'Name': 'Pro Controller', 'Alias': 'Pro Controller', 'Class': 1288, 'Icon': 'input-gaming', 'Paired': True, 'Trusted': True, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['00001000-0000-1000-8000-00805f9b34fb', '00001124-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v057Ep2009d0001', 'Adapter': '/org/bluez/hci0', 'ServicesResolved': False, 'WakeAllowed': True}
[... cut ...]
blueman-manager 21.04.55 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.manager.PulseAudioProfile.PulseAudioProfile object at 0x7fe42a783e20>
blueman-manager 21.04.55 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.manager.Notes.Notes object at 0x7fe42a783dc0>
blueman-manager 21.04.55 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.manager.Services.Services object at 0x7fe42b1d4490>
blueman-manager 21.04.55 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.manager.Info.Info object at 0x7fe42b1d4280>
```

infirit, 2022-05-02:
The failure is a timing thing. After we call `SetBluetoothStatus` we immediately ask for the status which is not updated (yet). Adding a simple sleep(1) fixes the exit being called. Not sure what the best solution is for this.

```patch
iff --git a/blueman/Functions.py b/blueman/Functions.py
index 73395c0f..68d52232 100644
--- a/blueman/Functions.py
+++ b/blueman/Functions.py
@@ -80,6 +80,7 @@ def check_bluetooth_status(message: str, exitfunc: Callable[[], Any]) -> None:
             return
 
     applet.SetBluetoothStatus('(b)', True)
+    sleep(1)
     if not applet.GetBluetoothStatus():
         print('Failed to enable bluetooth')
         exitfunc()
```

kuraga, 2022-05-02:
Confirm nor first nor second patch.

cschramm, 2022-05-02:
The blueman-manager log looks like BlueZ does not remove the devices and adapter. I'd expect that if no killswitch gets enabled. However,  blueman-applet's Killswitch plugin should do exactly that. Unless you disabled the plugin, please provide the `blueman-applet --loglevel debug` output of when you disable Bluetooth.

kuraga, 2022-05-03:
```
blueman-applet 11.54.54 INFO     PowerManager:139 request_power_state: Requesting False
blueman-applet 11.54.54 INFO     KillSwitch:138 on_power_state_change_requested: False
blueman-applet 11.54.54 DEBUG    KillSwitch:151 on_power_state_change_requested: Using mechanism to set state: False
blueman-applet 11.54.54 INFO     KillSwitch:112 io_event  : killswitch changed 1
blueman-applet 11.54.54 INFO     KillSwitch:122 io_event  : State: False
blueman-applet 11.54.54 INFO     PowerManager:178 update_power_state: off True | foff False | on True | current state True | new state False
blueman-applet 11.54.54 INFO     PowerManager:181 update_power_state: Signalling False
blueman-applet 11.54.54 INFO     KillSwitch:112 io_event  : killswitch changed 4
blueman-applet 11.54.54 INFO     KillSwitch:122 io_event  : State: False
blueman-applet 11.54.54 INFO     PowerManager:178 update_power_state: off True | foff False | on True | current state False | new state False
blueman-applet 11.54.54 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_1C_6E_4C_80_0D_6E {'Connected': False}
blueman-applet 11.54.54 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Class': 0, 'Powered': False, 'Discovering': False}
blueman-applet 11.54.54 DEBUG    DiscvManager:92 on_adapter_property_changed: prop Class 0
blueman-applet 11.54.54 INFO     PowerManager:178 update_power_state: off True | foff False | on False | current state False | new state False
blueman-applet 11.54.54 DEBUG    DiscvManager:92 on_adapter_property_changed: prop Powered False
blueman-applet 11.54.54 INFO     DiscvManager:109 on_adapter_property_changed: Stop
blueman-applet 11.54.54 DEBUG    DiscvManager:92 on_adapter_property_changed: prop Discovering False
blueman-applet 11.54.54 INFO     BluezAgent:222 _on_cancel: Agent.Cancel
blueman-applet 11.54.54 INFO     PowerManager:124 check     : callbacks done
blueman-applet 11.54.54 INFO     PowerManager:95 set_adapter_state: False
blueman-applet 11.54.54 INFO     PowerManager:178 update_power_state: off True | foff False | on False | current state False | new state False
```

cschramm, 2022-05-03:
Your system seems to behave differently than mine there. blueman does seem to set the killswitch states as expected, but BlueZ does not seem to remove the adapter. Quick cross-check: Do you still see the adapter with `bluetoothctl list` after an `rfkill block bluetooth`? That'd be surprising.

For me, bluetoothd shows something like this when disabling Bluetooth with blueman-applet:

```
bluetoothd[51949]: src/rfkill.c:rfkill_event() RFKILL event idx 0 type 2 op 2 soft 1 hard 0
bluetoothd[51949]: src/adapter.c:dev_class_changed_callback() Class: 0x000000
bluetoothd[51949]: src/adapter.c:new_settings_callback() Settings: 0x00000ad0
bluetoothd[51949]: src/adapter.c:settings_changed() Changed settings: 0x00000001
bluetoothd[51949]: src/adapter.c:settings_changed() Pending settings: 0x00000000
bluetoothd[51949]: src/adapter.c:cancel_passive_scanning() 
bluetoothd[51949]: src/adapter.c:adapter_stop() adapter /org/bluez/hci0 has been disabled
bluetoothd[51949]: src/rfkill.c:rfkill_event() RFKILL event idx 6 type 2 op 2 soft 1 hard 0
bluetoothd[51949]: src/adapter.c:index_removed() index 0
bluetoothd[51949]: src/adapter.c:adapter_unregister() Unregister path: /org/bluez/hci0
```

I guess in your case the `index_removed` and thus `adapter_unregister` does not show up. It's a reaction to a kernel event, so it might even depend on a driver or something.

Anyway, blueman is just showing what BlueZ provides, so if BlueZ still provides devices when blocked it's fine that blueman shows them. I guess you cannot do anything useful with them, though.

kuraga, 2022-05-03:
```
$ sudo rfkill block bluetooth
$ sudo rfkill
ID TYPE      DEVICE               SOFT      HARD
 0 wlan      ideapad_wlan      blocked unblocked
 1 bluetooth ideapad_bluetooth blocked unblocked
 3 wlan      phy0              blocked unblocked
 4 bluetooth hci0              blocked unblocked
$ sudo bluetoothctl list
Controller <MAC-address> <hostname> [default]
```

cschramm, 2022-05-03:
Got it. It's not the Bluetooth stack that removes the adapter for me but the driver or whatever. When I soft-block Bluetooth (`tpacpi_bluetooth_sw` in my case), I can see the USB device getting completely removed in `dmesg`. Even the `hci0` killswitch gets removed then. I guess that's specific to my driver, chip or whatever, so it's expected that blueman still shows the devices in your case.

infirit, 2022-05-03:
> Got it. It's not the Bluetooth stack that removes the adapter for me but the driver or whatever. When I soft-block Bluetooth (`tpacpi_bluetooth_sw` in my case), I can see the USB device getting [...]

This is new for me too. I didn't notice that when powering down or blocking the kernel/BlueZ didn't remove the adapter or devices, It definitely used to for my dongles. I'm on BlueZ 5.64.

cschramm, 2022-05-04:
I'm on 5.64 as well. To my understanding it's the ACPI hardward / thinkpad_acpi driver that physically unpowers the integrated adapter when `tpacpi_bluetooth_sw` gets blocked.

infirit, 2022-05-04:
Yep, that's what happened with btusb driver for my dongles, not sure when it changed.

Do we want the list to be cleared? I mean we could just make the Bluetooth disabled dialog model so you won't be able to interact with the list. And then make any menus or buttons in other places insensitive.

kuraga, 2022-05-04:
Nevertheless, propose to change adapter enabling behavior, too.

> > blueman-manager shows a hint and a way to power the adapter(s) on start.
>
> ...and closes the Bluetooth Devices window.

Propose not close the window.

infirit, 2022-05-06:
Not working but this is what I am thinking. This dialog shows when bluetooth is turned off and goes away if on.

[image]

```patch
diff --git a/blueman/main/Manager.py b/blueman/main/Manager.py
index 6248d244..401c3971 100644
--- a/blueman/main/Manager.py
+++ b/blueman/main/Manager.py
@@ -42,6 +42,7 @@ class Blueman(Gtk.Application):
         s.attach()
 
     window: Optional[Gtk.ApplicationWindow]
+    notpowered: Gtk.MessageDialog
 
     def do_startup(self) -> None:
         def doquit(_a: Gio.SimpleAction, _param: None) -> None:
@@ -93,6 +94,9 @@ class Blueman(Gtk.Application):
 
             def bt_status_changed(status: bool) -> None:
                 assert self.window is not None
+                self._show_bt_disabled(status)
+                return
+
                 if not status:
                     self.window.hide()
                     check_bluetooth_status(_("Bluetooth needs to be turned on for the device manager to function"),
@@ -177,6 +181,12 @@ class Blueman(Gtk.Application):
 
             Manager.watch_name_owner(on_dbus_name_appeared, on_dbus_name_vanished)
 
+        self.notpowered = Gtk.MessageDialog(
+            type=Gtk.MessageType.ERROR, visible=False, modal=True, icon_name="blueman", title="Bluetooth disabled",
+            secondary_text="Bluetooth is disabled do you want to turn it on?", parent=self.window)
+        self.notpowered.add_button("Quit", Gtk.ResponseType.NO)
+        self.notpowered.add_button("Turn on", Gtk.ResponseType.YES)
+
         self.window.present_with_time(Gtk.get_current_event_time())
 
     def _on_configure(self, _window: Gtk.ApplicationWindow, event: Gdk.EventConfigure) -> bool:
@@ -249,3 +259,9 @@ class Blueman(Gtk.Application):
     def remove(self, device: Device) -> None:
         assert self.List.Adapter
         self.List.Adapter.remove_device(device)
+
+    def _show_bt_disabled(self, status):
+        if not status:
+            self.notpowered.show()
+        else:
+            self.notpowered.hide()
```

closed 2024-02-20 by cschramm