blueman-project/blueman #1709: Device confusion

kuraga, 2022-05-02:
blueman: 2.2.3
BlueZ: 5.64
Distribution: Calculate Linux (Gentoo)
Desktop environment: XFCE 4.16

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

On turning on `MAJOR IV` headset (MAC `...:6E`) I see:
```
blueman-manager 22.39.15 INFO     ManagerDeviceList:369 row_update_event: Connected True
blueman-manager 22.39.15 DEBUG    ManagerDeviceMenu:297 generate  : JBL E45BT
blueman-manager 22.39.15 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_..._6E {'Connected': True}
```
(BTW, auto-connection is turned off for `MAJOR IV` if I look for menu.)

JBL E45BT!!!!!!

Possible connected issue: #1240.

cschramm, 2022-05-02:
I deduce that you are surprise to see "JBL E45BT" in the debug output when "MAJOR IV" connects. That is completely unrelated, don't worry. `ManagerDeviceMenu:297 generate` logs its selected row's device name.

#1240 is about an animation glitch.

closed 2022-05-02 by cschramm

kuraga, 2022-05-02:
Ok, but let me check `MAJOR IV` shouldn't auto-connect? I see that it shouldn't but does.

cschramm, 2022-05-02:
Both sides can initiate a connection (the remote side usually needs to be trusted explicitly for that), blueman might not be the only piece of software that initiates a connection on its side, and there are temporary connected states that do not refer to any services, e.g. for resolving LE services.

kuraga, 2022-09-19:
BTW, after this connection: `Audio profiles` menu item is a (empty) menu **item instead of submenu**.