blueman-project/blueman #1712: blueman-applet 2.2.4-2~ubuntu20.04 no longer functions

daniejstriata, 2022-05-09:
blueman: 2.2.4-2
BlueZ: 5.53-0ubuntu3.5
Distribution: Ubuntu 20.04
Desktop environment: Mint/ Cinnamon

The systemtray applet no longer functions/ displays on system tray. When I run it manually I get the following error and have to end process with CTRL-C:
```
→ ps faux | grep blueman
danie.d+   15999  0.0  0.0   8908   720 pts/1    S+   10:06   0:00          \_ grep --color=auto blueman
→ python3 /usr/bin/blueman-applet
blueman-applet 10.06.12 WARNING  PluginManager:147 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-applet 10.06.12 WARNING  PluginManager:147 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 10.06.12 ERROR    AgentManager:20 on_register_failed: /org/bluez/obex/agent/blueman org.bluez.obex.Error.AlreadyExists Agent already exists
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/main/Tray.py", line 29, in _on_name_appeared
    indicator_class = getattr(import_module('blueman.main.indicators.' + indicator_name), indicator_name)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 848, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/lib/python3/dist-packages/blueman/main/indicators/AppIndicator.py", line 6, in <module>
    gi.require_version('AyatanaAppIndicator3', '0.1')
  File "/usr/lib/python3/dist-packages/gi/__init__.py", line 129, in require_version
    raise ValueError('Namespace %s not available' % namespace)
ValueError: Namespace AyatanaAppIndicator3 not available
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 153, in apport_excepthook
    with os.fdopen(os.open(pr_filename,
FileNotFoundError: [Errno 2] No such file or directory: '/var/crash/_usr_bin_blueman-tray.1001.crash'

Original exception was:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/main/Tray.py", line 29, in _on_name_appeared
    indicator_class = getattr(import_module('blueman.main.indicators.' + indicator_name), indicator_name)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 848, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/lib/python3/dist-packages/blueman/main/indicators/AppIndicator.py", line 6, in <module>
    gi.require_version('AyatanaAppIndicator3', '0.1')
  File "/usr/lib/python3/dist-packages/gi/__init__.py", line 129, in require_version
    raise ValueError('Namespace %s not available' % namespace)
ValueError: Namespace AyatanaAppIndicator3 not available
```

cschramm, 2022-05-09:
Pardon my ignorance, but what is version 2.2.4-2~ubuntu20.04? I cannot spot anything like that in the Ubuntu repositories.

It looks like blueman-applet itself works fine but blueman-tray - the application that shows the tray icon - crashes as it cannot find AyatanaAppIndicator3. That's expected if you do not have the optional gir1.2-ayatanaappindicator3 package. However, blueman-tray will only expect to find it whenever blueman-applet tells blueman-tray to use it and blueman-applet can only tell blueman-tray to use it, if it successfully loaded it itself.

So, first off, do you have gir1.2-ayatanaappindicator3? Perceived blueman-applet behavior indicates yes, blueman-tray behavior indicates no. Start whichever application turns out to not match your system state with `--loglevel debug`, so that we can see what it does.

daniejstriata, 2022-05-09:
It is the version I got from installing using the official PPA. I did not know of AyatanaAppIndicator3. It was available to install. After installing it the tray icon worked again. It would appear the PPA needs to add AyatanaAppIndicator3 as a required dependency.

cschramm, 2022-05-09:
The blueman project does not provide any PPA.

AyatanaAppIndicator3 is an optional dependency. Things should work without it.

The blueman-applet log looks like AyatanaAppIndicator3 is loading fine. Am I right, that the log is from after you installed it?

daniejstriata, 2022-05-09:
The first logs I shared did not present me with the blueman tray icon. Now that AyatanaAppIndicator3 is installed the tray icon works.

closed 2022-05-09 by cschramm