blueman-project/blueman #1717: ValueError: Namespace AyatanaAppIndicator3 not available

raphaelh, 2022-05-17:
blueman: 2.1.2-1ubuntu0.3
BlueZ: 5.53-0ubuntu3.5
Distribution: Ubuntu 20.04.4 LTS
Desktop environment: XFCE

Problem similar to https://github.com/blueman-project/blueman/issues/963

```
$ blueman-tray
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/main/Tray.py", line 29, in _on_name_appeared
    indicator_class = getattr(import_module('blueman.main.indicators.' + indicator_name), indicator_name)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 848, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/lib/python3/dist-packages/blueman/main/indicators/AppIndicator.py", line 6, in <module>
    gi.require_version('AyatanaAppIndicator3', '0.1')
  File "/usr/lib/python3/dist-packages/gi/__init__.py", line 129, in require_version
    raise ValueError('Namespace %s not available' % namespace)
ValueError: Namespace AyatanaAppIndicator3 not available
```

I fixed it with :
```
$ sudo apt install gir1.2-ayatanaappindicator3-0.1 
```

Thanks

nexhub, 2022-05-17:
I can confirm the error and the fix is working for me :)
```
Ubuntu 20.04.4 LTS
blueman-git         2.2.4-2~ubuntu20.04
bluez                   5.53-0ubuntu3.5
xubuntu-desktop 2.233
```

cschramm, 2022-05-17:
Thanks for the report. The problem seems to be that blueman-applet tells blueman-tray to use the Ayatana indicator even though that's generally only possible if it's able to load it itself. Without the library installed, please stop blueman-applet and run it with `blueman-applet --loglevel debug` to collect its output.

nexhub, 2022-05-17:
```
% blueman-applet --loglevel debug
blueman-applet 15.37.47 INFO     PluginManager:85 load_plugin: ['PPPSupport', 'AuthAgent', 'Menu', 'SerialManager', 'RecentConns', 'StandardItems', 'ShowConnected', 'AutoConnect', 'GameControllerWakelock', 'DiscvManager', 'NMDUNSupport', 'ExitItem', 'AppIndicator', 'TransferService', 'Battery', 'DisconnectItems', 'PowerManager', 'DBusService', 'ConnectionNotifier', 'Networking', 'DhcpClient', 'StatusIcon', 'KillSwitch', 'NetUsage', 'NMPANSupport']
blueman-applet 15.37.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 15.37.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 15.37.47 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Address': '04:D3:B0:C0:06:5B', 'AddressType': 'public', 'Name': 'think', 'Alias': 'think', 'Class': 6029580, 'Powered': True, 'Discoverable': False, 'DiscoverableTimeout': 180, 'Pairable': False, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['00001133-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00001104-0000-1000-8000-00805f9b34fb', '00005005-0000-1000-8000-0002ee000001', '00001108-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '0000111f-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0535'}
blueman-applet 15.37.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 15.37.47 WARNING  PluginManager:147 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-applet 15.37.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 15.37.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 15.37.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 15.37.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 15.37.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 15.37.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 15.37.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 15.37.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'>
blueman-applet 15.37.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 15.37.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
blueman-applet 15.37.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 15.37.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.AppIndicator.AppIndicator'>
blueman-applet 15.37.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 15.37.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Battery.Battery'>
blueman-applet 15.37.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 15.37.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier'>
blueman-applet 15.37.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 15.37.47 INFO     Networking:35 load_nap_settings: Loading NAP settings
blueman-applet 15.37.47 WARNING  PluginManager:147 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 15.37.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 15.37.47 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'>
blueman-applet 15.37.47 INFO     KillSwitch:106 io_event  : killswitch registered 0
blueman-applet 15.37.47 INFO     KillSwitch:122 io_event  : State: True
blueman-applet 15.37.47 INFO     PowerManager:178 update_power_state: off False | foff False | on True | current state True | new state True
blueman-applet 15.37.47 INFO     KillSwitch:85 _on_connman_vanished: net.connman vanished
blueman-applet 15.37.47 INFO     KillSwitch:106 io_event  : killswitch registered 2
blueman-applet 15.37.47 INFO     KillSwitch:122 io_event  : State: True
blueman-applet 15.37.47 INFO     PowerManager:178 update_power_state: off False | foff False | on True | current state True | new state True
[... cut ...]
  File "<frozen importlib._bootstrap_external>", line 848, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/lib/python3/dist-packages/blueman/main/indicators/AppIndicator.py", line 6, in <module>
    gi.require_version('AyatanaAppIndicator3', '0.1')
  File "/usr/lib/python3/dist-packages/gi/__init__.py", line 129, in require_version
    raise ValueError('Namespace %s not available' % namespace)
ValueError: Namespace AyatanaAppIndicator3 not available

```

cschramm, 2022-05-25:
@nexhub: What is blueman-git 2.2.4-2~ubuntu20.04? You might be using https://launchpad.net/~pipewire-debian/+archive/ubuntu/testing. It's a third-party repository that says `Do not use this repo` and its `2.2.4-2`s contain a broken patch on top of the Debian patches (the packages require `gir1.2-ayatanaappindicator3-0.1` or `gir1.2-appindicator3-0.1` and their blueman-applet code works with either but their blueman-tray code only works with Ayatana).

nexhub, 2022-05-30:
I installed from [ppa:pipewire-debian/pipewire-upstream](https://launchpad.net/~pipewire-debian/+archive/ubuntu/pipewire-upstream) via these instructions: https://github.com/pipewire-debian/pipewire-debian
So this is a problem with the packagers patches? If you confirm, I'd file a bug with them. 
And if I can provide any additional information, please hit me up.

infirit, 2022-06-07:
> Yes, that package is broken, as described above.

Is this because if the ayatana patch? If so maybe we should just handle it up.

cschramm, 2022-06-07:
Debian migrated to Ayatana so the Debian package has a patch for the imports: https://salsa.debian.org/cschramm/blueman/-/blob/5a4c7343ce09633c0f332d3c6d2e8931a55dd6b2/debian/patches/ayatana.patch (just like basically all Debian packages that depend on libappindicator)

Ubuntu just uses the Debian versions unchanged and that works perfectly fine. For some reason, they did not fully follow the Ayatana migration, though, and are still dragging allow libappindicator3 next to libayatana-appindicator3. Does not matter for the blueman package: Just like in Debian it depends on libayatana-appindicator3 and has the patch.

The PPA seems to do their own packaging and did not have the Ayatana patch at all for some reason, then added a broken one that intends to try both Ayatana and non-Ayatana in https://github.com/pipewire-debian/pipewire-debian/pull/57, and just recently fixed it in https://github.com/pipewire-debian/pipewire-debian/issues/62.

Once again, I do maintain an APT repository very close to the Debian packaging (it actually has a branch at https://salsa.debian.org/cschramm/blueman/-/tree/git) at https://blueman.cschramm.eu/debian/ and am happy to support it here. It clearly does not claim to cover as many edge cases as the PPA, though, especially there are no builds for Python 3.6 and thus Ubuntu 18.04 (anymore).

pipewire-debian, 2022-06-08:
Sorry for the mess. @cschramm  I was initially taken the project from ubuntu source. Later I discovered your repo at https://salsa.debian.org/cschramm/blueman/-/tree/git. My Intention is to deliver latest version of blueman to the all existing version of ubuntu easiest possible way via PPA thats it. From now, PPA will respect your repo instead the ubuntu one. I appreciate you for being a Developer and Package maintainer both. Thank you.

cschramm, 2022-06-08:
> I was initially taken the project from ubuntu source.

Not sure how you mean that. The changelog looks like it would be based on 2.1.4-1. However, the ayatana patch got added in 2.1.3-2 already but did not get imported, so it looks like something went wrong initially.

> Need a suggestion, Should I drop that package as you already maintain that ?

I'm totally fine with the PPA. As I said it covers more "edge cases" than my repository, like Ubuntu 18.04 and to be honest I do not claim that e.g. the python3.7 repository's packages' dependencies actually fit the respective Ubuntu releases. They are just taken from https://salsa.debian.org/cschramm/blueman/-/blob/git/debian/control and I do not run any tests or anything on the builds (I do check available versions once when updating dependencies, though).

Out of curiosity I just checked the dependencies and it looks like neither my control file, nor yours, nor blueman's actual dependencies since 2.1 fit Ubuntu 18.04 as it does not have python3-gi >= 3.27.2. Not sure if the PPA actually adds anything, to the package then. Using a PPA might still be easier to some users than following my repository's instructions. :sweat_smile: There also might be some way to discover packages from the Launchpad platform which could be useful to users. :shrug:

pipewire-debian, 2022-06-08:
> Not sure how you mean that. The changelog looks like it would be based on 2.1.4-1. However, the ayatana patch got added in 2.1.3-2 already but did not get imported, so it looks like something went wrong initially.

Yeah my bad. Now remember I followed the [archlinux's packaging](https://archlinux.org/packages/community/x86_64/blueman/). There they don't have `AyatanaAppIndicator` dependencies. And dropped that patch without thinking the affect. 

> it does not have python3-gi >= 3.27.2. Not sure if the PPA actually adds anything

yeah PPA add that [package](https://launchpad.net/~pipewire-debian/+archive/ubuntu/pipewire-upstream/+build/21542953) 

> There also might be some way to discover packages from the Launchpad platform which could be useful to users. 

yeah I agree. Thank you for your support. :sweat_smile:

pipewire-debian, 2022-06-08:
Also curious, from https://salsa.debian.org/cschramm/blueman/-/commit/b6f11156a9f75ad361c54a75f0034506b8158daf officialy `Ayatana` patch is dropped. so https://github.com/blueman-project/blueman/issues/1717#issuecomment-1148799771 gonna happen right ?

cschramm, 2022-06-08:
Oh, I totally forgot about that. :sweat_smile: #1382 dropped the dependency on any shady appindicator lib in favor of a direct StatusNotifierItem implementation, so that there's nothing left to patch.

Now I'm confused. Isn't `blueman-git` meant to ship our `main` branch? How come it contains files that got dropped seven months ago? :confused:

pipewire-debian, 2022-06-09:
I was also thinking to move to the main for the PPA, as now battery indicator also added with experimental bluez (5.56+ AFAIK) feature. I was testing it in my ArchLinux, perfectly working and has no issue.

Is it ok to move to the main ? Actually wants suggestion from you. :smile:

infirit, 2022-06-09:
> I was also thinking to move to the main for the PPA, as now battery indicator also added with experimental bluez (5.56+ AFAIK) feature. I was testing it in my ArchLinux, perfectly working and has no issue.
> 
> Is it ok to move to the main ? Actually wants suggestion from you. 😄

@cschramm wants to release 2.3 soon(ish?) so my suggestion would be to wait for that.

closed 2022-08-16 by github-actions[bot]