blueman-project/blueman #1724: Code won't launch after compilation

Drugwash2, 2022-05-26:
blueman: **2.2.4**
BlueZ: **5.64**
Distribution: **Mint 19.2 x64**
Desktop environment: **Cinnamon 4.2.4**

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

After succesful compilation from either git clone or released package, launching `blueman-manager` yields the following:

```
$ blueman-manager
Traceback (most recent call last):
  File "/usr/local/bin/blueman-manager", line 16, in <module>
    from blueman.main.Manager import Blueman
ModuleNotFoundError: No module named 'blueman.main.Manager'
```
Files pertaining to the project are scattered all around the drive so I can't possibly fix it myself. Any idea...?

cschramm, 2022-05-26:
I guess you installed the package into a standard `site-packages` path but are using the Debian Python interpreter which is modified to use a custom `dist-packages` path. Moving things around might be sufficient to make it work if you do not want to build a proper Debian package.

clefebvre, 2022-05-27:
@Drugwash2 You can get the debian/ dir from the debian branch of https://github.com/linuxmint/blueman. Once you've got that, you can build and install blueman with:

```
dpkg-buildpackage
sudo dpkg -i ../blueman*.deb
```

Drugwash2, 2022-05-28:
> I guess you installed the package into a standard `site-packages` path

Most likely so. So hard for me to accustom with the Linux way(s). Some not-too-vague pointers in a Readme file regarding modifying paths for major distros would make wonders for newbies like me.

> You can get the debian/ dir from the debian branch

Thank you for the pointer, I'll try to do that. Probably tomorrow as it's been a long day. (actually it's been 'tomorrow' for 3.5 hours already) Hopefully it doesn't have too high version requirements.

Any hints as to how to completely clean the previous installation (performed through make & make install)? Yesterday in desperation I've reinstalled the default v2.0.5 from Synaptic and now it won't open the Devices window, says Blueman is already running.

Drugwash2, 2022-05-28:
OK, the compilation part has been fixed thanks to the Debian packaging data, v2.2.4 is up and running... but without the most important bit that prompted all this struggle: battery level info.This is not Blueman's fault, I've been performing tests and whatnot with the help of Tomasz Zamszowy developer of the **bluetooth-battery** Cinnamon applet [here](https://github.com/linuxmint/cinnamon-spices-applets/issues/4351) and still didn't manage to get to the bottom of why battery level is not being detected/displayed on my system for none of the seven headphones.

If anyone is willing to put up with this task please let me know and I'll open another issue. Or maybe it would be better to move to the Mint repository. No idea, they're all interconnected. Anyway, closing this now as it's been solved. Thank you both for your help.

closed 2022-05-28 by Drugwash2

cschramm, 2022-05-28:
Even if battery levels would get reported to bluetoothd, blueman currently would not pick them up, see #1703.

I just skimmed through https://github.com/linuxmint/cinnamon-spices-applets/issues/4351 but what immediately makes me suspicious is that I do not see a single mention of pulseaudio (or PipeWire) there. The initial bit you need is a component that reports battery levels to bluetoothd's experimental API. pulseaudio 15.99.1 can do that as well as Pipewire 0.3.24 (with additional coverage in 0.3.34). Most probably you're running pulseaudio 11.1. Only Ubuntu 22.04 and thus no release of Linux Mint  ships versions that support what you're trying to achieve.

Drugwash2, 2022-05-28:
> Most probably you're running pulseaudio 11.1.

That is true. Never would've thought the sound subsystem - or whatever is called - would be related to hardware specs such as battery level. Doesn't sound logical to me, but then again I've stumbled into enough not-so-logical stuff in Linux to begin doubting what the definition of logical is.

One question arises: what distro type and version could the applet author use to have the applet work for him? After all it's in the Mint Cinnamon Spices repository so -again, logically - he should be using Mint, and there is only Mint 21 at most, which I don't know what version of `pulseaudio` employs. Could it be something else? Or could it be a version lower than 15.99.1 that first implemented what is needed. By the way, could you please specify exactly what component/library/API/etc is the one that provides this battery function? I could at least attempt to upgrade pulseaudio just as I did with UPower (raised to 0.99.11) and Bluez (raised to 5.64).

Alternatively. You mentioned Pipewire. Vaguely heard of it, never stumbled into it, no idea what it is and what it's good for. Will search around. Any known conflicts/incompatibilities with pulseaudio or other system component? I'm a little wary as I have neither a backup nor free space to create one (and no money at all to buy external drives or flash sticks or similar).

Drugwash2, 2022-05-28:
Update: compiled and installed pulseaudio **15.99.1**. No dice, no change.
Then I noticed your reference to the [RFCOMM script](https://github.com/TheWeirdDev/Bluetooth_Headset_Battery_Level) and gave it a shot. After a couple scary errors about not finding RFCOMM port with the Akai **ABTS-S2** and another of 'connection refused' with the Panasonic **RP-BTD5**... it worked! The RP-BTD5 repeatedly reported 60% upon four queries (it stands 40h of continuous play so the drain is slow).

Now maybe it's me but after upgrading pulseaudio it feels the volume is much lower than before. Wondering if anybody else had that feeling after similar upgrade.

Gonna check all other devices when possible; right now I'm close to 36h of continuous messing with the computer, need to relax.

cschramm, 2022-05-28:
I do not have much experience there but let me try to give you some directions / explanations:

pulseaudio is currently "the standard" sound server on Linux desktops. PipeWire is a recent alternative to it.

Regarding battery levels as a "hardware spec": To my understanding classic Bluetooth does not have any generic support for a battery level. There are more or less proprietary or standardized battery level commands used in HIDP (input) and HFP / HSP (audio). BLE introduced a standard generic (read: independent of the other services that the device offers) battery service and battery level characteristic.

The BLE standard data is picked up by BlueZ and from there by recent versions of blueman. Also upower gets to know it from there. The HSP / HFP battery level commands could probably be implemented in BlueZ as well, however, from the "Bluetooth Headset Battery Level" script you see how difficult and shaky things are. The better alternative seems to be that the component that implements the audio profiles also reads their battery level information and that is exactly what happend in pulseaudio, PipeWire and bluez-alsa. bluetoothd supports reporting back the battery level information from there via its experimental provider API.

With pulseaudio 15.99.1 and the experimental API enabled I would expect things to work. It's unexpected that only the script does report something. (Keep in mind that 15.99.1 is a prerelease (no idea why it slipped into a stable - even LTS - release of Ubuntu :shrug:) but just while you were talking 16.0 got released.)

Drugwash2, 2022-05-29:
Thank you so much for the explanation, this is what I needed to clear things in my head. The Wikipedia article **[here](https://en.wikipedia.org/wiki/Bluetooth_Low_Energy)** offered even more in-depth information on the BLE standard, and from what I've read so far - still reading - this "new" standard has worse specs than the classic Basic/EDR one.

The important point here is that **headphones** and **stereo audio streaming** are not even mentioned among the primary use cases as was/is the case with the classic protocol. This makes me wonder why you would want to implement precisely that protocol
in Blueman's headphones/audio management since it's not even suitable to that purpose and possibly such device manufacturers won't implement it. Or am I missing something here?

Now, from the user point of view it doesn't matter which protocols or what kind of "hacks" the software uses internally as long as the visible/tangible outcome is the desired one. Personally I've turned this system upside-down, broke it and fixed it, risked unofficial upgrades to system services - all for this apparently tiny, insignificant feature: bluetooth battery level reporting. In the end one obscure project provided the desired result while all those major upgrades were useless. Generally people go with what works.

I don't know why this feature still isn't working after all those efforts (BlueZ 5.64, UPower 0.99.11, PulseAudio 15.99.1). Maybe I did something wrong. Or maybe the system is too old, maybe the kernel (5.0.0-31) doesn't have the appropriate drivers. Tried all tricks recommended so far and nothing worked. Maybe the devices don't have those services implemented despite them being relatively new (the RP-BTD5 manufacture date is August 2018, bought sometime in 2019 and the RP-HF400B was bought a couple weeks or so later). I can imagine many people still using relatively old devices where they could really use such battery level feature. So there may be certain value in trying to implement the RFCOMM protocol in Blueman at least as an alternative to the BLE one for devices that do not return the necessary response. At least that is my personal opinion as a simple user.

Regarding PulseAudio I did notice references to v16.0 when grabbing the git so the version number may just be a visual detail; for all intents and purposes the code itself that I built and installed may well be 16.0 considering the very close timing between my grabbing the code and their official release.

By the way, yesterday I had also implemented in Blueman the code/diff provided in some comment here regarding interfaces being added and removed. Didn't seem to be useful at anything, maybe it needs something else. Gonna dig deeper into this whole issue.

Drugwash2, 2022-05-30:
Follow up on previous comment: finally managed to get to the point where the system reports the battery, and the Cinnamon dedicated applet mentioned **[here](https://github.com/blueman-project/blueman/issues/1724#issuecomment-1140169717)** now works correctly in detecting and displaying the battery level. More details in **[this](https://github.com/linuxmint/cinnamon-spices-applets/issues/4351#issuecomment-1140767752)** comment.

However, as mentioned in the comment there Blueman somehow blocks the detection signals from the bus so they do not get to the applet while Blueman is running. Not sure whether this is not somehow my fault due to implementing those bits of code by  @infirit from the discussion **[here](https://github.com/blueman-project/blueman/pull/1710#issuecomment-1118541742)**. Currently I'm keeping one of the headphones in continuous usage to observe the evolution of the level reporting. Will revert the changes in Blueman sometime later - maybe after another portion of good sleep - and report back.

In the mean time it would be great if someone could point me to an efficient way of debugging the signals issue.

cschramm, 2022-05-30:
> The important point here is that **headphones** and **stereo audio streaming** are not even mentioned among the primary use cases as was/is the case with the classic protocol.

Not sure what you're referring to.

> This makes me wonder why you would want to implement precisely that protocol in Blueman's headphones/audio management

Note that blueman does not provide any* audio support at all. It just triggers a connect for the device via bluetoothd and in case of audio services bluetoothd expects some component that does handle the protocol, which could be pulseaudio, PipeWire, alsa-bluez, ...

(*) The only thing blueman adresses is switching profiles which is basically an alternative interface to pulseaudio (you can do the same e.g. in pavucontrol).

> However, as mentioned in the comment there Blueman somehow blocks the detection signals from the bus

You mean

> Blueman was/is blocking the UPower detection signals

? blueman does not have any stake in UPower.

Drugwash2, 2022-05-30:
> Not sure what you're referring to.

A table presented in that article compares the specs of the classic BT and BLE. In the **Primary use cases** section the classic protocol mentions headsets and stereo audio streaming while the LE protocol does not. Looking at the specs it is clear that those two, due to the relatively large volume of data transferred are not suitable for the LE protocol. At least that's how I understand things from what I read there. As such there may not be many audio devices in the future to implement the LE protocol **only**, or at all. But maybe I'm wrong.

> Note that blueman does not provide any* audio support at all.

That, I had understood. Wrong word 'protocol' by itself, maybe _info management using the data provided by the LE protocol_ would have been the right phrasing.

> blueman does not have any stake in UPower.

Again maybe wrong word/phrasing. Not sure what the precise inner workings are, could be dBus or something else that deals with sending the device connection/disconnection signals. All I know so far is that, with Blueman running the Cinnamon applet does not receive any of those signals and as such cannot provide the information. As soon as I close Blueman and turn the headphones on again the applet detects them and displays the battery level.
Repeated the sequence a couple times just to be sure, both with the 5.4 kernel and with the current older one 5.0. Therefore it became clear there was a direct connection between the events. Again, not sure if the changes could somehow play a role in this. Looks like it may take quite some time before the level drops from 40% - if it does - so can't check before that.
Not blaming or anything, just wanna make sure things work right with both apps as we never know what combination of apps people may use and what issues may arise.

cschramm, 2022-05-30:
Well, there is the pretty new "LE Audio" spec but not calling audio a *primary* use case of BLE makes sense, yes.

> Wrong word 'protocol' by itself, maybe info management using the data provided by the LE protocol would have been the right phrasing.

Then I think you're mixing things up in

> This makes me wonder why you would want to implement precisely that protocol
in Blueman's headphones/audio management since it's not even suitable to that purpose and possibly such device manufacturers won't implement it.

The LE battery service is the only standard way to communicate battery levels, so battery levels is a use case of LE *only*. That standard is supported by BlueZ and picked up by blueman, upower etc. It has nothing to do with audio. An LE headset still uses HFP / HSP / A2DP, i.e. non-LE profiles for its audio features while it could use the standard LE battery service to provide battery levels just like any other LE device.

There are proprietary ways of communicating battery levels tide to non-LE profiles like Logitech's HID++ protocol or Apple's HFP / HSP commands. Originally those extensions where based on the manufacturer controlling both ends of the connection, e.g. Apple audio devices and Apple iOS, however, they became a de-facto standard by other manufacturers adopting them and today you do have HID++ support including the battery data in Linux and just recently support for the Apple commands in pulseaudio, PipeWire, alsa-bluez etc. BlueZ will support forwarding it just like the LE standard way and blueman will support picking it up, promised (right now there's not much value in it as the BlueZ API is experimental and thus typically disabled). It's not in blueman's scope to implement the profiles itself.

> Again maybe wrong word/phrasing. Not sure what the precise inner workings are, could be dBus or something else that deals with sending the device connection/disconnection signals. All I know so far [...]

Very surprising. From a quick glance the applet just reads the data from UPower (via UPower's D-Bus API), so if you see the battery in UPower the applet should pick it up. blueman does not even use that API - not to mention doing anything that could influence it.

Drugwash2, 2022-05-30:
> The LE battery service is the only standard way to communicate battery levels [...] There are proprietary ways of communicating battery levels [...] however, they became a de-facto standard [...]

So in other words there are two different standards for the same thing although one is not "official". And unless I'm getting things wrong (again) there is the RFCOMM protocol used by HFP/HSP that can provide battery level value for devices that do not have any part of LE implemented. Please correct me if I'm wrong but I don't think ordinary people/users would care what's inside the devices - software-wise - as they only care for the "thing" to work. All I was trying to say was that the older, unofficial standard may still live for a while along with HFP/HSP-only, both older and newer devices may employ them, therefore it would be in users' interest to have Blueman operate with all possible standards for maximum coverage. At this point I'm wondering if somehow the Akai ABTS-S2 device employs only this old HFP/HSP protocol for battery level since it didn't provide anything through UPower/dBus like the other six devices did - one of them even being a lousy chinese copy after the original LG HBS800. I'll try with that RFCOMM script, see what the outcome will be.

Indeed the applet subscribes to the two signals (**DeviceAdded** and **DeviceRemoved**) through a Glib UPower implementation:

```
        dbusCon = proxy.get_connection();
        this.dbusAdd = dbusCon.signal_subscribe('org.freedesktop.UPower', 'org.freedesktop.UPower', 'DeviceAdded', null, null, 0, () => {
            this.setup();
        });
        this.dbusRemove = dbusCon.signal_subscribe('org.freedesktop.UPower', 'org.freedesktop.UPower', 'DeviceRemoved', null, null, 0, () => {
            this.setup();
        });
```

Theoretically there should be no interference between the two applications but somehow there is. And now that I looked closely at the modified file it's my fault for adding that code.

_Just a short rant break, this platform is pissing me off badly - wanted to quickly attach the modified Python file and got this inept notification; is this a **coding** platform or a mockery, Microsoft?!?_
[image]

Back to the issue. As you can see in the code below I've added both pieces of code from that comment, and there are calls to the dbus UPower **org.bluez.Battery1**, so that's where it might stumble and fail to release... something (not good with naming things, sorry). Well, if anything it's a proof that such implementation may not be ideal.
_(Level still didn't drop a bit from 40% - true 40h of continuous play as advertised, and I'm in my 25th hour of fiddling)_

<details>
  <summary>Modified blueman/bluez/Manager.py</summary>

```
import logging
from typing import List, Optional, Callable

from gi.repository import GObject, Gio

from blueman.bluez.Adapter import Adapter
from blueman.bluez.Device import Device
from blueman.bluez.errors import DBusNoSuchAdapterError
from blueman.gobject import SingletonGObjectMeta
from blueman.bluemantyping import GSignals

class Manager(GObject.GObject, metaclass=SingletonGObjectMeta):
    __gsignals__: GSignals = {
        'adapter-added': (GObject.SignalFlags.NO_HOOKS, None, (str,)),
        'adapter-removed': (GObject.SignalFlags.NO_HOOKS, None, (str,)),
        'device-created': (GObject.SignalFlags.NO_HOOKS, None, (str,)),
        'device-removed': (GObject.SignalFlags.NO_HOOKS, None, (str,)),
        'interface-added': (GObject.SignalFlags.NO_HOOKS, None, (str, str)),
        'interface-removed': (GObject.SignalFlags.NO_HOOKS, None, (str, str)),
        'battery-added': (GObject.SignalFlags.NO_HOOKS, None, (str,)),
        'battery-removed': (GObject.SignalFlags.NO_HOOKS, None, (str,)),
    }

    connect_signal = GObject.GObject.connect
    disconnect_signal = GObject.GObject.disconnect

    __bus_name = 'org.bluez'

    def __init__(self) -> None:
        super().__init__()
        self._object_manager = Gio.DBusObjectManagerClient.new_for_bus_sync(
            Gio.BusType.SYSTEM, Gio.DBusObjectManagerClientFlags.DO_NOT_AUTO_START,
            self.__bus_name, '/', None, None, None)

        self._object_manager.connect("object-added", self._on_object_added)
        self._object_manager.connect("object-removed", self._on_object_removed)
        self._object_manager.connect("interface-added", self._on_interface_added)
        self._object_manager.connect("interface-removed", self._on_interface_removed)

    def _on_object_added(self, _object_manager: Gio.DBusObjectManager, dbus_object: Gio.DBusObject) -> None:
        device_proxy = dbus_object.get_interface('org.bluez.Device1')
        adapter_proxy = dbus_object.get_interface('org.bluez.Adapter1')
[... 83 lines cut ...]
            if proxy:
                assert isinstance(proxy, Gio.DBusProxy)
                object_path = proxy.get_object_path()
                if object_path.startswith(adapter_path):
                    paths.append(object_path)

        return [Device(obj_path=path) for path in paths]

    def find_device(self, address: str, adapter_path: str = "/") -> Optional[Device]:
        for device in self.get_devices(adapter_path):
            if device['Address'] == address:
                return device
        return None

    @classmethod
    def watch_name_owner(
        cls,
        appeared_handler: Callable[[Gio.DBusConnection, str, str], None],
        vanished_handler: Callable[[Gio.DBusConnection, str], None],
    ) -> None:
        Gio.bus_watch_name(Gio.BusType.SYSTEM, cls.__bus_name, Gio.BusNameWatcherFlags.AUTO_START,
                           appeared_handler, vanished_handler)
```

</details>

cschramm, 2022-05-30:
Interactions are BlueZ <-> UPower, BlueZ <-> blueman, UPower <-> applet. As I said, if you see the battery in UPower the applet should pick it up and blueman does not interact with UPower at all.

The additions in your Manager.py basically just subscribe to additional D-Bus signals (subscribing does not globally consume them or anything) and emit log messages and GLib signals. Did you try resetting blueman to a clean state and checking the applet again? Did you even check if the battery is actually available in UPower? Otherwise there's nothing to expect from the applet.

Drugwash2, 2022-05-30:
Have not performed any further tests as I'm **still** waiting for the currently connected headphones' battery to change level. It's a long-run test, and I'm using precisely the same headphones that failed to show anything while Blueman was connected. Level used to be 60% at the beginning of the test, then dropped at 40% and holding. Playlist is only halfway done so all I need iscoffee and patience.

In the mean time, have you tried using the applet combined with the code as posted above?

cschramm, 2022-05-30:
I'm not using Cinnamon, so I don't know if spices is anything I could even run on my Desktop. :shrug:

Drugwash2, 2022-05-30:
Spices are just bits of javascript, optionally "spiced up" with Python bits and seasoned with a bit of JSON for settings. But yes, they can only run as is in the Cinnamon desktop as they depend on its libraries/implementations.
I'm sure you'll set up a virtual machine with some Mint Cinnamon and perform quite a few tests before my battery loses "weight". 😁

Drugwash2, 2022-05-30:
Ha! I must be cursed or something. Just when I went to the kitchen to grab a bite (bits, bit, bite - what's wrong with me?!) the headphones conveniently decided to drop from 40% to 10%.within a few secons, as if they waited for me to leave the room. Heard the beeps.

Back here started testing - everything seems to work without problems now, with both Panasonics. 😲 Blueman even displays the battery with both original and modified code - this was the very first time ever that I've seen the (ugly symbolic) battery icon in the devices list! Is it the planets or just me...? 🙄

Well, since I do need a little bit (bit?! again?!) of both "worlds" I'll keep using and tested both the applet and Blueman. Maybe I'll fix something for a change instead of breaking. 😇