blueman-project/blueman #1728: Tray icon is too small on hidpi screens

agorgl, 2022-05-30:
Current tray icon seems to have fixed height and does not scale to match other icons in the tray when used in a hidpi screen:
[image]

Distro: Arch
Panel: Polybar
Environment:
```
export GDK_SCALE=2
export GDK_DPI_SCALE=0.5
export QT_AUTO_SCREEN_SCALE_FACTOR=1
export QT_ENABLE_HIGHDPI_SCALING=1
export PLASMA_USE_QT_SCALING=1
```

agorgl, 2022-05-31:
But why all other icons are scaled properly? (a bit blurry from the upscaling, but still not tiny)

cschramm, 2022-05-31:
I quickly tested even with polybar and `GDK_SCALE` works perfectly fine to scale the icon in blueman-applet 2.2.4.

clefebvre, 2022-06-01:
> @clefebvre: Not sure what HiDPI issue you're mentioning in [#1633 (comment)](https://github.com/blueman-project/blueman/pull/1633#issuecomment-1138324790). Does it refer to the tray icon?

No, this refers to some of the icons used within the app.

That said I'm planning to improve two things in relation to tray which should address this issue:

- Port it from GtkStatusIcon to XappStatusIcon. This results in no changes in most DE (Xapp fallsback towards a GtkStatusIcon when no Xapp tray is found). In Xfce/MATE/Cinnamon this adds benefits such as sorting icons according to symbolic nature, HiDPI support, symbolic support, and most importantly the icon is then rendered by the tray itself, not by the app though some Gtk socket/plug (this is a nightmare to debug and it is responsible for many pixelated icons).
- Let users/distros switch between fullcolor and symbolic. I'm guessing what we're seeing here is a monochrome yet fullcolor pixel (png/jpg) icon coming from the icon theme. Symbolics aren't just symbolic they're also always scalable.

Can you tell us your icon theme @agorgl so we can make sure?

cschramm, 2022-06-01:
Yep, the theme only provides 22x22 and 24x24 (and 16x16 in case of the light version) panel icons. Not sure about the project status. Neither the website nor the repository say anything about the fact that it's archived. :shrug: