blueman-project/blueman #1731: Bluetooth can

gmernov, 2022-06-03:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2022-06-03:
Do I get it right, that it works fine on Ubuntu 20.04, thus newer versions of blueman and BlueZ. I'm not sure what kind of response you expect then. :sweat_smile:

gmernov, 2022-06-05:
Hi, i suppose dome kind of a debug advices :) Maybe i do something wrong. And yes, on my laptop Ubuntu it runs pretty smooth.

cschramm, 2022-06-05:
I guess the [bluez package from 20.04](https://packages.ubuntu.com/focal/amd64/bluez/download) should work on 18.04. You could try if that solves the problem. I noticed that while the frontends (blueman an bluetoothctl) report the "in progress" error, your bluetoothd logs only show timeouts (i.e. the device does not respond). Not sure how that fits together. As you see the same problem in bluetoothctl, the issue is probably not actually blueman-related.

gmernov, 2022-06-07:
Ok, i'll leave the issue on bluez repo. Maybe you're right, it's not directly blueman issue. Thanks. Good luck!

cschramm, 2022-06-07:
To fully outrule a blueman stake, you can try to replicate the problem in bluetoothctl without blueman running, i.e. especially no blueman-applet process.

If it's not related to the bluez package version, it could be e.g. firmware or the kernel. To be honest, I'm not sure if it makes much sense to dismantle such an old system. In the end you'll find a package that needs an update and major software updates are not part of the concept of Ubuntu releases.

gmernov, 2022-06-07:
Yeap. But the trouble is, that i have a setup on a NVIDIA's Jetson Nano platform. And i use Linux4Tegra distro. They do not have a latest Ubuntu release. The only supported version is 18.04. So, i have to be bound to this version.