blueman-project/blueman #1732: Feature Request: Disable auto-connect for a device

Merovius, 2022-06-04:
blueman: 2.2.3
BlueZ: 5.64
Distribution: Fedora
Desktop environment: i3

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

I have a pair of Bose QC 35. I have them paired with my phones and several of my Linux Desktops. When I turn my headphones on, I get a couple of prompts on all my Linux machines that my Headphones would like to connect to various services, with the option to "Allow always"/"Allow"/"Deny":

[image]

What's more, these dialogs don't disappear when the device disappears. So if I don't use a machine for a couple of days, I often have to click away a *lot* of these prompts which have accumulated. I would like to get rid of them. From what I can tell, my options are

1. "Always Allow" the connection. I don't want to do that, because my Headphones then connect to *both* my phone *and* one of my machines, which often ends up muting the audio from my phone, which is frustrating. Also, if there are usually 3-5 devices in range of my headphones, connecting to a specific one is hard, if they always accept any connection attempts.
2. Not have my headphones paired with my Linux machines. When I want to use them, pair them. This requires me to put my headphones in pairing mode, pair them, re-connect them, sometimes there's a hiccup with the available audio-profiles, requiring to repeat the process… it's a huge hassle. And given that I usually want to use them when I want to do a video call *right now*, that's not great.
3. Disable bluetooth (perhaps by removing the BT dongle, or by using rfkill). Which means I can't use BT for any *other* devices I use.

I don't like any of these options.

What I really want, is to be able to say "this particular paired device should *never* connect automatically and *only* connect when I manually do so in the device list". Effectively, I want an "Always Deny" button. Or maybe a setting that a particular device should behave as if I ignored/closed the popup - without showing it in the first place. I don't know enough about how BT connections and handshakes works to tell if these would be the same thing.

infirit, 2022-06-05:
The only way to do this is to block the device. I added this to the blueman ui in #1648 but this is not in a release yet. You can also use `bluetoothctl block btaddr`

Merovius, 2022-06-05:
Hm I thought I tried this and when I did, I could no longer connect manually either.

infirit, 2022-06-05:
Yes, you have to unblock it.

Merovius, 2022-06-05:
Hm okay, I just experimented a bit and it seems to work. Specifically, I have to set the Headphones as trusted and then I can use block/unblock to get the desired effect. I'll try this for a while, thanks.

cschramm, 2022-06-05:
In any case, I think closing the notification is a valid enhancement. We should geht a `Cancel` call from  bluetoothd for the authorization, so it should be possible. I'll check when I find some time.

Merovius, 2022-06-06:
> I just checked and found that we already handle the Cancel call correctly so that e.g. when I power off the device a confirmation request is for, the notification gets closed.

Hm is it possible that this is a recent-ish addition not yet in the version I use (2.2.3)? Because when I try it, it definitely doesn't close the notification when I turn off the headphones. Here's the output of `dbus-monitor` when I turn on my Headphones, the prompt appears, I wait for a few seconds and turn them off again. The prompt stayed:

[blueman.log](https://github.com/blueman-project/blueman/files/8844975/blueman.log)

From what I can tell, there is a `Cancel` call after I turned them off, followed by the `Connected` property changing to `false`. The prompt stays, however.

cschramm, 2022-06-06:
Nothing recent. It's in #1445, thus in all 2.2 releases.

The `Cancel` call should lead to blueman-applet closing the notification. Maybe your notification daemon does not support that. Which one is it?

There should be some output for the `Cancel` call from `blueman-applet --loglevel debug` (kill other instances of blueman-applet first).

Merovius, 2022-06-06:
> Maybe your notification daemon does not support that. Which one is it?

As far as I'm aware I'm not using a notification daemon. According to `xwininfo`/`xprop` the dialog window is created by `blueman-applet`.

> There should be some output for the Cancel call from blueman-applet --loglevel debug (kill other instances of blueman-applet first).

I will test that tomorrow.

cschramm, 2022-06-06:
Ah, yes, that's our fallback of no (sufficient) notification daemon is available. Anyway, I just tested that and it gets closed for me as well in response to the `Cancel` call.

cschramm, 2022-06-07:
When I connect to a BLE mouse and power it off while the authorization request is shown, blueman-applet shows the following log messages:

```
blueman-applet 11.57.46 INFO     BluezAgent:208 _on_request_confirmation: Agent.RequestConfirmation
blueman-applet 11.57.50 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_C1_64_FB_25_EF_9A {'ServicesResolved': False, 'Connected': False}
blueman-applet 11.57.50 INFO     BluezAgent:152 _on_cancel: Agent.Cancel
```

The first line leads to the notification being shown, the last line leads to its disappearance, independent of whether I use a notification daemon or the fallback dialog.

closed 2022-08-15 by github-actions[bot]

oleasteo, 2025-05-12:
While blocking a device is a mitigation of the issue, I wouldn't consider it a solution. It's unintuitive and badly supported by the UI. E.g. I cannot unblock & connect via the RecentConns plugin, nor disconnect & re-block via the DisconnectItems plugin.

There should be a per-device toggle to disable auto connection but still allow manual connections. This would be far better in UX.