blueman-project/blueman #1737: Preventing to disconnect after drop rfcomm port

dzbyr, 2022-06-06:
|**Package Versions:**| inf|
| ---| ---|
   | blueman| 2.2.4-1|
   | bluez| 5.64-2|
   | bluez-cups| 5.64-2|
   | bluez-hid2hci| 5.64-2|
   | bluez-libs| 5.64-2|
   | bluez-plugins| 5.64-2|
   | bluez-rfcomm| 5.63-2|
   | bluez-tools| 0.2.0-5|
   | bluez-utils| 5.64-2|
   | manjaro-bluetooth| 20210928-1|
   | pulseaudio-bluetooth| 15.0-4|

|Distribution:|  Info|
| ---| ---|
|Distributor ID:| ManjaroLinux|
|Description:| Manjaro Linux |
|Release:|	21.2.6|
|Codename:| Qonos|
|**Desktop environment**:|
|XDG_CURRENT_DESKTOP=|X-Cinnamon|
|VTE_VERSION=| 6800|

I connect serial port to BLE devices(HC-05 | HC-06) using **blueman** then takes data from **rfcomm0** via kind of serial plotting programs.
No problem so far, however whenever closing these kind programs (leaving rfcomm0 port) my Bluetooth connection drops.
I think the adapter kills the serial port connection somehow.
How can I prevent the automatically disconnecting from the device when drop the *rfcomm*?  I don't want disconnection.

Also, want to ensure whether this is deliberately choice for energy saving, etc. or not. 

* [x ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

[Here](https://gist.github.com/dzbyr/5a75c589d2fb6625669ae237dc1ffc2c) output of `blueman-applet --loglevel debug`

infirit, 2022-06-06:
Can you post the blueman-applet log for the whole rfcomm/serial session? The one posted does not show anything like this afaics.

cschramm, 2022-06-06:
It looks like you had another instance of blueman-applet still running so that the one with debug logging quit immediately.

Anyway, blueman runs a `blueman-rfcomm-watcher` process (as root) that keeps the device open and should prevent such a disconnect. Do you see such a process?

dzbyr, 2022-06-09:
@infirit Cathing what is different bit hard especially glancing 5 of them, luckily there is *vimdiff*

So, I cleaned each stage for detection error. Eliminated same lines.
In my method, I focused differentiation connection and drop moment but in case of ensure I added a step back and forth.
At the after a moment stage, I closed serial plotting script. 

This is not specific to the script, it also happens in different serial plotting apps. Such as [Serialplot](https://hackaday.io/project/5334-serialplot-realtime-plotting-software)

**hc-1_connected**
```
(blueman-applet:327502): Gtk-WARNING **: 18:48:02.869: Theme parsing error: gtk.css:7996:1: '/*' in comment block
INFO     Applet:54 _on_dbus_name_appeared: org.bluez :1.56
DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_98_D3_32_xxx {'Address': '98:D3:32:xxx', 'AddressType': 'public', 'Name': 'HC55_1', 'Alias': 'HC55_1', 'Class': 7936, 'Paired': True, 'Trusted': True, 'Blocked': False, 'LegacyPairing': False, 'Connected': True, 'UUIDs': ['00001101-0000-1000-8000-00805f9b34fb'], 'Adapter': '/org/bluez/hci0', 'ServicesResolved': False}
INFO     Functions:114 launch    : Gtk eventtime is 0, not using LaunchContext
INFO     Networking:71 set_nap   : set nap False
INFO     RecentConns:107 initialize: rebuilding menu
INFO     BluezAgent:56 register_agent: Register Agent
INFO     TransferService:234 _on_dbus_name_appeared: org.bluez.obex :1.73
INFO     ShowConnected:50 enumerate_connections: Found 1 existing connections
ERROR    AgentManager:20 on_register_failed: /org/bluez/obex/agent/blueman org.bluez.obex.Error.AlreadyExists Agent already exists
DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.DiscvManager.DiscvManager object at 0x7f0f58267100>
DEBUG    SerialManager:50 on_delete : Terminating any running scripts
```
**hc-1_connected_serial_bt_data_stream_on**

```(blueman-applet:329177): Gtk-WARNING **: 18:57:49.225: Theme parsing error: gtk.css:7996:1: '/*' in comment block
INFO     Applet:54 _on_dbus_name_appeared: org.bluez :1.56
DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_98_D3_32_xxx {'Address': '98:D3:32:xxx', 'AddressType': 'public', 'Name': 'HC55_1', 'Alias': 'HC55_1', 'Class': 7936, 'Paired': True, 'Trusted': True, 'Blocked': False, 'LegacyPairing': False, 'Connected': True, 'UUIDs': ['00001101-0000-1000-8000-00805f9b34fb'], 'Adapter': '/org/bluez/hci0', 'ServicesResolved': False}
INFO     Functions:114 launch    : Gtk eventtime is 0, not using LaunchContext
INFO     Networking:71 set_nap   : set nap False
INFO     RecentConns:107 initialize: rebuilding menu
INFO     BluezAgent:56 register_agent: Register Agent
INFO     TransferService:234 _on_dbus_name_appeared: org.bluez.obex :1.73
INFO     ShowConnected:50 enumerate_connections: Found 1 existing connections
ERROR    AgentManager:20 on_register_failed: /org/bluez/obex/agent/blueman org.bluez.obex.Error.AlreadyExists Agent already exists
DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.DiscvManager.DiscvManager object at 0x7f78c83fb100>
DEBUG    SerialManager:50 on_delete : Terminating any running scripts
```
**cat hc-1_connected_serial_bt_data_stream_on_disconnected**
```
(blueman-applet:329388): Gtk-WARNING **: 18:59:38.394: Theme parsing error: gtk.css:7996:1: '/*' in comment block
INFO     TransferService:234 _on_dbus_name_appeared: org.bluez.obex :1.73
INFO     Applet:54 _on_dbus_name_appeared: org.bluez :1.56
DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_98_D3_32_xxx {'Address': '98:D3:32:xxx', 'AddressType': 'public', 'Name': 'HC55_1', 'Alias': 'HC55_1', 'Class': 7936, 'Paired': True, 'Trusted': True, 'Blocked': False, 'LegacyPairing': False, 'Connected': True, 'UUIDs': ['00001101-0000-1000-8000-00805f9b34fb'], 'Adapter': '/org/bluez/hci0', 'ServicesResolved': False}
INFO     Functions:114 launch    : Gtk eventtime is 0, not using LaunchContext
INFO     Networking:71 set_nap   : set nap False
INFO     RecentConns:107 initialize: rebuilding menu
INFO     BluezAgent:56 register_agent: Register Agent
ERROR    AgentManager:20 on_register_failed: /org/bluez/obex/agent/blueman org.bluez.obex.Error.AlreadyExists Agent already exists
INFO     ShowConnected:50 enumerate_connections: Found 1 existing connections
DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.DiscvManager.DiscvManager object at 0x7f368437b100>
DEBUG    SerialManager:50 on_delete : Terminating any running scripts
```
**hc-1_connected_serial_bt_data_stream_on_disconnected_after_amoment**

```
(blueman-applet:329593): Gtk-WARNING **: 19:01:01.203: Theme parsing error: gtk.css:7996:1: '/*' in comment block
INFO     TransferService:234 _on_dbus_name_appeared: org.bluez.obex :1.73
INFO     Applet:54 _on_dbus_name_appeared: org.bluez :1.56
DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_98_D3_32_xxx {'Address': '98:D3:32:xxx',  'AddressType': 'public', 'Name': 'HC55_1', 'Alias': 'HC55_1', 'Class': 7936, 'Paired': True, 'Trusted': True, 'Blocked': False, 'LegacyPairing': False, 'Connected': True, 'UUIDs': ['00001101-0000-1000-8000-00805f9b34fb'], 'Adapter': '/org/bluez/hci0', 'ServicesResolved': False}
INFO     Functions:114 launch    : Gtk eventtime is 0, not using LaunchContext
INFO     Networking:71 set_nap   : set nap False
INFO     RecentConns:107 initialize: rebuilding menu
INFO     BluezAgent:56 register_agent: Register Agent
[... cut ...]
INFO     BluezAgent:56 register_agent: Register Agent
ERROR    AgentManager:20 on_register_failed: /org/bluez/obex/agent/blueman org.bluez.obex.Error.AlreadyExists Agent already exists
INFO     ShowConnected:50 enumerate_connections: Found 0 existing connections
DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.DiscvManager.DiscvManager object at 0x7fa061727100>
DEBUG    SerialManager:50 on_delete : Terminating any running scripts
```

@cschramm  I could see two of them while the connection active
[image]

infirit, 2022-06-09:
@dzbyr  Just post the full log please. Do something like `killall blueman-applet && blueman-applet --loglevel debug` and post all the lines it produces while connecting to the serial/rfcomm device and scripts being started and termintaed. We need to see the lines like below. 

```
blueman-applet 20.25.51 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_00_9E_C8_61_C7_7F {'Connected': True}
blueman-applet 20.25.52 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_00_9E_C8_61_C7_7F {'ServicesResolved': True}
blueman-applet 20.25.56 INFO     RecentConns:111 notify    : <Device.Device object at 0x7fc8991ec4c0 (blueman+bluez+Device+Device at 0x555a29f11160)> 00001101-0000-1000-8000-00805f9b34fb
rfcomm channel: 24
blueman-applet 20.25.56 INFO     SerialService:92 connect   : Starting rfcomm watcher as root
```

cschramm, 2022-06-09:
Again all outputs look like another blueman-applet process was already running, thus the error and immediate termination. A `killall blueman-applet` before running anything new should help.

dzbyr, 2022-06-11:
I didn't realize the debug season was going on. Sorry for not noticing the debug season for link purpose.

I tried two times same process to showing how repetetive this drop.
Disconnection time takes ~80-90 second after closing the serial ploting device.

```bash
killall blueman-applet && blueman-applet --loglevel debug

(blueman-applet:2643): Gtk-WARNING **: 11:39:52.128: Theme parsing error: gtk.css:4700:28: The style property GtkRange:slider-width is deprecated and shouldn't be used anymore. It will be removed in a future version

(blueman-applet:2643): Gtk-WARNING **: 11:39:52.128: Theme parsing error: gtk.css:4702:29: The style property GtkRange:trough-border is deprecated and shouldn't be used anymore. It will be removed in a future version

(blueman-applet:2643): Gtk-WARNING **: 11:39:52.133: Theme parsing error: gtk.css:6742:27: The style property GtkPaned:handle-size is deprecated and shouldn't be used anymore. It will be removed in a future version

(blueman-applet:2643): Gtk-WARNING **: 11:39:52.136: Theme parsing error: gtk.css:7996:1: '/*' in comment block
blueman-applet 11.39.52 INFO     PluginManager:85 load_plugin: ['DisconnectItems', 'ExitItem', 'AutoConnect', 'ShowConnected', 'Networking', 'Menu', 'KillSwitch', 'NetUsage', 'StandardItems', 'Battery', 'ConnectionNotifier', 'RecentConns', 'GameControllerWakelock', 'DBusService', 'DhcpClient', 'PowerManager', 'NMDUNSupport', 'TransferService', 'PPPSupport', 'StatusIcon', 'AppIndicator', 'AuthAgent', 'SerialManager', 'DiscvManager', 'NMPANSupport']
blueman-applet 11.39.52 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 11.39.52 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 11.39.52 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 11.39.52 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 11.39.52 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 11.39.52 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 11.39.52 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 11.39.52 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 11.39.52 INFO     Networking:35 load_nap_settings: Loading NAP settings
blueman-applet 11.39.52 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 11.39.52 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 11.39.52 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.PPPSupport.PPPSupport'>
blueman-applet 11.39.52 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.NetUsage.NetUsage'>
blueman-applet 11.39.52 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 11.39.52 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Battery.Battery'>
blueman-applet 11.39.52 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier'>
blueman-applet 11.39.52 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 11.39.52 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'>
blueman-applet 11.39.52 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DhcpClient.DhcpClient'>
blueman-applet 11.39.52 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 11.39.52 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.AppIndicator.AppIndicator'>
blueman-applet 11.39.52 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 11.39.52 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 11.39.52 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 11.39.52 INFO     KillSwitch:106 io_event  : killswitch registered 0
blueman-applet 11.39.52 INFO     KillSwitch:122 io_event  : State: False
blueman-applet 11.39.52 INFO     PowerManager:178 update_power_state: off True | foff False | on True | current state True | new state False
blueman-applet 11.39.52 INFO     PowerManager:181 update_power_state: Signalling False
blueman-applet 11.39.52 INFO     KillSwitch:85 _on_connman_vanished: net.connman vanished
blueman-applet 11.39.52 INFO     Applet:54 _on_dbus_name_appeared: org.bluez :1.21
blueman-applet 11.39.52 INFO     Functions:114 launch    : Gtk eventtime is 0, not using LaunchContext
blueman-applet 11.39.52 INFO     Networking:71 set_nap   : set nap False
blueman-applet 11.39.52 INFO     BluezAgent:56 register_agent: Register Agent
[... cut ...]
blueman-applet 11.44.10 INFO     RecentConns:107 initialize: rebuilding menu
rfcomm channel: 1
blueman-applet 11.44.15 INFO     SerialService:92 connect   : Starting rfcomm watcher as root
blueman-applet 11.44.15 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_98_D3_32_MM_MM_MM {'Connected': True}
blueman-applet 11.47.30 INFO     SerialManager:138 on_device_disconnect: Disconnecting /dev/rfcomm0
blueman-applet 11.47.30 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_98_D3_32_MM_MM_MM {'Connected': False}
```

infirit, 2022-06-28:
Did you see anything in `bluetoothd` debug output that suggest what is causing the disconnect?

dzbyr, 2022-06-28:
Yes, I'm trying to find a solution to that because that is looks like a common issue, but couldn't yet.
```bash
bluetoothd[36951]: src/adapter.c:connected_callback() hci0 device 98:D3:ccc:B5:19 connected eir_len 8
bluetoothd[36951]: src/adapter.c:dev_disconnected() Device 98:D3:ccc:B5:19 disconnected, reason 0
bluetoothd[36951]: src/adapter.c:adapter_remove_connection() 
bluetoothd[36951]: plugins/policy.c:disconnect_cb() reason 0
bluetoothd[36951]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr 98:D3:ccc:B5:19 type 0 status 0xe
bluetoothd[36951]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
bluetoothd[36951]: src/device.c:device_bonding_failed() status 14
bluetoothd[36951]: src/adapter.c:resume_discovery()
```
So there is a weird thing [ertm](https://wiki.archlinux.org/title/Gamepad#Old_workaround) that I couldn't prevent to set **ertm**. I think

```
sudo echo 1 > /sys/module/bluetooth/parameters/disable_ertm
bash: /sys/module/bluetooth/parameters/disable_ertm: Permission denied
```
or doing with another way that addin a bluetooth.conf file to the */etc/modprobe.d* 
```
options bluetooth disable_ertm=1
```

dzbyr, 2022-07-24:
Does the following *error* could be related with this drop?

```bash
 Connection Failed: File "/usr/lib/python3.10/site-packages/blueman/main/DbusService.py", line 124, in _handle_method_call
    method(*(args + (ok, lambda exception: self._return_dbus_error(invocation, exception))))
  File "/usr/lib/python3.10/site-packages/blueman/plugins/applet/DBusService.py", line 87, in connect_service
    service.connect(reply_handler=lambda port: ok(), error_handler=err)
  File "/usr/lib/python3.10/site-packages/blueman/services/meta/SerialService.py", line 93, in connect
    Mechanism().OpenRFCOMM('(d)', port_id)
  File "/usr/lib/python3.10/site-packages/gi/overrides/Gio.py", line 349, in __call__
    result = self.dbus_proxy.call_sync(self.method_name, arg_variant,
gi.repository.GLib.GError: g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name :1.100 was not provided by any .service files (2)
```

cschramm, 2022-07-24:
Oh hello. Yes, definitely. Looks like another manifestation of #1759.

Can you check `journalctl -u blueman-mechanism.service`? It looks like the D-Bus proxy invocation of blueman-mechanism succeeds but for some reason the process is already gone again when the client commits the immediately following method call.

dzbyr, 2022-07-24:
Looks bit different

```
-- Boot 18327e377c8244bdafd7eabac38bacc3 --
Ağu 26 04:49:10 axe-20aws2t12n systemd[1]: Starting Bluetooth management mechanism...
Ağu 26 04:49:10 axe-20aws2t12n blueman-mechanism[1583]: Unable to init server: Could not connect: Connection refused
Ağu 26 04:49:10 axe-20aws2t12n blueman-mechanism[1583]: Unable to init server: Could not connect: Connection refused
Ağu 26 04:49:10 axe-20aws2t12n systemd[1]: Started Bluetooth management mechanism.
Ağu 26 04:49:40 axe-20aws2t12n systemd[1]: blueman-mechanism.service: Deactivated successfully.
-- Boot fb19e776e4fe4cb7bc35b74cabfd7504 --
Ağu 26 05:30:48 axe-20aws2t12n systemd[1]: Starting Bluetooth management mechanism...
Ağu 26 05:30:48 axe-20aws2t12n blueman-mechanism[1456]: Unable to init server: Could not connect: Connection refused
Ağu 26 05:30:48 axe-20aws2t12n blueman-mechanism[1456]: Unable to init server: Could not connect: Connection refused
Ağu 26 05:30:48 axe-20aws2t12n systemd[1]: Started Bluetooth management mechanism.
```

infirit, 2022-07-24:
interesting. Looks like it sometimes starts, ignores the timer and exists immediately :thinking:

cschramm, 2022-07-24:
Hm, looks unsuspicious, yes. Do you have something from `blueman-mechanism.service` as well? Do you know the time of the error (There might be something in `~/.xsession-errors`)? Can you replicate it more or less reliably?

One thing that could hit us here - but chances are minimal and it's definitely not the cause in #1759 - is a small race condition: When the proxy object gets created, systemd might find the service running but it then exists before the method call arrives as it just ran into its timeout.

cschramm, 2022-07-27:
I'm a bit confused what the Gio proxy actually does. [Its documentation](https://docs.gtk.org/gio/class.DBusProxy.html) says

> The unique name owner of the proxy’s name is tracked

> calls are always sent to the current name owner

> if no name owner currently exists, then calls will be sent to the well-known name which may result in the message bus launching an owner (unless `G_DBUS_PROXY_FLAGS_DO_NOT_AUTO_START` is set)

so I would not expect the proxy to send calls to a name owner that's gone. :confused:

closed 2022-10-04 by github-actions[bot]