blueman-project/blueman #1743: 2.2.5 release tarball broken: configure has literal `PKG_CHECK_MODULES`

mgorny, 2022-06-10:
blueman: 2.2.5
BlueZ: 5.64
Distribution: Gentoo Linux
Desktop environment: Xfce

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

It seems that the 2.2.5 release tarball was generated on a system without pkg-config installed. It has literal:

    PKG_CHECK_MODULES(PYTHON, python-${PYTHON_VERSION})

which makes the script fail with:

    ./configure: line 14427: syntax error near unexpected token `PYTHON,'
    ./configure: line 14427: `PKG_CHECK_MODULES(PYTHON, python-${PYTHON_VERSION})'

infirit, 2022-06-10:
@mgorny I have updated the tarballs which should be ok as they pass distcheck.

cschramm, 2022-06-10:
How did that break? 🙄

mgorny, 2022-06-10:
Thanks, I can confirm that it works now. Also, seems to work nice with Python 3.11 :+1:.

infirit, 2022-06-12:
Thanks for letting us know so quickly  @mgorny .

closed 2022-06-12 by infirit