blueman-project/blueman #1759: Failed to apply network settings

alexandremsouza1, 2022-06-28:
blueman: 2.2.5
BlueZ:
Distribution: Ubuntu 22.04 LTS x86_64
Desktop environment: GNOME

I just updated and now I'm having this problem. My headphones are currently working fine. I'm just getting this dialog every time I go into plugins or when the system boots.

[image]

cschramm, 2022-06-28:
Ubuntu 22.04 does not provide blueman 2.2.5. Are you using the official 2.2.5-1 Debian package?

Please stop blueman-applet and start it manually with `blueman-applet --loglevel debug` to get proper output.

infirit, 2022-06-28:
From the message it most likely means you haven't installed the dbus service file correctly. Are you building blueman yourself?

alexandremsouza1, 2022-06-28:
no, I installed from apt-get `deb http://cz.archive.ubuntu.com/ubuntu kinetic main universe`

clefebvre, 2022-07-20:
This same error was reported in Mint 21 BETA (blueman 2.3): https://github.com/linuxmint/mint21-beta/issues/98

cschramm, 2022-07-20:
I'm able to provoke a similar exception with

```python
from blueman.main.DBusProxies import Mechanism
m = Mechanism()
```

```sh
sudo killall blueman-mechanism
```

```python
m.AnyMethod()
```

i.e. when our proxy object for the mechanism service survives the service itself. The Gio DBusProxy seems to resolve the well-known bus name `org.blueman.Mechanism` into a unique connection name and remembers just the latter, so that when it's gone everybody is lost.

That situation is totally unexpected, though. As the initialization of the proxy succeeds, we can conclude that blueman-mechanism started successfully and its name appeared on the bus. The method call then gets committed immediately, so that there's no chance that the service already reached its 30 seconds timeout. It seems like it's dying unexpectedly somewhere in `blueman.main.MechanismApplication.MechanismApplication.__init__`.

`journalctl -u blueman-mechanism.service` should be helpful.

cschramm, 2022-07-20:
Not really, `ReloadNetwork` gets called when the `Networking` applet plugin gets loaded. In main, #1564 actually avoids that by returning early from `_apply_nap_settings`.

clefebvre, 2022-07-21:
@cschramm https://github.com/blueman-project/blueman/commit/7c5f8756450a53aa6db4315428bbcb4091b5fed9 isn't included in our build. We're at 2.3 still, not 2.3.1.

cschramm, 2022-07-21:
@clefebvre: Yeah, sure. It's not included in 2.3.1 and just avoids that error "accidentally". It makes sense that it might show up in your version as well. The question remains why blueman-mechanism does not start / function as expected (and that's probably no better in main, it just would not bother the users).

infirit, 2022-07-23:
I opened #1795 to at least avoid people seeing this when NAP is not enabled. I do still want to see what happens with blueman-mechanism so please do share logs.

clefebvre, 2022-07-23:
We have some logs now in https://github.com/linuxmint/mint21-beta/issues/98. I do see "Loading NAP Settings" in there.

cschramm, 2022-07-24:
@clefebvre: Yeah, just as expected. blueman-mechanism's issue is the crucial part. We do not seem to have any data on that yet.

closed 2022-09-30 by github-actions[bot]