blueman-project/blueman #1761: How to display device battery percentage?

alexandremsouza1, 2022-06-28:
blueman: 2.2.5
BlueZ:
Distribution: Ubuntu 22.04 LTS x86_64
Desktop environment: GNOME

I'm not able to do it. And the version that this feature was available is 2.2.5?

https://drive.google.com/file/d/1UNK3gsZi53u0oBwSukks2kgpTHEbmVMd/view?usp=sharing

cschramm, 2022-06-28:
blueman 2.2 supports the LE Battery Service. Your device might not provide one but instead provide battery information via the audio profiles. That is supported by recent versions of audio servers, an experimental BlueZ API and blueman 2.3 (of which I just released beta1).

alexandremsouza1, 2022-06-28:
where do i download this beta 2.3?

cschramm, 2022-06-29:
https://github.com/blueman-project/blueman/releases/tag/2.3.beta1

A debian package is only at mentors yet. You can build a binary package if you like. In short, something like:
```
apt-get build-dep blueman
dget -u https://mentors.debian.net/debian/pool/main/b/blueman/blueman_2.3~beta1-1.dsc
cd blueman-2.3~beta1
debuild -b -uc -us
apt install ../blueman_2.3~beta1-1_amd64.deb
```

alexandremsouza1, 2022-06-29:
when trying to build an error occurs

[image]

alexandremsouza1, 2022-06-29:
passing the d parameter, this other error occurs

[image]

alexandremsouza1, 2022-06-29:
[image]

cschramm, 2022-06-29:
So you need to put (probably just uncomment) some deb-src URIs in your sources.list first.

alexandremsouza1, 2022-06-29:
could you point me to some from Ubuntu?

cschramm, 2022-06-30:
According to https://wiki.ubuntuusers.de/sources.list/#Jammy everything should be in place and, as I said, you just need to uncomment. Otherwise just duplicate the main deb URI as a deb-src URI.

Remember that 2.3 does not support audio profile battery levels on it's own. First step is to get batteries to show up in UPower or nothing will change in blueman. You can find plenty of hints here and on the Internets and it goes way beyond the non-blueman support I can provide here.

closed 2022-09-07 by github-actions[bot]