blueman-project/blueman #1777: 2.3 release has been generated with old, broken autoconf

mgorny, 2022-07-09:
blueman: 2.3
BlueZ: 5.64
Distribution: Gentoo Linux
Desktop environment: XFCE

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

The 2.3 release seems to have been made using autoconf 1.16.1 (2.2.5 used 1.16.5) that has broken support for Python newer than 3.9:

```
checking whether /usr/bin/python3.10 version is >= 3.6... yes
checking for /usr/bin/python3.10 version... 3.1
checking for /usr/bin/python3.10 platform... linux
checking for /usr/bin/python3.10 script directory... ${prefix}/lib/python3.10/site-packages
checking for /usr/bin/python3.10 extension module directory... ${exec_prefix}/lib/python3.10/site-packages
checking for PYTHON... no
configure: error: Package requirements (python-3.1) were not met:

Package 'python-3.1', required by 'virtual:world', not found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables PYTHON_CFLAGS
and PYTHON_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.

!!! Please attach the following file when seeking support:
!!! /tmp/portage/net-wireless/blueman-2.3/work/blueman-2.3/config.log
```

souravdas142, 2022-07-09:
please see : https://projects.gentoo.org/python/guide/porting.html#configure-no-package-python-3-1-found

cschramm, 2022-07-09:
The blueman project is now officially looking for a release manager. The only requirement is being able to do releases, a capability the maintainer does not seem to have. :see_no_evil:

I added 2b5f850079d07cfb0d3e0af9fee141cf2a9a4596 and updated the archives with it. Will make sure to add it to 2-3-stable and main as well...

infirit, 2022-07-09:
If I'm not mistaken we can force automake version with

```
diff --git a/configure.ac b/configure.ac
index ddaadcb4..b8ab8795 100644
--- a/configure.ac
+++ b/configure.ac
@@ -3,7 +3,7 @@ AC_PREREQ(2.61)
 AC_INIT([blueman], [2.3], [https://github.com/blueman-project/blueman/issues])
 AC_CONFIG_HEADERS(config.h)
 AC_CONFIG_MACRO_DIRS([m4])
-AM_INIT_AUTOMAKE([foreign dist-xz])
+AM_INIT_AUTOMAKE([1.16.4 foreign dist-xz])
 
 AM_MAINTAINER_MODE([enable])
 AC_PROG_CC
```

infirit, 2022-07-17:
We changed the release process and force a recent automake version. Let me know if you the latest 2.3.1 release has issues for you.

mgorny, 2022-07-17:
2.3.1 looks good, thanks!

closed 2022-07-18 by cschramm