blueman-project/blueman #1783: BlueMan 2.3 using old icons (theme)

birdie-github, 2022-07-15:
Blueman 2.3 comes with gray quite unpleasant [to me] icons.

Is it possible to revert to or use the icons found in version 2.2.5?

Thank you!

cschramm, 2022-07-15:
Unfortunately no. We decided to switch to symbolic icons (only) in blueman 2.3 and against making it an option (due to the obvious effects on maintainability).

birdie-github, 2022-07-15:
> We decided to switch to symbolic icons (only) in blueman 2.3 and against making it an option (due to the obvious effects on maintainability).

Grey icons are really hard to distinguish for many people :-(

OK, I'm going to check the code and try to recompile it manually with the old icons set once the XFCE panel crash has been taken care of.

closed 2022-07-15 by birdie-github