blueman-project/blueman #1791: Right mouse click popup menu is a complete mess

birdie-github, 2022-07-20:
blueman: 2.3.1-1.fc36 (an update from the updates-testing repo, `dnf upgrade --enablerepo=updates-testing blueman`)
BlueZ: bluez-5.64-1.fc36
Distribution: Fedora 36
Desktop environment: XFCE4

Check the attached screenshot:
[image]

This all looks utterly wrong:

Three times **Label Empty**
Send _Files to Device...
Recent _Connections
_Local Services

birdie-github, 2022-07-20:
Never mind, I've restarted it twice and it's now all good. Though I'm still not sure why it happened.

i-rinat, 2022-07-21:
This may be caused by a specific implementation of the system tray in XFCE4. I have the same issue, but it never really went away. Sometimes the (systray applet context) menu looks as usual, and sometimes it breaks into a single-level mess as in the screenshot above. I didn't notice if any specific actions cause that. Looks like it breaks at random and fixes itself at random.

cschramm, 2022-07-21:
Weird. Can you check which indicator implementation is used? `killall blueman-tray`, then `blueman-tray --loglevel info`. It's either the `StatusNotifierItem` (i.e. Xfce takes the data from D-Bus and renders the menu itself) or `GtkStatusIcon` (i.e. blueman-tray renders a GTK+ menu).

Did it start with 2.3 or does it also exist in 2.2?

> Looks like it breaks at random and fixes itself at random.

Is that from one execution to another or even between multiple invocations of the menu?

birdie-github, 2022-07-21:
It's always broken after a cold boot, then when I restart it, it looks OK.

```
blueman-tray --loglevel info
blueman-tray 20.32.19 INFO     Tray:46 _on_name_appeared: Using indicator "StatusNotifierItem"
```

infirit, 2022-07-21:
> It's always broken after a cold boot, then when I restart it, it looks OK.
> 
> I never had this problem with 2.2.x. It looks like a regression.

I setup a fresh F36 XFCE vm and I can't reproduce this. 

> 
> BTW, what's the relationship between `blueman-applet` and `blueman-tray`? Both are running here in Fedora 36.

Applet is the core that does most things. Tray is what provides the tray icon and by itself does very little than receive and display menu data from the applet.

cschramm, 2022-07-21:
Hm, ok, sounds like Xfce has some (another 🎉) problem with our StatusNotifierItem in 2.3 opposed to libappindicator's in 2.2... 😕

i-rinat, 2022-07-24:
Module "DisconnectItems" seems to helps to trigger the issue. It adds an item to the menu for each connected device and than somehow breaks the menu on xfce4-panel side. A possible workaround is to disable module "DisconnectItems".

I made some experiments, and while I don't have a working solution, here is what I found:

* `blueman-applet` source code seems to be okay. Tracing dbus messages shows that `blueman-applet` generates `LayoutUpdated` signals when its menu is updated and replies to `GetLayout` calls just fine;
* the source code of `libsystray.so` from `xfce4-panel` seems to be also okay. It doesn't parse dbus messages itself. Instead it relies on `libdbusmenu-gtk3` library. That library provides a replacement for GtkMenu widget which is expected to update automatically. But in this particular scenario I believe it fails. I'm not sure yet;
* `waybar` panel running under `sway` Wayland compositor has the same bug with the menu;
* `lxqt-panel` which uses separate implementation of DbusMenu, `libdbusmenu-qt5`, handles menu changes without the bug.

I don't think it's possible to do anything about the bug in the blueman source code.

cschramm, 2022-07-24:
Well, seems like `libdbusmenu-gtk3` has some weird issue with what we put on the bus then (`waybar` uses it as well). In 2.2 we did not handle the StatusNotifierItem ourselves but use `libappindicator` which also uses `libdbusmenu-gtk3`. Seems like we have some compatibility issue with `libdbusmenu-gtk3` now.

cschramm, 2022-07-25:
I just stumbled across this with Xfce on Debian testing and can actually reproduce it rather reliably - with some randomness or timing, though - by just turning Bluetooth on / off.

It seems like libdbusmenu does not update the types (standard or separator) of existing items under some circumstances, so that standard items that replaced separators show up with grey background and unprocessed text and cannot get activated while separators that replaced standard items appear as such with the default text "Label Empty". In the initial screenshot here, types are off by one, probably because the disconnect item got inserted. By turning Bluetooth on and off, you get the same effect due to the discoverability item.

Adding the default `type: standard` property to standard items does not fix this.

cschramm, 2022-07-25:
I failed to replicate the problem with the following script, simulating the Bluetooth toggle every second:

```python3
from blueman.main.indicators.StatusNotifierItem import StatusNotifierItem
from gi.repository import GLib

status = True

def flip():
  global status
  print("flip")
  item.set_menu([
    {"text": "A", "icon_name": "bluetooth-symbolic", "sensitive": True},
    {},
    {},
    {"text": "A", "icon_name": "blueman-send-symbolic", "sensitive": False},
    {},
    {"text": "A", "icon_name": "document-open-recent-symbolic", "sensitive": False},
    {},
    {"text": "A", "icon_name": "bluetooth-symbolic", "sensitive": False},
    {"text": "A", "icon_name": "bluetooth-symbolic", "sensitive": False},
    {"text": "A", "icon_name": "document-properties-symbolic", "sensitive": True},
    {},
    {"text": "A", "icon_name": "application-x-addon-symbolic", "sensitive": True},
    {"text": "A", "icon_name": "help-about-symbolic", "sensitive": True},
    {"text": "A", "icon_name": "application-exit-symbolic", "sensitive": True},
  ] if status else [
    {"text": "A", "icon_name": "bluetooth-disabled-symbolic", "sensitive": True},
    {"text": "A", "icon_name": "edit-find-symbolic", "sensitive": True},
    {},
    {},
    {"text": "A", "icon_name": "blueman-send-symbolic", "sensitive": True},
    {},
    {"text": "A", "icon_name": "document-open-recent-symbolic", "sensitive": False},
    {},
    {"text": "A", "icon_name": "bluetooth-symbolic", "sensitive": True},
    {"text": "A", "icon_name": "bluetooth-symbolic", "sensitive": True},
    {"text": "A", "icon_name": "document-properties-symbolic", "sensitive": True},
    {},
    {"text": "A", "icon_name": "application-x-addon-symbolic", "sensitive": True},
    {"text": "A", "icon_name": "help-about-symbolic", "sensitive": True},
    {"text": "A", "icon_name": "application-exit-symbolic", "sensitive": True},
    ]
  )
  status = not status
  return True

item = StatusNotifierItem("blueman-tray", None, None)
flip()

GLib.timeout_add(1000, flip)

GLib.MainLoop().run()
```

I can very much replicate it with `rfkill toggle bluetooth` in the same interval, though (when toggling is done with a script, I can just keep the menu open and watch the artifacts). Adding an icon change to the simulation does not trigger it either. :confused:

birdie-github, 2022-07-25:
Yeah, and there's another 2.3.x regression as well.

When you restart the XFCE panel (`xfce4-panel -r`), the blueman systray icon disappears. That wasn't the case with version 2.2.x :-(

cschramm, 2022-07-26:
@birdie-github: #1801

cschramm, 2022-07-27:
I also fail to replicate the problem with the exact same sequence (not timing, though) that I observed in blueman-tray when starting it and then blocking Bluetooth (~4 seconds after start) with the menu broken afterward.

<details>
<summary>Script</summary>

```python3
from blueman.main.indicators.StatusNotifierItem import StatusNotifierItem
from gi.repository import GLib

item = StatusNotifierItem("blueman-tray", None, None)

def init():
    item.set_tooltip_title("Bluetooth aktiviert")
    item.set_tooltip_text("")
    item.set_visibility(True)
    item.set_menu([{'text': '<b>Bluetooth _ausschalten</b>', 'markup': True, 'icon_name': 'bluetooth-disabled-symbolic', 'tooltip': 'Alle Adapter ausschalten', 'sensitive': True}, {'text': '_Erkennbar machen', 'markup': False, 'icon_name': 'edit-find-symbolic', 'tooltip': 'Den Standard-Adapter vorübergehend sichtbar machen', 'sensitive': True}, {'markup': False, 'sensitive': True}, {'markup': False, 'sensitive': True}, {'text': '_Dateien an Gerät senden…', 'markup': False, 'icon_name': 'blueman-send-symbolic', 'sensitive': True}, {'markup': False, 'sensitive': True}, {'text': 'Zuletzt benutzte _Verbindungen…', 'markup': False, 'icon_name': 'document-open-recent-symbolic', 'sensitive': False}, {'markup': False, 'sensitive': True}, {'text': '_Geräte…', 'markup': False, 'icon_name': 'bluetooth-symbolic', 'sensitive': True}, {'text': 'Adap_ter…', 'markup': False, 'icon_name': 'bluetooth-symbolic', 'sensitive': True}, {'text': '_Lokale Dienste…', 'markup': False, 'icon_name': 'document-properties-symbolic', 'sensitive': True}, {'markup': False, 'sensitive': True}, {'text': '_Erweiterungen', 'markup': False, 'icon_name': 'application-x-addon-symbolic', 'sensitive': True}, {'text': '_Hilfe', 'markup': False, 'icon_name': 'help-about-symbolic', 'sensitive': True}, {'text': 'B_eenden', 'markup': False, 'icon_name': 'application-exit-symbolic', 'sensitive': True}])

def flip():
    item.set_visibility(True)
    item.set_menu([{'text': '<b>Bluetooth _einschalten</b>', 'markup': True, 'icon_name': 'bluetooth-disabled-symbolic', 'tooltip': 'Alle Adapter ausschalten', 'sensitive': True}, {'text': '_Erkennbar machen', 'markup': False, 'icon_name': 'edit-find-symbolic', 'tooltip': 'Den Standard-Adapter vorübergehend sichtbar machen', 'sensitive': True}, {'markup': False, 'sensitive': True}, {'markup': False, 'sensitive': True}, {'text': '_Dateien an Gerät senden…', 'markup': False, 'icon_name': 'blueman-send-symbolic', 'sensitive': True}, {'markup': False, 'sensitive': True}, {'text': 'Zuletzt benutzte _Verbindungen…', 'markup': False, 'icon_name': 'document-open-recent-symbolic', 'sensitive': False}, {'markup': False, 'sensitive': True}, {'text': '_Geräte…', 'markup': False, 'icon_name': 'bluetooth-symbolic', 'sensitive': True}, {'text': 'Adap_ter…', 'markup': False, 'icon_name': 'bluetooth-symbolic', 'sensitive': True}, {'text': '_Lokale Dienste…', 'markup': False, 'icon_name': 'document-properties-symbolic', 'sensitive': True}, {'markup': False, 'sensitive': True}, {'text': '_Erweiterungen', 'markup': False, 'icon_name': 'application-x-addon-symbolic', 'sensitive': True}, {'text': '_Hilfe', 'markup': False, 'icon_name': 'help-about-symbolic', 'sensitive': True}, {'text': 'B_eenden', 'markup': False, 'icon_name': 'application-exit-symbolic', 'sensitive': True}])
    item.set_menu([{'text': '<b>Bluetooth _einschalten</b>', 'markup': True, 'icon_name': 'bluetooth-symbolic', 'tooltip': 'Alle Adapter ausschalten', 'sensitive': True}, {'text': '_Erkennbar machen', 'markup': False, 'icon_name': 'edit-find-symbolic', 'tooltip': 'Den Standard-Adapter vorübergehend sichtbar machen', 'sensitive': True}, {'markup': False, 'sensitive': True}, {'markup': False, 'sensitive': True}, {'text': '_Dateien an Gerät senden…', 'markup': False, 'icon_name': 'blueman-send-symbolic', 'sensitive': True}, {'markup': False, 'sensitive': True}, {'text': 'Zuletzt benutzte _Verbindungen…', 'markup': False, 'icon_name': 'document-open-recent-symbolic', 'sensitive': False}, {'markup': False, 'sensitive': True}, {'text': '_Geräte…', 'markup': False, 'icon_name': 'bluetooth-symbolic', 'sensitive': True}, {'text': 'Adap_ter…', 'markup': False, 'icon_name': 'bluetooth-symbolic', 'sensitive': True}, {'text': '_Lokale Dienste…', 'markup': False, 'icon_name': 'document-properties-symbolic', 'sensitive': True}, {'markup': False, 'sensitive': True}, {'text': '_Erweiterungen', 'markup': False, 'icon_name': 'application-x-addon-symbolic', 'sensitive': True}, {'text': '_Hilfe', 'markup': False, 'icon_name': 'help-about-symbolic', 'sensitive': True}, {'text': 'B_eenden', 'markup': False, 'icon_name': 'application-exit-symbolic', 'sensitive': True}])
    item.set_menu([{'text': '<b>Bluetooth _einschalten</b>', 'markup': True, 'icon_name': 'bluetooth-symbolic', 'tooltip': 'Alle Adapter einschalten', 'sensitive': True}, {'text': '_Erkennbar machen', 'markup': False, 'icon_name': 'edit-find-symbolic', 'tooltip': 'Den Standard-Adapter vorübergehend sichtbar machen', 'sensitive': True}, {'markup': False, 'sensitive': True}, {'markup': False, 'sensitive': True}, {'text': '_Dateien an Gerät senden…', 'markup': False, 'icon_name': 'blueman-send-symbolic', 'sensitive': True}, {'markup': False, 'sensitive': True}, {'text': 'Zuletzt benutzte _Verbindungen…', 'markup': False, 'icon_name': 'document-open-recent-symbolic', 'sensitive': False}, {'markup': False, 'sensitive': True}, {'text': '_Geräte…', 'markup': False, 'icon_name [... cut ...] icon_name': 'document-properties-symbolic', 'sensitive': True}, {'markup': False, 'sensitive': True}, {'text': '_Erweiterungen', 'markup': False, 'icon_name': 'application-x-addon-symbolic', 'sensitive': True}, {'text': '_Hilfe', 'markup': False, 'icon_name': 'help-about-symbolic', 'sensitive': True}, {'text': 'B_eenden', 'markup': False, 'icon_name': 'application-exit-symbolic', 'sensitive': True}])

GLib.timeout_add(100, init)
GLib.timeout_add(2000, flip)

GLib.MainLoop().run()

```
</details>

cschramm, 2022-07-27:
Ok, I tracked it down. I can replicate something very close rather reliably with:

```python3
from blueman.main.indicators.StatusNotifierItem import StatusNotifierItem
from gi.repository import GLib

item = StatusNotifierItem("blueman", None, None)

def step1():
    item.set_menu([
        {'text': '', 'markup': False, 'icon_name': 'bluetooth', 'sensitive': True},
    ])

def step2():
    item.set_menu([
        {'markup': False, 'sensitive': True},
    ])

GLib.timeout_add(100, step1)
GLib.timeout_add(103, step2)
GLib.MainLoop().run()
```

Timing is super important. The necessary delay between both steps depends on the complexity of the menu and probably on the system as well. When the issue hits, the entry has a grey background but the height of standard item. By increasing the second timeout you can reliably make things work, i.e. it becomes a proper separator.

What I saw happening in the `com.canonical.dbusmenu` interface is that when the issue shows up, the second `LayoutUpdated` signal gets sent and reacted to with a `GetLayout` call while a `GetGroupProperties` call is still running, so something in `libdbusmenu-glib`'s `LayoutUpdated` handling overlaps and gets things wrong. When the gap is too low, the issue does not show up, probably if processing did not reach the `GetGroupProperties` call at all yet or something similar.

I guess the best idea here would be to just make sure that `LayoutUpdated` does not get sent too quickly. using some random threshold and some queue or something. Meh...

closed 2022-07-27 by cschramm

TomBayne, 2022-08-26:
Doesn't seem like this issue is fixed :(

birdie-github, 2022-08-26:
> Doesn't seem like this issue is fixed :(

Which version are you using?

i-rinat, 2022-09-09:
Took a dive into libdbusmenu code. Turns out, it expects numeric IDs to be unique across menu updates. Code in the library reuses already built menu items if IDs match. Implementation in Blueman is not tracking IDs, and instead enumerates menu items from scratch.

cschramm, 2022-09-10:
@i-rinat: Yeah, assuming "unique" identifiers would be a good apology for not caring about that race condition. However, the identifier is - well - the identity of an item. Receiving the same identifier again means it is the same item. Receiving the same identifier with a different type means it is the same item and changed its type. I don't see why that case should not work. It's not forbidden to change items' type or something. However, that's exactly where the race condition takes effect when libdbusmenu receives rapid updates. If you do not want to call it a bug, then the assumption is that identifiers do not get re-used *and* items do not change their type (at least; there might be other problematic changes as well).

With #1847 we now track items with identifiers instead of enumerating them so that we do not re-use identifiers for replaced items. That's probably a workaround on its own here and libdbusmenu receivers would not need #1803 anymore. It doesn't hurt though, as it reduces the signal noise with just a couple of extra lines and a negligible delay.

PR #1803 (2022-07-27): Avoid rapid LayoutUpdated signals
libdbusmenu gets things wrong when signals arrive too frequently (but still with some gap). The interval is completely random - 10 ms turned out too fast as the issue still shows up (probably the gap between the signals is actually lower in those cases), 1 s sounds a lot too long to delay things (just to make broken receiver happy).

Closes #1791
commit 090b8526 (2022-07-27): Avoid rapid LayoutUpdated signals

libdbusmenu gets things wrong when signals arrive too frequently (but still with some gap). The interval is completely random - 10 ms turned out too fast as the issue still shows up (probably the gap between the signals is actually lower in those cases), 1 s sounds a lot too long to delay things (just to make broken receiver happy).

Closes #1791
commit 30c2529f (2022-08-01): Avoid rapid LayoutUpdated signals

libdbusmenu gets things wrong when signals arrive too frequently (but still with some gap). The interval is completely random - 10 ms turned out too fast as the issue still shows up (probably the gap between the signals is actually lower in those cases), 1 s sounds a lot too long to delay things (just to make broken receiver happy).

Closes #1791