blueman-project/blueman #1797: Systray icon hint could show a lot more info

birdie-github, 2022-07-23:
Right now in the hint I can only see this:

**Bluetooth active
1 Active connection**

A hint could contain a whole lot more information, e.g.:

**Bluetooth active
Device: Bluedio T headset
Receive Signal Strength: 50% (Optimal)
Link Quality: 70%
Transmit Power Level: 55% (Optimal)
Battery Level: 80%
Time connected: 5 hours, 10 minutes**

Thank you!

cschramm, 2022-09-08:
Two questions come to my mind:

- Is that data actually that useful?
- How do we handle multiple connections?

Link quality is gone, see #1841. At least for BLE RSSI and Tx power are often pointless default values, see #1642.

I'm not aware of anything that would persist the connection time.

That leaves us with just device name and possibly battery level. We could fit that into one line per connected device. Probably a good idea. :thinking:

birdie-github, 2022-09-08:
> Is that data actually that useful?

Yes, it is. If you're busy doing something else, you could leave it open, maybe someone will write a patch.

> How do we handle multiple connections?

1. Absolute most users normally have a single device connected. In fact multiple connections if I'm not mistaken are allowed only with BT 5.2 and not that many PC/laptop BT adapters.
2. Show all of them? ;-)
```
Bluetooth active

Device1: Bluedio T headset
Receive Signal Strength: 50% (Optimal)
Link Quality: 70%
Transmit Power Level: 55% (Optimal)
Battery Level: 80%
Time connected: 5 hours, 10 minutes

Device2: Something else
[the same as above]
```

cschramm, 2022-09-08:
> Absolute most users normally have a single device connected. In fact multiple connections if I'm not mistaken are allowed only with BT 5.2 and not that many PC/laptop BT adapters are at this level.

?? I'd say it's rather standard to have e.g. a mouse and headphones connected.

birdie-github, 2022-09-08:
I didn't know it was possible. My bad.

In fact I've just connected my smartphone to my PC and BT headphones are still working. Wow.

birdie-github, 2022-11-08:
It's relevant.

> This issue has been automatically marked as stale because it has not had recent activity. It will be closed if no further activity occurs. Thank you for your contributions.

birdie-github, 2023-01-09:
Not stale.

igorsantos07, 2023-03-16:
Showing the list of connected devices and their battery levels, when available, is definitely useful.

Other than that... Been using bluetooth for _many years_ and I never, _ever_, needed to know the connection strength. Coming from a user who didn't even know you could connect more than one device to the computer, I wouldn't trust that the connection strength is of any use lol

infirit, 2023-03-17:
> Other than that... Been using bluetooth for _many years_ and I never, _ever_, needed to know the connection strength. Coming from a user who didn't even know you could connect more than one device to the computer, I wouldn't trust that the connection strength is of any use lol

Maybe not to you but I've had connection issues and these bars showed the connection quality was abysmal. Also keep in mind blueman is about 15 years old and some of the design and features are based on what was needed back then. A few more years and it will be allowed to drink and drive :rofl: .

thiagonebuloni, 2023-08-02:
Hi, how is this update going? I found an issue marked as resolved in this battery subject, stating that in version 2.2 would be avaiable. But i'm using version 2.3.5 in Mint and it doesn't show battery life.
#1453

cschramm, 2023-08-03:
There is #1911. I still have to get back to it.

What's available is a battery level in blueman-manager (the devices window) and desktop notifications with the battery level. If those do not work for you, a battery level tooltip won't either.

commit c208cac0 (2023-09-03): Add battery levels to status icon tooltip

Closes #1797
closed 2023-09-03 by cschramm