blueman-project/blueman #1809: Improve the notification error message for a failed connect

birdie-github, 2022-08-01:
blueman: blueman-2.2.5-1.fc36.x86_64
BlueZ: bluez-5.65-1.fc36.x86_64
Distribution: Fedora 36
Desktop environment: XFCE

When trying to manually connect to a known BT device this message is shown which is not human readable. Maybe you could show something more comprehensible.

Thanks!

[image]

birdie-github, 2022-08-20:
@cschramm 

Error messages in `ManagerDeviceMenu.py` are not exactly human readable and kinda misleading as well. :(

E.g. `"Host is down", "br-connection-page-timeout"`? What's the "host"? Is it my laptop? Is it its Bluetooth adapter? Is it the device I'm trying to connect to?

Also, to me it makes sense to show a human readable code as well as a low level original code, so that it would be easier to track it down because it actually comes from https://github.com/bluez/bluez

So, it would be great if you showed something like:

Failed to connect
br-connection-page-timeout: Timeout initiating a connection with the remote device
br-connection-profile-unavailable: The remote device does not support the selected protocol

Maybe these messages should be reported to upstream instead considering your merge request, dunno.

cschramm, 2022-08-21:
Not sure what you mean. Where do you get to see `Host is down`?

birdie-github, 2022-08-21:
> Not sure what you mean. Where do you get to see `Host is down`?

https://github.com/cschramm/blueman/commit/f72c0cbbbdce1172fbf8b28df30ae677779b2d62

cschramm, 2022-08-21:
That's the code where we catch the bluetoothd / libc error message. The message we show in the UI for it is `Device did not respond`.

commit 1b02f9e5 (2022-11-17): Handle BlueZ 5.62+ error codes

Closes #1809
closed 2022-10-29 by github-actions[bot]