blueman-project/blueman #1814: Bluetooth down after last debian dist-upgrade

domix75, 2022-08-04:
blueman: 2.3.1-1
BlueZ: 5.64-2
Distribution: Debian GNU/Linux bookworm/sid
Desktop environment: i3-wm 4.20.1-1

* [ x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Since update yesterday bluetooth stopped working properly. Mouses are not recognised (I tried a second one). I can pair one (Logitech Laser Travel Mouse) but it connects and disconnects immediately, and the one I used before daily is not even seen when trying to pair. 

I tried a headset and it's the same, it pairs but that's all. Then it connects and disconnects immediately.

here is the log (last 500 including bluetoothd in syslog -- I have enabled bluez debug) : https://pastebin.com/MRLuqJxL

thanks

cschramm, 2022-08-04:
Does not seem blueman-related at all. First step is to find the update that caused it, so check the logs and try some downgrades. I don't know your update cycles but bluez 5.64-2 has been in the repositories for quite a while now, so my first guess would be the kernel.

domix75, 2022-08-04:
Thanks for your answer.
Actually I tried previous kernel (5.18.0-2) but it's not working either
In fact I was using this since mid June I think, so it was working
then probably the problem comes from bluez
cheers

Carl393, 2022-08-07:
i had a similar problem. I know this sounds strange. are you using bluetoothctl? if so you have to type yes for the pairing process... not simply Y... otherwise it doesn't work for some reason.

domix75, 2022-08-07:
> i had a similar problem. I know this sounds strange. are you using bluetoothctl? if so you have to type yes for the pairing process... not simply Y... otherwise it doesn't work for some reason.

thank you for your reply but I don't have this question when I use bluetoothctl in interactive mode
anyway it works again (see next post)
thx again

domix75, 2022-08-07:
finally i got it to work again but I'm not sure what was up
I purged bluez (which unstalled blueman too)
then I reinstalled it, and could pair using bluetoothclt while blueman-applet was not able too)
then I disconnected it using bluetoothctl, and started again the process with blueman-applet and this time it worked normally
there was probably something wrong in a config file, so the purge did some good. 
thanks for your anwsers and consideration

closed 2022-08-07 by domix75