blueman-project/blueman #1818: "segfault in libgtk-3.so.0.2404.29" when trying to start on wayland

Random-Chain, 2022-08-10:
blueman: 2.2.4-1 (latest available in Pop!_OS repo)
Distribution: Pop!_OS 22.04
Desktop environment: Wayland

When running on xorg, everything is fine.
When trying to run on Wayland, blueman crashes on start, or when trying to open the Devices window.

I can't find anything related to Blueman on Wayland or any related open issues, so I don't know if it's not supported or there's something else I need to configure.

Syslog:
```
kernel: [19281.824016] blueman-manager[110422]: segfault at 7ffe90518ff8 ip 00007f8c41e6c2ac sp 00007ffe90519000 error 6 in libgtk-3.so.0.2404.29[7f8c41d5e000+383000]
kernel: [19281.824026] Code: 1f 80 00 00 00 00 48 be 00 00 00 00 04 00 00 00 48 89 df e8 46 fa ff ff eb c9 0f 1f 40 00 41 56 41 55 49 89 f5 41 54 49 89 fc <55> 53 48 83 ec 30 48 8b 7f 18 64 48 8b 04 25 28 00 00 00 48 89 44
systemd[105468]: blueman-manager.service: Main process exited, code=dumped, status=11/SEGV
```

Errors I see in Blueman log:
```
blueman-applet 11.09.26 ERROR    PluginManager:90 load_plugin: Unable to load plugin module GameControllerWakelock
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/main/PluginManager.py", line 88, in load_plugin
    importlib.import_module(self.module_path.__name__ + f".{plugin}")
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/usr/lib/python3/dist-packages/blueman/plugins/applet/GameControllerWakelock.py", line 15, in <module>
    raise ImportError('This is not an X11 screen')
ImportError: This is not an X11 screen
```
And:
```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/main/DBusProxies.py", line 52, in call_finish
    proxy.call_finish(resp)
gi.repository.GLib.GError: g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.NoReply: Remote peer disconnected (4)
```
Full log here: https://pastebin.com/5F1dZecR

I also have the core dump that I can upload if it's relevant.

=-=-=

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

infirit, 2022-08-10:
Can you disable the gamecontrollerwakelock plugin in the plugin dialog? I don't have any segfaults myself when I run blueman under Wayland (KDE).

Random-Chain, 2022-08-12:
I don't see the gamecontrollerwakelock plugin in the list when running on wayland, so can't disable it. 
Just to check, I logged in to xorg, disabled all plugins, then logged in to wayland with everything off. It still crashes.

I'm also now seeing this assertion that I somehow missed before:
```
(blueman-applet:29822): Gdk-CRITICAL **: 12:43:02.561: gdk_wayland_window_set_dbus_properties_libgtk_only: assertion 'GDK_IS_WAYLAND_WINDOW (window)' failed
```

infirit, 2022-08-13:
Look like I'll have to check this in a VM. Not sure if I can do wayland in a VM...

infirit, 2022-08-13:
I can confirm the segfault but also this is very likely a problem with PopOS. The tranmission folks also ran into this, see https://github.com/transmission/transmission/issues/2572. I see `menutracker: action unity.-Help missing from the start` as well over and over untill it segfaults.

I'd suggest you raise this with the PopOS folks.

You can workaround this by launching blueman-manager with `GDK_BACKEND=x11 blueman-manager`.

closed 2022-10-22 by github-actions[bot]