blueman-project/blueman #1819: NAP connection fails

sm8ps, 2022-08-10:
blueman: 2.2.4
BlueZ: 5.64
Distribution: Ubuntu 22.04
Desktop environment: Budgie

* [x ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

This used to work about a year ago (likely on U21.04): connecting a certain device as NAP. With the recent upgrade (U22.04) it fails.

The (translated) error messages is: 
```
Connection failed: File "/usr/lib/python3/dist-packages/blueman/main/NetworkManager.py", line 80, in on_connection_activate
    self.active_connection = client.activate_connection_finish(result)
gi.repository.GLib.GError: nm-manager-error-quark: Connection 'DEVICE Network' is not available on device 00:16:53:46:CF:76 because profile is not compatible with device (devices bdaddr setting mismatches) (2)`
```

Here is a step-by-step account with actions and commands:
https://gist.github.com/sm8ps/db85216964b5b8bd9205534d70c1fbaa

Here the sylog as provided by journalctll:
https://gist.github.com/sm8ps/f43fcaa21796198fddb87b2842e17d83

infirit, 2022-08-11:
I'll have to look into this but it looks like the NetworkManager connection profile for this device contains an incorrect Bluetoot address.

Blueman will recreate the connection if you delete it drom NM. Let me know if this helps.

sm8ps, 2022-08-12:
Right on! Thanks @infirit, that was indeed the case. Myself, I would not have known where to start looking for the issue.
Deleting the connection file from /etc/NetworkManager/system-connections solved the issue.

PR #1826 (2022-08-15): Find NM PANU connection by bdaddr
...not device name.

Fixes #1819
commit b524334e (2022-08-15): Find NM PANU connection by bdaddr

...not device name.

Fixes #1819
commit ad068136 (2022-10-06): Find NM PANU connection by bdaddr

...not device name.

Fixes #1819
closed 2022-08-12 by sm8ps