blueman-project/blueman #1821: When connecting bluetooth headphones the sound doesn't switch immediately to bluetooth headphones

jsarkari, 2022-08-11:
blueman:
BlueZ:
Distribution: Linux Mint 21
Desktop environment: Cinnamon

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

infirit, 2022-08-11:
We have no control over which audio device is selected when a new device shows up. Blueman only takes care if the Bluetooth connection and then the pulseaudio or pipewire take over.

Both pulseaudio and pipewire have ways to control what device will get priority. Best is to ask on the mint forums how to configure this.

birdie-github, 2022-08-22:
This how it works in recent versions of PipeWire. Not sure PulseAudio is capable of that.

cschramm, 2022-08-23:
@birdie-github: Capable of what? Switching devices? It has `module-switch-on-connect` to switch the default device to a newly added one, yes.

closed 2022-10-31 by github-actions[bot]