blueman-project/blueman #1823: The 'daemon is not running' service is unhelpful and not even grammatical

LinuxOnTheDesktop, 2022-08-11:
blueman: `command not found`
BlueZ:  `command not found`
Distribution: Mint 21
Desktop environment: Cinnamon

* [X ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow. [But my point is about the error message itself.]

[image]

Surely, and at the least, the message should impart the name of the daemon. As the message stands at present, it is rather hard for the user - or at least if the user is new to the software - to troubleshoot the problem. I suppose one can _guess_ that the service is called 'bluez'. Is it?

Also: commas cannot conjoin clauses that could stand separately as independent sentences.

LinuxOnTheDesktop, 2022-08-12:
It is about the message. I _had_ a problem. I resolved the problem. That I resolved the problem was no thanks to the message.

cschramm, 2022-08-15:
I'm not sure about the expectations here.

What is a name of a daemon? Its binary is `/usr/sbin/bluetoothd`. Its init script is `/etc/init.d/bluetooth`. Its systemd service unit is `bluetooth.service`. Calling it "Bluetooth daemon" actually makes a lot of sense to me. An alternative could be `bluetoothd` but what would that be worth?

The situation when the message appears is that no D-Bus service provides the name `org.bluez`. I think it's pointless to inform the user about exactly that.

LinuxOnTheDesktop, 2022-08-15:
I had not anticipated that the daemon would have (in a sense) multiple names. Yet, how about amending `Bluetooth daemon was not started` to `'Bluetooth' daemon was not started`? With that change, the user at least gets a firm steer that the problem owes to something that (in the case of the init and systemd entities) is called, specifically, 'bluetooth'.

One might take the opportunity too to improve the English in the message - both in the sentence that mentions the daemon and in the preceding sentence. I suggest that the text as a whole be amended to the following.

> The blueman-manager cannot continue, because the bluez daemon is not running. It is likely that no bluetooth adaptor was detected or else that the 'bluetooth' daemon was not started.

cschramm, 2022-08-30:
I'm totally fine with that (I just don't like the article in front of `blueman-manager`) but actually have additional concerns:

- I don't really think it's worth mentioning that blueman-manager cannot continue.
- It's weird to call the same thing `the bluez daemon` in one sentence and `the 'bluetooth' daemon` in the other one. It makes sense to mention BlueZ but that's already done in the title.
- The fact that the daemon is not running and that it was not started is basically the same to me.

(A little background on the "no adapter" hint: A reason for the `bluetooth.service` unit to not auto-start is missing Bluetooth support in the kernel, typically as the module is not loaded. In a properly configured "standard" system that can only happen if no (supported) hardware is present because otherwise the kernel would load the necessary modules automagically.)

How about something like just

> The 'bluetooth' daemon does not seem to be running, possibly because no Bluetooth adapter was detected.

?

LuciusofLegend, 2023-12-11:
My idea, something like:

The 'bluetooth' daemon does not seem to be running.  Try manually starting the daemon with *command to start it*, and check that you have a supported bluetooth adapter.

Also, it seems like it could be a nice feature to have a link to the troubleshooting page on all error messages.  That would make it much more straightforward.

cschramm, 2023-12-12:
There is no ubiquitous `command to start it` unless you e.g. assume systemd with a standard configuration.