blueman-project/blueman #1824: Goldtouch wireless keyboard not paring/showing on search - previously worked on Linux 20.3

8x1053, 2022-08-12:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

8x1053, 2022-08-12:
System:
  Kernel: 5.15.0-41-generic x86_64 bits: 64 compiler: gcc v: 11.2.0 Desktop: Cinnamon 5.4.8
    tk: GTK 3.24.33 wm: Mutter dm: LightDM Distro: Linux Mint 21 Vanessa base: Ubuntu 22.04 jammy
Machine:
  Type: Laptop System: HP product: HP ENVY x360 Convertible v: Type1ProductConfigId
    serial: <superuser required> Chassis: type: 10 serial: <superuser required>
  Mobo: HP model: 8420 v: 58.67 serial: <superuser required> UEFI: Insyde v: F.51
    date: 10/17/2017
Battery:
  ID-1: BAT0 charge: 25.7 Wh (100.0%) condition: 25.7/25.7 Wh (100.0%) volts: 16.9 min: 15.4
    model: 313-54-41-A MB04055XL serial: <filter> status: Full
CPU:
  Info: quad core model: Intel Core i7-8550U bits: 64 type: MT MCP arch: Coffee Lake rev: A cache:
    L1: 256 KiB L2: 1024 KiB L3: 8 MiB
  Speed (MHz): avg: 800 min/max: 400/4000 cores: 1: 800 2: 800 3: 800 4: 800 5: 800 6: 800
    7: 800 8: 800 bogomips: 31999
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3
Graphics:
  Device-1: Intel UHD Graphics 620 vendor: Hewlett-Packard driver: i915 v: kernel ports:
    active: HDMI-A-1 off: eDP-1 empty: none bus-ID: 00:02.0 chip-ID: 8086:5917
  Device-2: Sunplus Innovation HP Wide Vision HD type: USB driver: uvcvideo bus-ID: 1-4:4
    chip-ID: 1bcf:2c83
  Device-3: Logitech BRIO Ultra HD Webcam type: USB
    driver: hid-generic,snd-usb-audio,usbhid,uvcvideo bus-ID: 2-2.2:3 chip-ID: 046d:085e
  Display: x11 server: X.Org v: 1.21.1.3 driver: X: loaded: modesetting unloaded: fbdev,vesa
    gpu: i915 display-ID: :0 screens: 1
  Screen-1: 0 s-res: 1920x1080 s-dpi: 96
  Monitor-1: HDMI-1 mapped: HDMI-A-1 model: AOC 32G2WG8 res: 1920x1080 dpi: 70
    diag: 801mm (31.5")
  OpenGL: renderer: Mesa Intel UHD Graphics 620 (KBL GT2) v: 4.6 Mesa 22.0.5 direct render: Yes
Audio:
  Device-1: Intel Sunrise Point-LP HD Audio vendor: Hewlett-Packard driver: snd_hda_intel
    v: kernel bus-ID: 00:1f.3 chip-ID: 8086:9d71
  Device-2: Logitech BRIO Ultra HD Webcam type: USB
    driver: hid-generic,snd-usb-audio,usbhid,uvcvideo bus-ID: 2-2.2:3 chip-ID: 046d:085e
  Sound Server-1: ALSA v: k5.15.0-41-generic running: yes
  Sound Server-2: PulseAudio v: 15.99.1 running: yes
  Sound Server-3: PipeWire v: 0.3.48 running: yes
Network:
  Device-1: Intel Wireless 7265 driver: iwlwifi v: kernel pcie: speed: 2.5 GT/s lanes: 1
    bus-ID: 01:00.0 chip-ID: 8086:095a
  IF: wlo1 state: up mac: <filter>
Bluetooth:
  Device-1: Intel Bluetooth wireless interface type: USB driver: btusb v: 0.8 bus-ID: 1-6:6
    chip-ID: 8087:0a2a
  Report: hciconfig ID: hci0 rfk-id: 0 state: up address: <filter> bt-v: 2.1 lmp-v: 4.2
    sub-v: 1000
Drives:
  Local Storage: total: 715.97 GiB used: 94.92 GiB (13.3%)
  ID-1: /dev/mmcblk0 model: GF8S5 size: 477.5 GiB serial: <filter>
  ID-2: /dev/nvme0n1 vendor: Toshiba model: N/A size: 238.47 GiB speed: 31.6 Gb/s lanes: 4
    serial: <filter> temp: 26.9 C
Partition:
  ID-1: / size: 233.18 GiB used: 18.64 GiB (8.0%) fs: ext4 dev: /dev/nvme0n1p3
  ID-2: /boot/efi size: 512 MiB used: 5.2 MiB (1.0%) fs: vfat dev: /dev/nvme0n1p2
Swap:
  ID-1: swap-1 type: file size: 2 GiB used: 287.8 MiB (14.1%) priority: -2 file: /swapfile
Sensors:
  System Temperatures: cpu: 40.0 C pch: 33.5 C mobo: 33.0 C
  Fan Speeds (RPM): N/A
Repos:
  Packages: 2371 apt: 2364 flatpak: 7
  No active apt repos in: /etc/apt/sources.list
  Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list
    1: deb http: //packages.linuxmint.com vanessa main upstream import backport
    2: deb http: //archive.ubuntu.com/ubuntu jammy main restricted universe multiverse
    3: deb http: //archive.ubuntu.com/ubuntu jammy-updates main restricted universe multiverse
    4: deb http: //archive.ubuntu.com/ubuntu jammy-backports main restricted universe multiverse
    5: deb http: //security.ubuntu.com/ubuntu/ jammy-security main restricted universe multiverse
  Active apt repos in: /etc/apt/sources.list.d/signal-xenial.list
    1: deb [arch=amd64 signed-by=/usr/share/keyrings/signal-desktop-keyring.gpg] https: //updates.signal.org/desktop/apt xenial main
    2: deb [arch=amd64 signed-by=/usr/share/keyrings/signal-desktop-keyring.gpg] https: //updates.signal.org/desktop/apt xenial main
    3: deb [arch=amd64 signed-by=/usr/share/keyrings/signal-desktop-keyring.gpg] https: //updates.signal.org/desktop/apt xenial main
  Active apt repos in: /etc/apt/sources.list.d/winehq-impish.sources
    1: deb [arch=amd64 i386] https: //dl.winehq.org/wine-builds/ubuntu impish main
Info:
  Processes: 289 Uptime: 7h 27m Memory: 7.64 GiB used: 3.14 GiB (41.1%) Init: systemd v: 249
  runlevel: 5 Compilers: gcc: 11.2.0 alt: 11 Client: Cinnamon v: 5.4.8 inxi: 3.3.13

8x1053, 2022-08-12:
`blueman-applet 22.34.23 INFO     PluginManager:87 load_plugin: ['NMDUNSupport', 'StandardItems', 'AutoConnect', 'PowerManager', 'StatusNotifierItem', 'Networking', 'Menu', 'ExitItem', 'ShowConnected', 'StatusIcon', 'RecentConns', 'SerialManager', 'AuthAgent', 'ConnectionNotifier', 'PPPSupport', 'DhcpClient', 'DisconnectItems', 'DiscvManager', 'NMPANSupport', 'DBusService', 'KillSwitch', 'TransferService', 'GameControllerWakelock', 'NetUsage']
blueman-applet 22.34.23 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 22.34.23 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 22.34.23 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Address': '10:F0:05:F5:BF:E4', 'AddressType': 'public', 'Name': 'Wisteria', 'Alias': 'Wisteria', 'Class': 8126732, 'Powered': True, 'Discoverable': True, 'DiscoverableTimeout': 0, 'Pairable': False, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['00001133-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00001104-0000-1000-8000-00805f9b34fb', '00005005-0000-1000-8000-0002ee000001', '00001108-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '0000111f-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0540', 'Roles': ['central', 'peripheral']}
blueman-applet 22.34.23 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 22.34.23 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 22.34.23 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 22.34.23 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 22.34.23 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusNotifierItem.StatusNotifierItem'>
blueman-applet 22.34.23 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 22.34.23 INFO     Networking:35 load_nap_settings: Loading NAP settings
blueman-applet 22.34.23 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 22.34.23 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 22.34.23 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 22.34.23 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 22.34.23 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 22.34.23 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier'>
blueman-applet 22.34.23 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.PPPSupport.PPPSupport'>
blueman-applet 22.34.23 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DhcpClient.DhcpClient'>
blueman-applet 22.34.23 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 22.34.23 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 22.34.23 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 22.34.23 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 22.34.23 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'>
blueman-applet 22.34.23 INFO     KillSwitch:106 io_event  : killswitch registered 0
blueman-applet 22.34.23 INFO     KillSwitch:122 io_event  : State: True
blueman-applet 22.34.23 INFO     PowerManager:179 update_power_state: off False | foff False | on True | current state True | new state True
blueman-applet 22.34.23 INFO     KillSwitch:85 _on_connman_vanished: net.connman vanished
blueman-applet 22.34.23 INFO     TransferService:234 _on_dbus_name_appeared: org.bluez.obex :1.55
blueman-applet 22.34.23 INFO     Applet:63 _on_dbus_name_appeared: org.bluez :1.11
blueman-applet 22.34.23 INFO     Functions:114 launch    : Gtk eventtime is 0, not using LaunchContext
blueman-applet 22.34.23 INFO     Networking:71 set_nap   : set nap False
blueman-applet 22.34.23 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_AC_B1_EE_6D_F2_61 {'Address': 'AC:B1:EE:6D:F2:61', 'AddressType': 'public', 'Name': 'KMOUK HSB001', 'Alias': 'KMOUK HSB001', 'Class': 2360324, 'Icon': 'audio-headset', 'Paired': True, 'Trusted': True, 'Blocked': False, 'LegacyPairing': False, 'Con [... cut ...]
blueman-applet 22.34.50 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.DBusService.DBusService object at 0x7f0a85debdc0>
blueman-applet 22.34.50 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <StatusIcon.StatusIcon object at 0x7f0a90075100 (blueman+plugins+applet+StatusIcon+StatusIcon at 0x558f41db3ae0)>
blueman-applet 22.34.50 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.Menu.Menu object at 0x7f0a8ec4da20>
`

8x1053, 2022-08-12:
When I try to get this see the keyboard (which, again, was working under 20.3 Linux Mint, I get this...`blueman-applet 22.38.21 INFO     Applet:100 on_device_removed: Device removed /org/bluez/hci0/dev_D6_CC_4D_0F_6B_D4
blueman-applet 22.38.40 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Discovering': True}
blueman-applet 22.38.40 DEBUG    DiscvManager:92 on_adapter_property_changed: prop Discovering True
blueman-applet 22.38.40 DEBUG    Manager:52 _on_object_added: Device1: /org/bluez/hci0/dev_D6_CC_4D_0F_6B_D4
blueman-applet 22.38.40 INFO     Applet:95 on_device_created: Device created /org/bluez/hci0/dev_D6_CC_4D_0F_6B_D4
blueman-applet 22.38.41 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_CC_BE_39_95_25_4B {'RSSI': -73}
blueman-applet 22.38.50 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Discovering': False}
blueman-applet 22.38.50 DEBUG    DiscvManager:92 on_adapter_property_changed: prop Discovering False
`

8x1053, 2022-08-12:
The UI doesn't show the keyboard at anytime during the paring attempt.

8x1053, 2022-08-12:
Trying this...

https://forums.linuxmint.com/viewtopic.php?t=271905

infirit, 2022-08-13:
We hide unnamed devices by default, you can disable this in the view menu.

If that does not help post the log for blueman-applet from start till after the search ends. Do make sure to kill blueman-applet `killall blueman-applet` and then run it in a terminal.

8x1053, 2022-08-13:
This worked and I am a bit sheepish.  Thanks for the get back.  I should have checked the toolbar before digging the community.

Thanks! One less cord to deal with and ergonomic goodness released.

closed 2022-08-13 by 8x1053

cschramm, 2022-08-14:
So you were actually able to pair the device even though it did not expose a name?

@infirit: If I got you right in https://github.com/blueman-project/blueman/pull/1258#issuecomment-629785356 one would not be able to use it at all with Android and probably Windows. That sounds super weird. :thinking:

8x1053, 2022-08-14:
@cschramm It is not all joy. 

The root problem was not being able to see the unnamed device. By unchecking the View/Hide unnamed devices - the MAC Address of the keyboard became visible, and I was able to pair/trust and connect with it. Once I did, it became a named device.

[image]

The lack of joy comes from the ongoing behavior.  The keyboard is not repairing (ie it paired once, is visible now... but will not re-pair.) I have since gone back to a wired keyboard.  

Whatever the issue, it is not the same as the original, so I closed this issue.

This is the log when I try to pair my keyboard now. I asks me for the pin (which it did under the 20.3 install - initially as well - but the wireless keyboard is not able to input. I have to use either the virtual keyboard or a wired one.) Even then the wireless one won't work.

I'm learning a ton about Bluetooth, but not getting far.  I am putting about a 30 mins a day into this, trying to get it to work.  I will repost to community if I can't find a solution on my own... but the root problem here is resolved - I can see my device now, it worked/paired when I first tried it (then it went south... ) 

Joy.

Here is the entire log from a killall to a pairing pin request that I input on my wired keyboard.  Still trying to figure this out.

Note this worked without issue on Mint 20.3.

`blueman-applet 06.52.47 INFO     PluginManager:87 load_plugin: ['NMDUNSupport', 'StandardItems', 'AutoConnect', 'PowerManager', 'StatusNotifierItem', 'Networking', 'Menu', 'ExitItem', 'ShowConnected', 'StatusIcon', 'RecentConns', 'SerialManager', 'AuthAgent', 'ConnectionNotifier', 'PPPSupport', 'DhcpClient', 'DisconnectItems', 'DiscvManager', 'NMPANSupport', 'DBusService', 'KillSwitch', 'TransferService', 'GameControllerWakelock', 'NetUsage']
blueman-applet 06.52.47 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 06.52.47 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 06.52.47 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Address': '10:F0:05:F5:BF:E4', 'AddressType': 'public', 'Name': 'Wisteria', 'Alias': 'Wisteria', 'Class': 8126732, 'Powered': True, 'Discoverable': True, 'DiscoverableTimeout': 0, 'Pairable': True, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['00001133-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00001104-0000-1000-8000-00805f9b34fb', '00005005-0000-1000-8000-0002ee000001', '00001108-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '0000111f-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb', '00001112-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0540', 'Roles': ['central', 'peripheral']}
blueman-applet 06.52.47 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 06.52.47 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 06.52.47 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 06.52.47 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 06.52.47 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusNotifierItem.StatusNotifierItem'>
blueman-applet 06.52.47 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 06.52.47 INFO     Networking:35 load_nap_settings: Loading NAP settings
blueman-applet 06.52.47 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 06.52.47 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 06.52.47 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 06.52.47 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 06.52.47 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 06.52.47 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier'>
blueman-applet 06.52.47 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.PPPSupport.PPPSupport'>
blueman-applet 06.52.47 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DhcpClient.DhcpClient'>
blueman-applet 06.52.47 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 06.52.47 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
[... cut ...]
blueman-applet 06.53.39 DEBUG    Notification:250 do_g_signal: The notification was dismissed by the user.
blueman-applet 06.53.59 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_08_B7_38_24_31_FB {'Connected': False}
blueman-applet 06.53.59 INFO     BluezAgent:152 _on_cancel: Agent.Cancel
blueman-applet 06.54.38 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_08_B7_38_24_31_FB {'Class': 1344, 'Icon': 'input-keyboard', 'Connected': True}
blueman-applet 06.54.38 INFO     BluezAgent:164 _on_request_pin_code: Agent.RequestPinCode
`

cschramm, 2022-08-14:
> It asks me for the pin (which it did under the 20.3 install - initially as well - but the wireless keyboard is not able to input. I have to use either the virtual keyboard or a wired one.) Even then the wireless one won't work.

The idea there is that you enter the displayed code into the device involved in the pairing process for confirmation. Typing it on your PC that is displaying the code is pointless. Check the device's manual for instructions but typically you just type the sequence and then often press enter.

> the device pairs with my Android, NovaAirC Ereader, Windows 10 Machine and IPad without issue as well.

As you mention that: How did the device show up on those systems before the initial pairing? With its Bluetooth address instead of a name? (Opposed to your initial issue the device shows up with its name unpaired in the log but I guess that's just due to the name being cache by BlueZ.)

8x1053, 2022-08-14:
@cschramm 

> The idea there is that you enter the displayed code into the device

The wireless keyboard is non functional though paired - so that doesn't work.  The manual states that the code can be whatever the user wants - but I could not input any value or return even - except with the virtual keyboard or wired keyboard.

> How did the device show up on those systems

By name in all cases, including previously under LInux Mint 20 install.

I'm thinking about doing a fresh install.  I haven't done a ton of work on the machine yet due to my default bluetooth keyboard roadblock. I have install GIMP and a few odd things.

cschramm, 2022-08-14:
> The wireless keyboard is non functional though paired

I'm not able to follow. Your log shows that it (device 08:B7:38:24:31:FB) is not paired.

infirit, 2022-08-15:
> So you were actually able to pair the device even though it did not expose a name?
> 
> @infirit: If I got you right in [#1258 (comment)](https://github.com/blueman-project/blueman/pull/1258#issuecomment-629785356) one would not be able to use it at all with Android and probably Windows. That sounds super weird. 🤔

It wouldn't if it actually doesn't have a name. But I am getting the suspicion that either windows/android try harder to get a name or BlueZ is just failing to populate the name in newer/some versions (we had one instance it got populated a second or so later).

I am contemplating to flip the default though as it causing more confusion rn.

@8x1053 if you are willing, can you test something for me. Does the keyboard show up with a name on your phone when it's not paired (removed from the list)? Doesn't matter really which one but I only have android devices.

cschramm, 2022-08-15:
> > So you were actually able to pair the device even though it did not expose a name?
> > @infirit: If I got you right in [#1258 (comment)](https://github.com/blueman-project/blueman/pull/1258#issuecomment-629785356) one would not be able to use it at all with Android and probably Windows. That sounds super weird. thinking
> 
> It wouldn't if it actually doesn't have a name. But I am getting the suspicion that either windows/android try harder to get a name or BlueZ is just failing to populate the name in newer/some versions (we had one instance it got populated a second or so later).

A delay should not be a problem as we unset `no_name` if we get an update for the `Name` property in `ManagerDeviceList.row_update_event`.

> I am contemplating to flip the default though as it causing more confusion rn.

Is it? This is the first case where generally hiding unnamed devices might be a problem, no?

In any case, we could think of alternative solutions like just forcing unnamed devices to the end of the list. We did have constant complaints that the list was too cluttered with all the unnamed devices, so just bringing them back like it was does not sound like a good idea to me. 

> @8x1053 if you are willing, can you test something for me. Does the keyboard show up with a name on your phone when it's not paired (removed from the list)? Doesn't matter really which one but I only have android devices.

It does in https://github.com/blueman-project/blueman/issues/1824#issuecomment-1212746332 but that's totally expected as the name is still cached.

```sh
$ sudo head -2 /var/lib/bluetooth/10:F0:05:F5:BF:E4/cache/08:B7:38:24:31:FB
[General]
Name=Goldtouch Bluetooth Keyboard
```
(I totally made this up but I'd bet on it :wink:)

cschramm, 2022-12-04:
@8x1053: In case you're able to replicate the initial issue, could you please try the following for us:

- Start with the device removed from the system.
- Make the device discoverable.
- Open blueman-manager.
- Run `bluetoothctl scan on` in a terminal.

Wait for a while (>> 1 min). Does the device show up (in case you have unnamed devices hidden) or get a name (in case you have them displayed)?

8x1053, 2022-12-04:
Unfortunately after re-install I can't duplicate the issue.