blueman-project/blueman #1827: Lost functionality on my bluetooth devices post pairing. Fresh install LInux 21 Cinnamon

8x1053, 2022-08-14:
blueman:2.3.1
BlueZ:5.64
Distribution:Linux Mint 21 Cinnamon
Desktop environment:Cinnamon
Kernel:5.15.0-41-generic

head -1 $(which blueman-applet)
#! /usr/bin/python3

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

I just created a fresh install for Linux Mint 21, I was having issue pairing my devices ==> https://github.com/blueman-project/blueman/issues/1824 , which resolved around a View setting on the hidden devices.

Current Problem Statement ==> all my devices are pairing - but not working.
The most important device - my headphones ... are not showing up as a Sound Device.
The headphones were previously working out of the box when I did a fresh Linux Mint 21 install a couple days ago.
When I got my wireless keyboard to pair ==> https://github.com/blueman-project/blueman/issues/1824 - all my devices started to not work by the next day.

I am working trying to get my headphones to work to troubleshoot.  They connect/pair/are trusted... but do not show on Sound UI.

[image] 

I have rebooted several times, and below is are my logs per troubleshooting page BKM - from killall  to pairing requests.

`blueman-applet 10.29.04 INFO     PluginManager:87 load_plugin: ['NMDUNSupport', 'StandardItems', 'AutoConnect', 'PowerManager', 'StatusNotifierItem', 'Networking', 'Menu', 'ExitItem', 'ShowConnected', 'StatusIcon', 'RecentConns', 'SerialManager', 'AuthAgent', 'ConnectionNotifier', 'PPPSupport', 'DhcpClient', 'DisconnectItems', 'DiscvManager', 'NMPANSupport', 'DBusService', 'KillSwitch', 'TransferService', 'GameControllerWakelock', 'NetUsage']
blueman-applet 10.29.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 10.29.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 10.29.04 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Address': '10:F0:05:F5:BF:E4', 'AddressType': 'public', 'Name': 'Wisteria', 'Alias': 'Wisteria', 'Class': 8126732, 'Powered': True, 'Discoverable': False, 'DiscoverableTimeout': 0, 'Pairable': False, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['00001133-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00001104-0000-1000-8000-00805f9b34fb', '00005005-0000-1000-8000-0002ee000001', '00001112-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '0000111f-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb', '00001108-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0540', 'Roles': ['central', 'peripheral']}
blueman-applet 10.29.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 10.29.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 10.29.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 10.29.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 10.29.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusNotifierItem.StatusNotifierItem'>
blueman-applet 10.29.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 10.29.04 INFO     Networking:35 load_nap_settings: Loading NAP settings
blueman-applet 10.29.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 10.29.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 10.29.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 10.29.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 10.29.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 10.29.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier'>
blueman-applet 10.29.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.PPPSupport.PPPSupport'>
blueman-applet 10.29.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DhcpClient.DhcpClient'>
blueman-applet 10.29.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 10.29.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 10.29.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 10.29.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 10.29.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'>
blueman-applet 10.29.04 INFO     KillSwitch:106 io_event  : killswitch registered 0
blueman-applet 10.29.04 INFO     KillSwitch:122 io_event  : State: True
blueman-applet 10.29.04 INFO     PowerManager:179 update_power_state: off False | foff False | on True | current state True | new state True
blueman-applet 10.29.04 INFO     KillSwitch:85 _on_connman_vanished: net.connman vanished
blueman-applet 10.29.04 INFO     Applet:63 _on_dbus_name_appeared: org.bluez :1.10
blueman-applet 10.29.04 INFO     Functions:114 launch    : Gtk eventtime is 0, not using LaunchContext
blueman-applet 10.29.04 INFO     Networking:71 set_nap   : set nap False
blueman-applet 10.29.04 INFO     BluezAgent:56 register_agent: Register Agent
blueman-applet 10.29.04 INFO     TransferService:234 _on_dbus_name_appeared: org.bluez.obex :1.71
blueman-applet 10.29.04 INFO     ShowConnected:50 enumerate_connections: Found 0 existing connections
blueman-applet 10.29.04 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Pairable': True}
blueman-applet 10.29.04 DEBUG    DiscvManager:92 on_adapter_property_changed: prop Pairable True
blueman-applet 10.29.04 INFO     AgentManager:17 on_registered: /org/bluez/obex/agent/blueman
blueman-applet 10.29.05 INFO     PowerManager:140 request_power_state: Requesting True
blueman-applet 10.29.05 INFO     KillSwitch:138 on_power_state_change_requested: True
blueman-applet 10.29.05 DEBUG    KillSwitch:151 on_power_state_change_requested: Using mechanism to set state: True
blueman-applet 10.29.05 INFO     PowerManager:125 check     : callbacks done
blueman-applet 10.29.05 INFO     PowerManager:96 set_adapter_state: True
blueman-applet 10.29.05 INFO     PowerManager:179 update_power_state: off False | foff False | on True | current state True | new state True
blueman-applet 10.29.12 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Discovering': True}
blueman-applet 10.29.12 DEBUG    DiscvManager:92 on_adapter_property_changed: prop Discovering True
blueman-applet 10.29.13 DEBUG    Manager:52 _on_object_added: Device1: /org/bluez/hci0/dev_AC_B1_EE_6D_F2_61
[... 37 lines cut ...]
blueman-applet 10.31.34 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.DisconnectItems.DisconnectItems object at 0x7f58425bce50>
blueman-applet 10.31.34 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.DhcpClient.DhcpClient object at 0x7f58425bd300>
blueman-applet 10.31.34 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.PPPSupport.PPPSupport object at 0x7f58425bccd0>
blueman-applet 10.31.34 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier object at 0x7f58425bd1b0>
[... cut ...]
blueman-applet 10.31.34 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <StatusIcon.StatusIcon object at 0x7f58425a5d40 (blueman+plugins+applet+StatusIcon+StatusIcon at 0x56101f054500)>
blueman-applet 10.31.34 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.Menu.Menu object at 0x7f584b3e9bd0>
`

I deleted the audiopulse config file and rebooted to troubleshoot. ==>https://askubuntu.com/a/1155043

This is a blueman issue I think.  All my devices worked fine with Linux Mint 20.3.
I'm not adverse to doing another fresh install, but something whacky has gone down with my bluetooth.  
Any help would be appreciated.

cschramm, 2022-08-14:
If connecting to that device works - and everything looks like it does - pulseaudio already handled the connection, so I clearly expect it to know the device. I'd first check if it shows up in `pactl list cards`.

8x1053, 2022-08-14:
> I'd first check if it shows up in pactl list cards.
I'm not seeing anything in the list bluetooth like.

`tim@Wisteria:~$ pactl list cards
Card #0
	Name: alsa_card.usb-046d_Logitech_BRIO_B267094D-03
	Driver: module-alsa-card.c
	Owner Module: 7
	Properties:
		alsa.card = "1"
		alsa.card_name = "Logitech BRIO"
		alsa.long_card_name = "Logitech BRIO at usb-0000:00:14.0-2.2, super speed"
		alsa.driver_name = "snd_usb_audio"
		device.bus_path = "pci-0000:00:14.0-usb-0:2.2:1.3"
		sysfs.path = "/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2.2/2-2.2:1.3/sound/card1"
		udev.id = "usb-046d_Logitech_BRIO_B267094D-03"
		device.bus = "usb"
		device.vendor.id = "046d"
		device.vendor.name = "Logitech, Inc."
		device.product.id = "085e"
		device.product.name = "BRIO Ultra HD Webcam"
		device.serial = "046d_Logitech_BRIO_B267094D"
		device.form_factor = "webcam"
		device.string = "1"
		device.description = "BRIO Ultra HD Webcam"
		module-udev-detect.discovered = "1"
		device.icon_name = "camera-web-usb"
	Profiles:
		input:analog-stereo: Analog Stereo Input (sinks: 0, sources: 1, priority: 65, available: yes)
		input:iec958-stereo: Digital Stereo (IEC958) Input (sinks: 0, sources: 1, priority: 55, available: yes)
		off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
	Active Profile: input:analog-stereo
	Ports:
		analog-input-mic: Microphone (type: Mic, priority: 8700, latency offset: 0 usec, availability unknown)
			Properties:
				device.icon_name = "audio-input-microphone"
			Part of profile(s): input:analog-stereo
		iec958-stereo-input: Digital Input (S/PDIF) (type: SPDIF, priority: 0, latency offset: 0 usec, availability unknown)
			Part of profile(s): input:iec958-stereo

Card #1
	Name: alsa_card.pci-0000_00_1f.3
	Driver: module-alsa-card.c
	Owner Module: 8
	Properties:
		alsa.card = "0"
		alsa.card_name = "HDA Intel PCH"
		alsa.long_card_name = "HDA Intel PCH at 0xb1328000 irq 153"
		alsa.driver_name = "snd_hda_intel"
		device.bus_path = "pci-0000:00:1f.3"
		sysfs.path = "/devices/pci0000:00/0000:00:1f.3/sound/card0"
		device.bus = "pci"
		device.vendor.id = "8086"
		device.vendor.name = "Intel Corporation"
		device.product.id = "9d71"
		device.product.name = "Sunrise Point-LP HD Audio"
		device.form_factor = "internal"
		device.string = "0"
		device.description = "Built-in Audio"
		module-udev-detect.discovered = "1"
		device.icon_name = "audio-card-pci"
	Profiles:
		input:analog-stereo: Analog Stereo Input (sinks: 0, sources: 1, priority: 32833, available: yes)
		output:analog-stereo: Analog Stereo Output (sinks: 1, sources: 0, priority: 39268, available: yes)
		output:analog-stereo+input:analog-stereo: Analog Stereo Duplex (sinks: 1, sources: 1, priority: 39333, available: yes)
		output:hdmi-stereo: Digital Stereo (HDMI) Output (sinks: 1, sources: 0, priority: 38668, available: yes)
		output:hdmi-stereo+input:analog-stereo: Digital Stereo (HDMI) Output + Analog Stereo Input (sinks: 1, sources: 1, priority: 38733, available: yes)
		output:hdmi-stereo-extra1: Digital Stereo (HDMI 2) Output (sinks: 1, sources: 0, priority: 5700, available: no)
		output:hdmi-stereo-extra1+input:analog-stereo: Digital Stereo (HDMI 2) Output + Analog Stereo Input (sinks: 1, sources: 1, priority: 5765, available: yes)
		output:hdmi-surround-extra1: Digital Surround 5.1 (HDMI 2) Output (sinks: 1, sources: 0, priority: 600, available: no)
[... 36 lines cut ...]
		analog-output-headphones: Headphones (type: Headphones, priority: 9900, latency offset: 0 usec, not available)
			Properties:
				device.icon_name = "audio-headphones"
			Part of profile(s): output:analog-stereo, output:analog-stereo+input:analog-stereo
		hdmi-output-0: HDMI / DisplayPort (type: HDMI, priority: 5900, latency offset: 0 usec, available)
			Properties:
				device.icon_name = "video-display"
				device.product.name = "32G2WG8"
			Part of profile(s): output:hdmi-stereo, output:hdmi-stereo+input:analog-stereo
		hdmi-output-1: HDMI / DisplayPort 2 (type: HDMI, priority: 5800, latency offset: 0 usec, not available)
			Properties:
				device.icon_name = "video-display"
			Part of profile(s): output:hdmi-stereo-extra1, output:hdmi-stereo-extra1+input:analog-stereo, output:hdmi-surround-extra1, output:hdmi-surround-extra1+input:analog-stereo, output:hdmi-surround71-extra1, output:hdmi-surround71-extra1+input:analog-stereo
		hdmi-output-2: HDMI / DisplayPort 3 (type: HDMI, priority: 5700, latency offset: 0 usec, not available)
			Properties:
				device.icon_name = "video-display"
			Part of profile(s): output:hdmi-stereo-extra2, output:hdmi-stereo-extra2+input:analog-stereo, output:hdmi-surround-extra2, output:hdmi-surround-extra2+input:analog-stereo, output:hdmi-surround71-extra2, output:hdmi-surround71-extra2+input:analog-stereo
		hdmi-output-3: HDMI / DisplayPort 4 (type: HDMI, priority: 5600, latency offset: 0 usec, not available)
			Properties:
				device.icon_name = "video-display"
			Part of profile(s): output:hdmi-stereo-extra3, output:hdmi-stereo-extra3+input:analog-stereo, output:hdmi-surround-extra3, output:hdmi-surround-extra3+input:analog-stereo, output:hdmi-surround71-extra3, output:hdmi-surround71-extra3+input:analog-stereo
		hdmi-output-4: HDMI / DisplayPort 5 (type: HDMI, priority: 5500, latency offset: 0 usec, not available)
			Properties:
				device.icon_name = "video-display"
			Part of profile(s): output:hdmi-stereo-extra4, output:hdmi-stereo-extra4+input:analog-stereo, output:hdmi-surround-extra4, output:hdmi-surround-extra4+input:analog-stereo, output:hdmi-surround71-extra4, output:hdmi-surround71-extra4+input:analog-stereo`

cschramm, 2022-08-14:
Assuming that the device was connected, there's in fact no card. However, pulseaudio (or another server) must have handled the connection or it would have failed otherwise, so I'd check its and BlueZ' logs (`journalctl --user -u pulseaudio.service` and `sudo journalctl -u bluetooth.service`) at the time of the connection.

8x1053, 2022-08-14:
Aug 14 14:35:06 Wisteria bluetoothd[795]: src/profile.c:connect_io() Unable to connect Hands-Free Voice gateway: connect to CC:98:8B:C1:5D:C7: Device or resource busy (16)
Aug 14 14:35:06 Wisteria bluetoothd[795]: src/profile.c:record_cb() Connecting Hands-Free Voice gateway failed: Input/output error
Aug 14 14:35:06 Wisteria bluetoothd[795]: src/profile.c:ext_connect() Headset Voice gateway failed connect to CC:98:8B:C1:5D:C7: Connection refused (111)

8x1053, 2022-08-14:
Aug 14 14:33:44 Wisteria pulseaudio[1246]: Could not find org.bluez.BatteryProviderManager1.RegisterBatteryProvider(), is bluetoothd started with experimental features enabled (-E flag)?

8x1053, 2022-08-14:
@cschramm 
I did a fresh install of Linux Mint 21and my headphones are working now, but my other devices are not showing on the search (they briefly showed when I first brought up Blueman?? Not sure what is going on.  My wireless keyboard is not coming up either.  I'm wondering if there is some setting.

8x1053, 2022-08-14:
The problem statement has changed post re-install of OS, but it is still problematic.  I will close this issue since it is different now, and push on.  I am going to try and reinstall one more time and connect / pair / trust all my go to Bluetooth devices post bootup, to see if I can get these to work.  My headphones are working post second reboot once I have them paired.  

If I still can't get this to go ... I will repost the issue. 

Thanks for looking at this.

closed 2022-08-14 by 8x1053

8x1053, 2022-08-14:
OK... got all my devices to work!!!!

Re-install of Linux was the key for me.  Bluetooth is still kludgy for me but hopefully this will settle as I use LM 21 going forward.

powerfrontier, 2022-09-12:
I have the same issue. For me, re-install is not an option. Anybody could help, please?

`journalctl --user -u pulseaudio.service`
```
-- Boot 6bd81a7797e948cd83bd118a9bb14eb0 --
sep 12 18:05:59 avalon-mini systemd[1211]: Starting Sound Service...
sep 12 18:05:59 avalon-mini pulseaudio[1318]: libva info: VA-API version 1.14.0
sep 12 18:05:59 avalon-mini pulseaudio[1318]: libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
sep 12 18:05:59 avalon-mini pulseaudio[1318]: libva info: Found init function __vaDriverInit_1_14
sep 12 18:05:59 avalon-mini pulseaudio[1318]: libva info: va_openDriver() returns 0
sep 12 18:06:00 avalon-mini systemd[1211]: Started Sound Service.
sep 12 18:06:00 avalon-mini pulseaudio[1220]: Could not find org.bluez.BatteryProviderManager1.RegisterBatteryProvider(), is bluetoothd started with experimental features enabled (-E flag)?
lines 300-356/356 (END)
```
`sudo journalctl -u bluetooth.service`
```
-- Boot 6bd81a7797e948cd83bd118a9bb14eb0 --
sep 12 18:05:35 avalon-mini systemd[1]: Starting Bluetooth service...
sep 12 18:05:36 avalon-mini bluetoothd[747]: Bluetooth daemon 5.64
sep 12 18:05:36 avalon-mini bluetoothd[747]: Starting SDP server
sep 12 18:05:36 avalon-mini bluetoothd[747]: Bluetooth management interface 1.21 initialized
sep 12 18:05:36 avalon-mini systemd[1]: Started Bluetooth service.
sep 12 18:05:38 avalon-mini bluetoothd[747]: Endpoint registered: sender=:1.34 path=/MediaEndpoint/A2DPSource/ldac_hq
sep 12 18:05:38 avalon-mini bluetoothd[747]: Endpoint registered: sender=:1.34 path=/MediaEndpoint/A2DPSource/ldac_sq
sep 12 18:05:38 avalon-mini bluetoothd[747]: Endpoint registered: sender=:1.34 path=/MediaEndpoint/A2DPSource/ldac_mq
sep 12 18:05:38 avalon-mini bluetoothd[747]: Endpoint registered: sender=:1.34 path=/MediaEndpoint/A2DPSink/aptx_hd
sep 12 18:05:38 avalon-mini bluetoothd[747]: Endpoint registered: sender=:1.34 path=/MediaEndpoint/A2DPSource/aptx_hd
sep 12 18:05:38 avalon-mini bluetoothd[747]: Endpoint registered: sender=:1.34 path=/MediaEndpoint/A2DPSink/aptx
sep 12 18:05:38 avalon-mini bluetoothd[747]: Endpoint registered: sender=:1.34 path=/MediaEndpoint/A2DPSource/aptx
sep 12 18:05:38 avalon-mini bluetoothd[747]: Endpoint registered: sender=:1.34 path=/MediaEndpoint/A2DPSink/sbc
sep 12 18:05:38 avalon-mini bluetoothd[747]: Endpoint registered: sender=:1.34 path=/MediaEndpoint/A2DPSource/sbc
sep 12 18:05:38 avalon-mini bluetoothd[747]: Endpoint registered: sender=:1.34 path=/MediaEndpoint/A2DPSink/sbc_xq_453
sep 12 18:05:38 avalon-mini bluetoothd[747]: Endpoint registered: sender=:1.34 path=/MediaEndpoint/A2DPSource/sbc_xq_453
sep 12 18:05:38 avalon-mini bluetoothd[747]: Endpoint registered: sender=:1.34 path=/MediaEndpoint/A2DPSink/sbc_xq_512
sep 12 18:05:38 avalon-mini bluetoothd[747]: Endpoint registered: sender=:1.34 path=/MediaEndpoint/A2DPSource/sbc_xq_512
sep 12 18:05:38 avalon-mini bluetoothd[747]: Endpoint registered: sender=:1.34 path=/MediaEndpoint/A2DPSink/sbc_xq_552
sep 12 18:05:38 avalon-mini bluetoothd[747]: Endpoint registered: sender=:1.34 path=/MediaEndpoint/A2DPSource/sbc_xq_552
sep 12 18:05:59 avalon-mini bluetoothd[747]: Endpoint unregistered: sender=:1.34 path=/MediaEndpoint/A2DPSource/ldac_hq
sep 12 18:05:59 avalon-mini bluetoothd[747]: Endpoint unregistered: sender=:1.34 path=/MediaEndpoint/A2DPSource/ldac_sq
sep 12 18:05:59 avalon-mini bluetoothd[747]: Endpoint unregistered: sender=:1.34 path=/MediaEndpoint/A2DPSource/ldac_mq
sep 12 18:05:59 avalon-mini bluetoothd[747]: Endpoint unregistered: sender=:1.34 path=/MediaEndpoint/A2DPSink/aptx_hd
sep 12 18:05:59 avalon-mini bluetoothd[747]: Endpoint unregistered: sender=:1.34 path=/MediaEndpoint/A2DPSource/aptx_hd
sep 12 18:05:59 avalon-mini bluetoothd[747]: Endpoint unregistered: sender=:1.34 path=/MediaEndpoint/A2DPSink/aptx
sep 12 18:05:59 avalon-mini bluetoothd[747]: Endpoint unregistered: sender=:1.34 path=/MediaEndpoint/A2DPSource/aptx
sep 12 18:05:59 avalon-mini bluetoothd[747]: Endpoint unregistered: sender=:1.34 path=/MediaEndpoint/A2DPSink/sbc
sep 12 18:05:59 avalon-mini bluetoothd[747]: Endpoint unregistered: sender=:1.34 path=/MediaEndpoint/A2DPSource/sbc
sep 12 18:05:59 avalon-mini bluetoothd[747]: Endpoint unregistered: sender=:1.34 path=/MediaEndpoint/A2DPSink/sbc_xq_453
sep 12 18:05:59 avalon-mini bluetoothd[747]: Endpoint unregistered: sender=:1.34 path=/MediaEndpoint/A2DPSource/sbc_xq_453
sep 12 18:05:59 avalon-mini bluetoothd[747]: Endpoint unregistered: sender=:1.34 path=/MediaEndpoint/A2DPSink/sbc_xq_512
sep 12 18:05:59 avalon-mini bluetoothd[747]: Endpoint unregistered: sender=:1.34 path=/MediaEndpoint/A2DPSource/sbc_xq_512
sep 12 18:05:59 avalon-mini bluetoothd[747]: Endpoint unregistered: sender=:1.34 path=/MediaEndpoint/A2DPSink/sbc_xq_552
sep 12 18:05:59 avalon-mini bluetoothd[747]: Endpoint unregistered: sender=:1.34 path=/MediaEndpoint/A2DPSource/sbc_xq_552
sep 12 18:06:00 avalon-mini bluetoothd[747]: Endpoint registered: sender=:1.49 path=/MediaEndpoint/A2DPSource/ldac_hq
sep 12 18:06:00 avalon-mini bluetoothd[747]: Endpoint registered: sender=:1.49 path=/MediaEndpoint/A2DPSource/ldac_sq
sep 12 18:06:00 avalon-mini bluetoothd[747]: Endpoint registered: sender=:1.49 path=/MediaEndpoint/A2DPSource/ldac_mq
sep 12 18:06:00 avalon-mini bluetoothd[747]: Endpoint registered: sender=:1.49 path=/MediaEndpoint/A2DPSink/aptx_hd
sep 12 18:06:00 avalon-mini bluetoothd[747]: Endpoint registered: sender=:1.49 path=/MediaEndpoint/A2DPSource/aptx_hd
sep 12 18:06:00 avalon-mini bluetoothd[747]: Endpoint registered: sender=:1.49 path=/MediaEndpoint/A2DPSink/aptx
sep 12 18:06:00 avalon-mini bluetoothd[747]: Endpoint registered: sender=:1.49 path=/MediaEndpoint/A2DPSource/aptx
sep 12 18:06:00 avalon-mini bluetoothd[747]: Endpoint registered: sender=:1.49 path=/MediaEndpoint/A2DPSink/sbc
sep 12 18:06:00 avalon-mini bluetoothd[747]: Endpoint registered: sender=:1.49 path=/MediaEndpoint/A2DPSource/sb
```

cschramm, 2022-09-12:
@powerfrontier: What problem do you have? No card showing up in pulseaudio after connecting to an audio device? Check pulseaudio's logs at the time of connecting the device (In case that's what your dumps show, there's nothing special. You might want to enable some verbose logging or something.). It's most likely an issue with (Linux Mint 21's) pulseaudio. (We also have dubious reports of Linux Mint 21 using PipeWire instead in #1837.)

powerfrontier, 2022-09-14:
@cschramm  Yes, no device showing on the pulseaudio's output screen. Also, what I was saying in the edit, no "Audio profile" menu item info, this time on blueman. I'm not on the problematic system right now, but I will try to update the logs with more verbose info tomorrow. Thanks for response anyway

cschramm, 2022-09-14:
That menu item is an interface to pulseaudio, so it is expected to be absent if there is no matching card in pulseaudio.

powerfrontier, 2022-09-15:
`tail -f /var/log/syslog`
```
Sep 15 04:18:30 avalon-mini systemd[1208]: Reached target Bluetooth.
Sep 15 04:18:38 avalon-mini dbus-daemon[1225]: [session uid=1000 pid=1225] Activating via systemd: service name='org.blueman.Manager' unit='blueman-manager.service' requested by ':1.60' (uid=1000 pid=1525 comm="/usr/bin/python3 /usr/bin/blueman-applet " label="unconfined")
Sep 15 04:18:38 avalon-mini systemd[1208]: Starting Bluetooth Manager...
Sep 15 04:18:38 avalon-mini dbus-daemon[1225]: [session uid=1000 pid=1225] Successfully activated service 'org.blueman.Manager'
Sep 15 04:18:38 avalon-mini systemd[1208]: Started Bluetooth Manager.
Sep 15 04:18:38 avalon-mini blueman-manager[17963]: Theme parsing error: gtk.css:5465:0: 'none' is not a valid color name
Sep 15 04:18:38 avalon-mini blueman-manager[17963]: Theme parsing error: softwarecenter.css:49:29: The style property GtkButton:inner-border is deprecated and shouldn't be used anymore. It will be removed in a future version
Sep 15 04:18:38 avalon-mini blueman-manager[17963]: Theme parsing error: softwarecenter.css:50:31: The style property GtkButton:default-border is deprecated and shouldn't be used anymore. It will be removed in a future version
Sep 15 04:18:38 avalon-mini blueman-manager[17963]: Theme parsing error: softwarecenter.css:51:39: The style property GtkButton:default-outside-border is deprecated and shouldn't be used anymore. It will be removed in a future version

```
Some more debug logs. This seems ok, the only errors seems unrelated to my issue. I don't know how to get more useful logs. Any suggestion?

cschramm, 2022-09-15:
Again, the sound server is the only interesting part here. If pulseaudio does not log any error, see e.g. https://fedoraproject.org/wiki/How_to_debug_PulseAudio_problems#General_advice for verbose logging.

<details>
<summary>A proper setup looks something like this</summary>

```
(  14.707|   9.302) D: [pulseaudio] bluez5-util.c: Properties changed in device /org/bluez/hci0/dev_CC_9F_7A_A5_47_34
(  15.888|   1.180) D: [pulseaudio] backend-native.c: dbus: path=/Profile/HSPHSProfile, interface=org.bluez.Profile1, member=NewConnection
(  15.888|   0.000) D: [pulseaudio] backend-native.c: dbus: NewConnection path=/org/bluez/hci0/dev_CC_9F_7A_A5_47_34, fd=22, profile headset_audio_gateway
(  15.888|   0.000) I: [pulseaudio] backend-native.c: doing listen
(  15.888|   0.000) D: [pulseaudio] bluez5-util.c: Transport /org/bluez/hci0/dev_CC_9F_7A_A5_47_34/fd22 state: disconnected -> idle
(  15.888|   0.000) D: [pulseaudio] bluez5-util.c: Checking if device Nokia 8 (CC:9F:7A:A5:47:34) supports profile a2dp_sink: false
(  15.888|   0.000) D: [pulseaudio] bluez5-util.c: Checking if device Nokia 8 (CC:9F:7A:A5:47:34) supports profile a2dp_source: true
(  15.888|   0.000) D: [pulseaudio] bluez5-util.c: Checking if device Nokia 8 (CC:9F:7A:A5:47:34) supports profile headset_head_unit: false
(  15.888|   0.000) D: [pulseaudio] bluez5-util.c: Checking if device Nokia 8 (CC:9F:7A:A5:47:34) supports profile headset_audio_gateway: true
(  15.888|   0.000) D: [pulseaudio] bluez5-util.c: Checking if device Nokia 8 (CC:9F:7A:A5:47:34) supports profile handsfree_head_unit: false
(  15.888|   0.000) D: [pulseaudio] bluez5-util.c: Checking if device Nokia 8 (CC:9F:7A:A5:47:34) supports profile handsfree_audio_gateway: false
(  15.888|   0.000) D: [pulseaudio] backend-native.c: Transport /org/bluez/hci0/dev_CC_9F_7A_A5_47_34/fd22 available for profile headset_audio_gateway
(  16.640|   0.752) D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Introspectable found, skipping
(  16.640|   0.000) D: [pulseaudio] bluez5-util.c: Parsing remote endpoint /org/bluez/hci0/dev_CC_9F_7A_A5_47_34/sep1
(  16.640|   0.000) D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Properties found, skipping
(  16.641|   0.000) D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Introspectable found, skipping
(  16.641|   0.000) D: [pulseaudio] bluez5-util.c: Parsing remote endpoint /org/bluez/hci0/dev_CC_9F_7A_A5_47_34/sep3
(  16.641|   0.000) D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Properties found, skipping
(  16.642|   0.000) D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Introspectable found, skipping
(  16.642|   0.000) D: [pulseaudio] bluez5-util.c: Parsing remote endpoint /org/bluez/hci0/dev_CC_9F_7A_A5_47_34/sep4
(  16.642|   0.000) D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Properties found, skipping
(  16.643|   0.000) D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Introspectable found, skipping
(  16.643|   0.000) D: [pulseaudio] bluez5-util.c: Parsing remote endpoint /org/bluez/hci0/dev_CC_9F_7A_A5_47_34/sep5
(  16.643|   0.000) D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Properties found, skipping
(  16.644|   0.000) D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Introspectable found, skipping
(  16.644|   0.000) D: [pulseaudio] bluez5-util.c: Parsing remote endpoint /org/bluez/hci0/dev_CC_9F_7A_A5_47_34/sep6
(  16.645|   0.000) D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Properties found, skipping
(  16.962|   0.317) D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Introspectable found, skipping
(  16.962|   0.000) D: [pulseaudio] bluez5-util.c: Unknown interface org.bluez.MediaTransport1 found, skipping
(  16.962|   0.000) D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Properties found, skipping
(  16.962|   0.000) D: [pulseaudio] bluez5-util.c: dbus: path=/MediaEndpoint/A2DPSink/aptx_hd, interface=org.bluez.MediaEndpoint1, member=SetConfiguration
(  16.963|   0.000) D: [pulseaudio] bluez5-util.c: Transport /org/bluez/hci0/dev_CC_9F_7A_A5_47_34/fd2 state: disconnected -> idle
(  16.963|   0.000) D: [pulseaudio] bluez5-util.c: Checking if device Nokia 8 (CC:9F:7A:A5:47:34) supports profile a2dp_sink: false
(  16.963|   0.000) D: [pulseaudio] bluez5-util.c: Checking if device Nokia 8 (CC:9F:7A:A5:47:34) supports profile a2dp_source: true
(  16.963|   0.000) D: [pulseaudio] bluez5-util.c: Checking if device Nokia 8 (CC:9F:7A:A5:47:34) supports profile headset_head_unit: false
(  16.963|   0.000) D: [pulseaudio] bluez5-util.c: Checking if device Nokia 8 (CC:9F:7A:A5:47:34) supports profile headset_audio_gateway: true
(  16.963|   0.000) D: [pulseaudio] bluez5-util.c: Checking if device Nokia 8 (CC:9F:7A:A5:47:34) supports profile handsfree_head_unit: false
(  16.963|   0.000) D: [pulseaudio] bluez5-util.c: Checking if device Nokia 8 (CC:9F:7A:A5:47:34) supports profile handsfree_audio_gateway: false
(  16.963|   0.000) D: [pulseaudio] module-bluez5-discover.c: Loading module-bluez5-device path=/org/bluez/hci0/dev_CC_9F_7A_A5_47_34 autodetect_mtu=0 output_rate_refresh_interval_ms=500 avrcp_absolute_volume=1
[... cut ...]
(  16.975|   0.000) D: [pulseaudio] sink-input.c: Negotiated format: pcm, format.sample_format = "\"s24le\""  format.rate = "48000"  format.channels = "2"  format.channel_map = "\"front-left,front-right\""
(  16.975|   0.000) I: [pulseaudio] module-stream-restore.c: Restoring volume for sink input sink-input-by-media-role:music.
(  16.975|   0.000) D: [pulseaudio] module-suspend-on-idle.c: Sink alsa_output.pci-0000_00_1b.0.analog-stereo becomes busy, resuming.
(  16.975|   0.000) D: [pulseaudio] reserve-wrap.c: Successfully acquired reservation lock on device 'Audio1'
```
</details>

powerfrontier, 2022-09-16:
Thanks for the tip @cschramm. Here is last part of the output of `pulseaudio -vvvv` 

[](https://pastebin.com/BPCdyTD2)

The pairing of the headphone begins at line 51. I don't see any error, and the output seems to me very different than yours.

cschramm, 2022-09-16:
Yep, looks like (that instance of) pulseaudio does not handle the profile at all. Something else on your system takes care of the connection and that explains that nothing shows up in pulseaudio even though the connection works.

Some options:

* Another pulseaudio instance (but multiple instances normally do not run in parallel)
* PipeWire (but PipeWire has a pulseaudio compatibility layer; not sure if that would expose the card to pulseaudio itself as you're not supposed to run both sound servers to my understanding)
* BlueALSA (but pulseaudio would probably pick up the alsa cards)

In the bluetoothd you see the registration of media endpoints together with a D-Bus sender. You can try e.g. `d-feet` to find the corresponding process.

powerfrontier, 2022-09-20:
Sorry for the delay in responding. Do you mean that D-Bus message appears on the previous ` sudo journalctl -u bluetooth.service` log? or in the more recent `/var/log/syslog`? If it's the first one, I can't see it; and if it's the second one it appears to be the blueman process, isn't it?

cschramm, 2022-09-21:
In the bluetooth.service log, some sender `:1.34` on the system bus registers media endpoints, then surprisingly seems to unregister them, and then a sender `:1.49` registers some. You should be able to find the latest sender that registered media endpoints. Note that those names (numbers) are temporary, so do not expect to still find a / the correct `:1.49`.